raspberrypi/rpicam-apps #2: imx290/327 issue?

vincentneo, 2021-02-03:
Hello,

I am trying to use a IMX 327 board with the raspberry pi but kept getting:
```
pi@raspberrypi:~/libcamera-apps/build $ ./libcamera-hello
libEGL warning: DRI2: failed to authenticate
[0:00:37.985155973] [1011]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2201-fb9051ff
[0:00:38.002265088] [1021] ERROR V4L2 v4l2_device.cpp:191 'imx290 10-001a': Control 0x009a0922 not found
[0:00:38.459742431] [1021] ERROR IPAProxy ipa_proxy.cpp:141 Configuration file 'imx290.json' not found for IPA module 'raspberrypi'
[0:00:38.465821858] [1011] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:00:38.466761442] [1011] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:00:38.467361129] [1011] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:00:38.467883421] [1011]  INFO Camera camera.cpp:890 configuring streams: (0) 1280x960-YUV420
[0:00:38.469223786] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt RG10 Score: 1856.67 (best 1856.67)
[0:00:38.469480244] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1280x720 fmt RG10 Score: 2146.67 (best 1856.67)
[0:00:38.469623577] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt RG12 Score: 856.667 (best 856.667)
[0:00:38.469747119] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1280x720 fmt RG12 Score: 1146.67 (best 856.667)
[0:00:38.469878161] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt pRAA Score: 1356.67 (best 856.667)
[0:00:38.469999931] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1280x720 fmt pRAA Score: 1646.67 (best 856.667)
[0:00:38.470192692] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt pRCC Score: 356.667 (best 356.667)
[0:00:38.470313994] [1021]  INFO RPI raspberrypi.cpp:119 Mode: 1280x720 fmt pRCC Score: 646.667 (best 356.667)
[0:00:38.470497275] [1021]  INFO RPI raspberrypi.cpp:594 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected mode: 1920x1080-pRCC
[0:00:38.474327796] [1021] ERROR IPARPI raspberrypi.cpp:545 Unable to find sensor control 0x009e0903
[0:00:38.474604671] [1021] ERROR IPARPI raspberrypi.cpp:302 Sensor control validation failed.
[0:00:38.474788213] [1021] ERROR RPI raspberrypi.cpp:1217 IPA configuration failed!
[0:00:38.475047900] [1021] ERROR RPI raspberrypi.cpp:709 Failed to configure the IPA: -32
ERROR: *** failed to configure streams ***
```

I have already set the dtoverlay to imx290 in /boot/config.txt
May I know what went wrong?

Thank you.

kbingham, 2021-02-03:
There are some controls missing in the sensor driver, in the kernel you're using. We should make those prints easier to read and interpret, but ultimately you'll need to extend the driver accordingly

vincentneo, 2021-02-03:
I'm using kernel 5.4.83, which should have IMX290/327 drivers, I thought?

kbingham, 2021-02-03:
You very likely have the driver, yes as i can see reference to imx290 in the log. The issue is that newer features are required by libcamera which must be added to the driver.

I'll dig out the list of required controls when I get to my laptop later

vincentneo, 2021-02-03:
hmm I must have done something wrongly?

took my official v1 raspberry pi camera to try and this is what I got:
```
libEGL warning: DRI2: failed to authenticate
[0:00:34.259020872] [1019]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2201-fb9051ff
[0:00:34.319523215] [1029] ERROR V4L2 v4l2_device.cpp:191 'ov5647 10-0036': Control 0x009a0922 not found
[0:00:34.320119049] [1029] ERROR V4L2 v4l2_subdevice.cpp:285 'ov5647 10-0036': Unable to get rectangle 2 on pad 0: Invalid argument
[0:00:34.320294257] [1029]  WARN CameraSensor camera_sensor.cpp:288 'ov5647 10-0036': The PixelArraySize property has been defaulted to (0x0)/2592x1944
[0:00:34.320449413] [1029]  WARN CameraSensor camera_sensor.cpp:313 'ov5647 10-0036': The sensor kernel driver needs to be fixed
[0:00:34.320503059] [1029]  WARN CameraSensor camera_sensor.cpp:315 'ov5647 10-0036': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:00:34.777910038] [1019]  INFO Camera camera.cpp:890 configuring streams: (0) 1280x960-YUV420
[0:00:34.779339517] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 2592x1944 fmt BG10 Score: 2074 (best 2074)
[0:00:34.779703736] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt BG10 Score: 1856.67 (best 1856.67)
[0:00:34.779887382] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 1296x972 fmt BG10 Score: 1507 (best 1507)
[0:00:34.780123944] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt BG10 Score: 3740 (best 1507)
[0:00:34.780313319] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt BA81 Score: 4740 (best 1507)
[0:00:34.780490715] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 2592x1944 fmt pBAA Score: 1574 (best 1507)
[0:00:34.780902746] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt pBAA Score: 1356.67 (best 1356.67)
[0:00:34.781081392] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 1296x972 fmt pBAA Score: 1007 (best 1007)
[0:00:34.781371965] [1029]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt pBAA Score: 3240 (best 1007)
[0:00:34.781820246] [1029]  INFO RPI raspberrypi.cpp:594 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[0:00:35.018748319] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: Sharpness = 1.000000
[0:00:35.019087069] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: Saturation = 1.000000
[0:00:35.019187538] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: Contrast = 1.000000
[0:00:35.019272538] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: AeMeteringMode = 0
[0:00:35.019415090] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: FrameDurations = [ 33333, 33333 ]
[0:00:35.019539725] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: AeExposureMode = 0
[0:00:35.019615923] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: ExposureValue = 0.000000
[0:00:35.019734048] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: AwbMode = 0
[0:00:35.019815402] [1032]  INFO IPARPI raspberrypi.cpp:623 Request ctrl: Brightness = 0.000000
Segmentation fault
```
can't be that this doesn't support that camera right?

davidplowman, 2021-02-04:
Hi

Thanks for the report. As Kieran says it sounds like the driver needs updating to support libcamera properly. There are a couple of other things you would need to do as well if you want to support a new sensor.

1. The Raspberry Pi pipeline handler will require a CamHelperImx290 class. You could base it on, for example, libcamera/src/ipa/raspberrypi/cam_helper_imx219.cpp.
2. You'll need a camera tuning file. To get you going just make a copy of libcamera/src/ipa/raspberrypi/data/uncalibrated.json in the same folder renaming it as imx290.json (matching the name advertised in your driver).

There's more explanation of this in https://www.raspberrypi.org/documentation/linux/software/libcamera/rpi_SOFT_libcamera_1p2.pdf (sections 3.2 and 6.5, among others).

davidplowman, 2021-02-04:
To answer the question about the ov5647. It seems to work OK for me, but there are a couple of things in the log you posted.

1. I don't see that "Unable to get rectangle" message, so it might be worth checking software versions. I get:
```bash
$ uname -a
Linux raspberrypi 5.10.9-v7l+ #1396 SMP Thu Jan 21 15:57:42 GMT 2021 armv7l GNU/Linux
$ sudo vcgencmd version
Jan 21 2021 16:02:55
Copyright (c) 2012 Broadcom
version a5bf70c44ba93e99f955a66530916240a1661132 (clean) (release) (start)\
```
2. I'm also bothered by that EGL "failed to authenticate" message. Actually I wonder if this is more likely to be the cause for the crash. It's rather beyond my area of expertise but maybe it relates to the display drivers you're using. In my `/boot/config.txt` I have
```
dtoverlay=vc4-fkms-v3d
```
If this remains a problem it's probably worth posting on the Raspberry Pi forums where people who know more about this should be able to answer. Thanks!

vincentneo, 2021-02-07:
Alright then, I guess its not so easy to get an imx290 working...

By the way, I do not have `dtoverlay=vc4-fkms-v3d` on my pi. Might try putting that in.

Thanks.

davidplowman, 2021-03-09:
Hi again, you might be interested to know that libcamera now includes support for imx290/imx327 sensors running on the Raspberry Pi. This probably seems a reasonable point to close this issue. Thanks!

closed 2021-03-09 by davidplowman