raspberrypi/rpicam-apps #3: DRM Error: No crts

Jul3k, 2021-02-06:
Running libcamera-apps except `libcamera-raw` fails instantly with:
`ERROR: *** drm: no crts ***`

I am having trouble to find resources on the cause of this problem. I am running Raspberry Pi OS Lite without a desktop server. Does libcamera-apps require a display attached?

Jul3k, 2021-02-06:
I hope I came the solution one step closer. I did not noticed that there was another uncommented line defining the dtoverlay.
Now I only have `dtoverlay=imx477` in the `boot/config.text`. But I still get an DRM error:

ERROR: *** drmOpen failed: No such file or directory ***

davidplowman, 2021-02-07:
I've certainly always run it with a display. However, there is a -n (no preview) option. Does that help?

Jul3k, 2021-02-07:
Yes! The -n option did the trick. Thank you.

closed 2021-02-07 by Jul3k