raspberrypi/rpicam-apps #6: ERROR: *** drm: no crts ***

initBasti, 2021-02-11:
Hello, Raspberry team :),

I am currently trying to install this wonderful project and the build process finishes without any errors. As soon as I try to start the `LIBCAMERA_LOG_LEVELS=0 ./libcamera-hello -v` command, I get the following:
```
pi@raspberrypi:~/libcamera-apps $ LIBCAMERA_LOG_LEVELS=0 ./build/libcamera-hello -v
[0:17:07.396442208] [858] DEBUG IPAProxy ipa_proxy.cpp:290 Registered proxy "IPAProxyLinux"
[0:17:07.397143071] [858] DEBUG IPAProxy ipa_proxy.cpp:290 Registered proxy "IPAProxyThread"
Options:
    verbose: 1
    timeout: 5000
    width: 0
    height: 0
    output: 
    rawfull: 0
    preview: default
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers, and configuration
Camera closed
ERROR: *** drm: no crts ***
```

My installation process is documented within the following ansible-playbook:
[https://termbin.com/xv90](https://termbin.com/xv90)

Additionally, I looked for other drm packages and tried out these two: `libdrm2`, `libdrm-common`
but without any success.

I recompiled the program with debugging information and found out that it crashes at libcamera-app.hpp:129

Greetings,
Sebastian

davidplowman, 2021-02-11:
Hi, are you running on a Pi 3? You may need to double-check whether you have `dtoverlay=vc4-fkms-v3d` in the `[all]` section of your `/boot/config.txt` file.

initBasti, 2021-02-13:
Hey David,
sorry for the delayed response, I am running a Pi 4.
Greetings,
Sebastian

davidplowman, 2021-02-13:
Puzzling. Probably still worth checking your /boot/config.txt for that dtoverlay. If it's still no good, perhaps you can post what `uname -a` and `sudo vcgencmd version` report.

initBasti, 2021-02-13:
Ah, the option was actually deactivated in the `[all]` section and was only active in the `[pi4]` section.

```
pi@raspberrypi:~/libcamera-apps $ uname -a
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~/libcamera-apps $ sudo vcgencmd version
\Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
```
Here is the full config.txt:
[https://termbin.com/lqnn](https://termbin.com/lqnn)

I rebooted after the config modification but was still greeted by a friendly:
```
pi@raspberrypi:~/libcamera-apps $ ./build/libcamera-vid -o ~/test.mp4
ERROR: *** drm: no crts ***
```

But it works good when I pass `-n`.

Greetings,
Sebastian

davidplowman, 2021-02-13:
I've just performed a fresh install of Raspberry Pi OS 32-bit, followed by the README.md instructions, and have working preview windows. I get:
```
pi@raspberrypi:~/libcamera-apps $ uname -a
Linux raspberrypi 5.10.14-v7l+ #1401 SMP Mon Feb 8 14:27:07 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~/libcamera-apps $ sudo vcgencmd version
Feb  8 2021 14:32:22 
Copyright (c) 2012 Broadcom
version 73b3cad64181954e67f6e9fe6d275378d3eefa10 (clean) (release) (start)
```
which is all a bit newer, though there's no reason that I know of to suppose it's related to the problem. Did you install everything onto a clean version of the OS?

initBasti, 2021-02-13:
Oh, I am sorry, I run a headless setup over ssh... I thought you meant that it is possible to run the apps without specifying the no preview button on headless setups. But I notice that I never mentioned that I run it over ssh ...  :S

davidplowman, 2021-02-14:
Ah OK, I'd not really understood. Maybe it would be friendlier if it just (issued a warning and) carried on without a preview window in these circumstances?

initBasti, 2021-02-14:
Yes, I believe that would be optimal, as the current behavior gives the feeling that the installation went wrong or that some dependency is missing, while in reality, it is just a default option that is incompatible with my setup.

davidplowman, 2021-02-15:
I've made this change now and pushed it, let me know if it doesn't work! Thanks.

closed 2021-02-15 by davidplowman