raspberrypi/rpicam-apps #13: install unsuccessful with HQ camera

felixloftus, 2021-03-01:
Hi, 
I've previously used libcam with a V2 camera module on my rpi4. I've just changed to a HQ camera module and am struggling. Initially I got a couple of images working but now I always get this error:

```
[0:01:24.772144596] [969]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2309-cc22d220
[0:01:24.781873166] [971]  WARN CameraSensor camera_sensor.cpp:307 'imx477 10-001a': Recommended V4L2 control 0x009a0922 not supported
[0:01:24.781994168] [971]  WARN CameraSensor camera_sensor.cpp:353 'imx477 10-001a': The sensor kernel driver needs to be fixed
[0:01:24.782038799] [971]  WARN CameraSensor camera_sensor.cpp:355 'imx477 10-001a': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:01:24.786257118] [971]  WARN CameraSensor camera_sensor.cpp:466 'imx477 10-001a': Failed to retrieve the camera location, setting to External
[0:01:25.014436689] [969]  WARN V4L2 v4l2_pixelformat.cpp:180 Unsupported V4L2 pixel format XB24
[0:01:25.015593878] [969]  INFO Camera camera.cpp:890 configuring streams: (0) 1280x960-YUV420
[0:01:25.016195333] [971]  INFO RPI raspberrypi.cpp:122 Mode: 1012x760 fmt RG10 Score: 2441.26 (best 2441.26)
[0:01:25.016321021] [971]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt RG12 Score: 1714.33 (best 1714.33)
[0:01:25.016378078] [971]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt RG12 Score: 827.329 (best 827.329)
[0:01:25.016434393] [971]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt RG12 Score: 921.167 (best 827.329)
[0:01:25.016493135] [971]  INFO RPI raspberrypi.cpp:122 Mode: 1012x760 fmt pRAA Score: 1941.26 (best 827.329)
[0:01:25.016553136] [971]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt pRCC Score: 1214.33 (best 827.329)
[0:01:25.016605600] [971]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt pRCC Score: 327.329 (best 327.329)
[0:01:25.016657157] [971]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt pRCC Score: 421.167 (best 327.329)
[0:01:25.016737177] [971]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
[0:01:25.125306683] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: Sharpness = 1.000000
[0:01:25.125471520] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: Saturation = 1.000000
[0:01:25.125559299] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: Brightness = 0.000000
[0:01:25.125630745] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: FrameDurations = [ 33333, 33333 ]
[0:01:25.125729802] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: AeMeteringMode = 0
[0:01:25.125869583] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: Contrast = 1.000000
[0:01:25.125938306] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: NoiseReductionMode = 1
[0:01:25.126054253] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: ExposureValue = 0.000000
[0:01:25.126148273] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: AwbMode = 0
[0:01:25.126211145] [974]  INFO IPARPI raspberrypi.cpp:614 Request ctrl: AeExposureMode = 0
[0:01:25.128044883] [971]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:01:25.128116032] [971]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:01:25.128157440] [971]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:01:25.825085736] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (6 left)
[0:01:25.860133682] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (5 left)
[0:01:25.895647692] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (4 left)
[0:01:25.929257020] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (3 left)
[0:01:25.962839348] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (2 left)
[0:01:25.996346805] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (1 left)
[0:01:26.030356862] [971]  INFO RPI raspberrypi.cpp:1620 Dropping frame at the request of the IPA (0 left)
terminate called after throwing an instance of 'std::runtime_error'
  what():  drmModeSetPlane failed: Permission denied
Aborted
```

I'm a bit lost as to whats going wrong. Any help would be appreciated

davidplowman, 2021-03-03:
It's evidently trying to show a DRM-based preview window, rather than an EGL one, and failing in the process. Can you say a bit more about your setup? (for what it's worth, the camera itself seems to be working fine)

- What kind of Pi are you using? (Pi 3 or 4, how much RAM)
- Do you have an HDMI display directly attached?
- Are you running X-Windows?
- Are you logged in directly to the Pi? (or, for example, over ssh)
- What command line are you using?
- From your description I understood that all you've done is replace the v2 module by an HQ cam (and edited the dtoverlay to match). Everything else about your setup and how you run it is identical. Is that right? If you simply switch back to the v2 module does the preview window work properly again?

Thanks!

felixloftus, 2021-03-03:
Hi, thanks for your response. 

- I'm running a pi4 with 2GB RAM

- I have a screen attached. It is the Adafruit PiTFT running as a FBCP, attached to the GPIO pins.

- I'm unsure if I'm running X-Windows. I've just looked this up but I'm a bit unsure how to check this

- I'm controlling the pi via SSH

- I'm using  Terminal on a Mac

- I tried out your final suggestion and I get the same error with the V2 module. It isn't working, so I guess the problem might be something to do with the screen. I'm happy to uninstall the drivers if you think that's worth trying

Thanks so much for your help

davidplowman, 2021-03-04:
So it sounds to me like the problem is that you're logged in remotely via ssh and can't open the display. And from the error message I would guess you're running X, otherwise I think the DRM window would have worked.

Assuming you're logged in as the same user as the Pi dekstop (user "pi" by default), then I believe you should have permissions to open an X window on the main display. Go to the folder contiaining the applications and try typing this: `DISPLAY=:0 ./libcamera-hello`.

Some observations on this:

- If you aren't logged in as the same user as the desktop, you'll have to look into X window permissions (beyond my area of expertise, I'm afraid). You can type `xhost +` to get round them (you'll probably have to do that from the Pi itself, not an ssh login), but be aware that you're opening a security hole with that.
- raspivid etc. will work because they are direct hardware overlays from the GPU and don't interact at all with the windowing system.
- Another option might be to avoid booting your Pi into X, and then I think the DRM preview would work. But obviously not if you want other applications and windows to run!

felixloftus, 2021-03-12:
Thank you very much for all this advice. For the time being, I've returned to the standard raspicam but I'll try some of these fixes when I have more time!

davidplowman, 2021-03-12:
Thanks for letting us know! I'll close the issue at this point, but please submit a new one if you come back to this and need further help.