raspberrypi/rpicam-apps #17: segmentation fault when using 1080p res (imx327)

vincentneo, 2021-03-13:
Hi,

I am getting Segmentation fault when setting --width 1920 --height 1080 when using the imx327, with my pi 3b+.
The preview window will appear only for a moment, and segmentation fault appears.

It does not happen when width and height is not set.

```
[0:13:54.413228670] [1334]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2390-e32d33b1
[0:13:54.430912021] [1338]  WARN CameraSensor camera_sensor.cpp:306 'imx290 10-001a': Recommended V4L2 control 0x009a0922 not supported
[0:13:54.431568269] [1338]  WARN CameraSensor camera_sensor.cpp:358 'imx290 10-001a': The sensor kernel driver needs to be fixed
[0:13:54.431746446] [1338]  WARN CameraSensor camera_sensor.cpp:360 'imx290 10-001a': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:13:54.442069238] [1338]  WARN CameraSensor camera_sensor.cpp:472 'imx290 10-001a': Failed to retrieve the camera location, setting to External
[0:13:54.616388274] [1334]  WARN V4L2 v4l2_pixelformat.cpp:180 Unsupported V4L2 pixel format XB24
[0:13:54.617047491] [1334] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:13:54.617675563] [1334] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:13:54.618043583] [1334] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:13:54.618370770] [1334]  INFO Camera camera.cpp:890 configuring streams: (0) 1920x1080-YUV420
[0:13:54.619355247] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 1500 (best 1500)
[0:13:54.619602434] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG10 Score: 3500 (best 1500)
[0:13:54.619725298] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG12 Score: 500 (best 500)
[0:13:54.619825455] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG12 Score: 2500 (best 500)
[0:13:54.619929673] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1000 (best 500)
[0:13:54.620076392] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRAA Score: 3000 (best 500)
[0:13:54.620186600] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRCC Score: 0 (best 0)
[0:13:54.620286339] [1338]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRCC Score: 2000 (best 0)
[0:13:54.620433214] [1338]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected mode: 1920x1080-pRCC
[0:13:54.738936733] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: Sharpness = 1.000000
[0:13:54.739254232] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: Saturation = 1.000000
[0:13:54.739365274] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: Brightness = 0.000000
[0:13:54.739455221] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: FrameDurations = [ 33333, 33333 ]
[0:13:54.739593554] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: AeMeteringMode = 0
[0:13:54.739834856] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: Contrast = 1.000000
[0:13:54.739911002] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: NoiseReductionMode = 1
[0:13:54.740162408] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: ExposureValue = 0.000000
[0:13:54.740270584] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: AwbMode = 0
[0:13:54.740348345] [1341]  INFO IPARPI raspberrypi.cpp:612 Request ctrl: AeExposureMode = 0
[0:13:54.743313495] [1338]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:13:54.743495526] [1338]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:13:54.743582453] [1338]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:13:55.019033479] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (6 left)
[0:13:55.068152081] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (5 left)
[0:13:55.132388570] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (4 left)
[0:13:55.192103193] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (3 left)
[0:13:55.241578305] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (2 left)
[0:13:55.295289607] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (1 left)
[0:13:55.334941300] [1338]  INFO RPI raspberrypi.cpp:1621 Dropping frame at the request of the IPA (0 left)
Segmentation fault
```

davidplowman, 2021-03-18:
Hi, sorry - I didn't see this issue come in.

I've tried this running on a Pi 3 with the imx327 module that I have  (an Innomaker one) as follows:
```./libcamera-vid -o test.h264 --width 1920 --height 1080```
but don't see a problem.

Can you say which application you're running and the exact command line? Is everything up to date? Is there anything unusual about your setup or how you're driving it? Failing that we may need to get a backtrace and see what that tells us.

vincentneo, 2021-03-22:
Hi,

I am using a pi 3+, with the imx327 (innomaker).

The command that I was using was `./libcamera-vid --width 1920 --height 1080`. Having -o or not makes no difference in this issue.

I remembered running `rpi-update` before trying and experiencing this issue. libcamera-apps, libcamera was also newly compiled from source. (as of time of first post)
``` 
uname -a
Linux raspberrypi 5.10.20-v7+ #1404 SMP Thu Mar 4 19:40:23 GMT 2021 armv7l GNU/Linux
```

First run always results in `Segmentation fault`, but second run will result in:
```
./libcamera-vid --width 1920 --height 1080
[0:01:49.201691257] [1355]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2390-e32d33b1
[0:01:49.227770858] [1359]  WARN CameraSensor camera_sensor.cpp:306 'imx290 10-001a': Recommended V4L2 control 0x009a0922 not supported
[0:01:49.227989922] [1359]  WARN CameraSensor camera_sensor.cpp:358 'imx290 10-001a': The sensor kernel driver needs to be fixed
[0:01:49.228051694] [1359]  WARN CameraSensor camera_sensor.cpp:360 'imx290 10-001a': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:01:49.237700873] [1359]  WARN CameraSensor camera_sensor.cpp:472 'imx290 10-001a': Failed to retrieve the camera location, setting to External
[0:01:49.410258019] [1355]  WARN V4L2 v4l2_pixelformat.cpp:180 Unsupported V4L2 pixel format XB24
[0:01:49.410964586] [1355] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:01:49.411726362] [1355] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:01:49.412142667] [1355] ERROR Controls controls.cpp:990 Control 0x00000002 not found
[0:01:49.412485325] [1355]  INFO Camera camera.cpp:890 configuring streams: (0) 1920x1080-YUV420
[0:01:49.413498978] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 1500 (best 1500)
[0:01:49.413698146] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG10 Score: 3500 (best 1500)
[0:01:49.413783772] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG12 Score: 500 (best 500)
[0:01:49.413851585] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG12 Score: 2500 (best 500)
[0:01:49.413923981] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1000 (best 500)
[0:01:49.413991169] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRAA Score: 3000 (best 500)
[0:01:49.414068149] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRCC Score: 0 (best 0)
[0:01:49.414136014] [1359]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRCC Score: 2000 (best 0)
[0:01:49.414240181] [1359]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected mode: 1920x1080-pRCC
[0:01:49.527162188] [1359] ERROR V4L2 v4l2_videodevice.cpp:1133 /dev/video14[17:cap]: Not enough buffers provided by V4L2VideoDevice
[0:01:49.529292410] [1359] ERROR RPI raspberrypi.cpp:788 Failed to allocate buffers
ERROR: *** failed to start camera ***
```

My /boot/config.txt: 
```
dtoverlay=vc4-fkms-v3d,cma=512
max_framebuffers=2
framebuffer_priority=2
dtparam=i2cz_vc=on
dtoverlay=imx290,clock-frequency=74250000
```
GPU Memory set at 380

Thank you.

davidplowman, 2021-03-22:
Hi, thanks for the extra information. I've had a go at reproducing the problem with those parameters but haven't so far been successful, the module - same or very similar to yours by the sound of it - just won't "not work". A couple of suggestions:

- This morning I've been through a completely fresh install with the latest Pi OS and the raspberrypi/libcamera-apps instructions. `uname -a` reports Mar 15, and `vcgencmd version` reports Mar 19. The only changes to the default config.txt are to add the following to the [all] section:
```
dtoverlay=vc4-fkms-v3d
dtoverlay=imx290,clock-frequency=74250000
```
and it runs fine. Are you able to try this experiment and see if it works for you? You'd then be able to make your other alterations one at a time and hopefully spot when it goes wrong.
- Can you get a stack trace for the segmentation fault? The fact that it then fails to allocate buffers after that also seems suspicious, I wonder if the kernel is getting unhappy. It might be worth checking dmesg, perhaps.

(Slightly unrelated: I think your cma syntax may be wrong, it should be `cma-512` not `cma=512`. Having said that, 512MB sounds huge for a platform with only 1GB; it may even be harmful. Also 380MB sounds large for the `gpu_mem`. I realise of course that you may be running other software that needs this!)

vincentneo, 2021-03-22:
Hi, 

Will test out the experiment when free in the coming days.

I set the gpu_mem to 380MB because previously, it was at a lower value that made even the hello app not run at all. I simply randomly chose a memory value, and it just happened to work.

I also tried no cma previously. With or without, same results. 

> Can you get a stack trace for the segmentation fault? The fact that it then fails to allocate buffers after that also seems suspicious, I wonder if the kernel is getting unhappy. It might be worth checking dmesg, perhaps.

How do I get the stack trace? in dmesg?

Thanks for taking the time to look into this!

davidplowman, 2021-03-22:
So to get a user space stack trace you'll have to run under gdb. Enter the following commands:
```
gdb libcamera-vid
run --width 1920 --height 1080
bt
```
(each command will take a while). The final one should print out a backtrace which you can post here.

In dmesg, just look for anything that contains contains lots of hex and looks really scary, especially anything that says "call trace" or something like that. But there may well be nothing, which would be worth knowing in itself.

naushir, 2021-08-13:
Closing this since we have not heard back in a while.

closed 2021-08-13 by naushir