raspberrypi/rpicam-apps #18: Using the RAW video

catchchaos, 2021-03-16:
Hi,

I recently captured a RAW bayer video using libcamera_raw. How can I extract individual frames from the generated video? I've got the HQ camera with the Sony IMX477. Normally, I use ffmpeg to extract frames, but it doesn't support 12-bit bayer formats.

I could simply read the binary file assuming there is no additional information. But surely this is incorrect as there are some offsets at the beggining/end and to indicate that frame transitions?

davidplowman, 2021-03-17:
Hi, libcamera-raw outputs a message saying exactly what the image size, stride (bytes per line) and format are, something like this: 
```Raw stream: <width>x<height> stride <stride> format <format>```

It then writes a completely plain binary dump of each frame into a flat file. If you multiply `<height>` by `<stride>`, this tells you exactly how many bytes per frame you need to read.

Alternatively you can add the arguments `--segment 1` and it will be split into separate frames. You would probably want to put a counter into the output filename, for example:
```./libcamera-raw -o frame%05d.raw --segment 1```

Note that you end up with a plain binary dump of Bayer pixels, and they don't give you much clue as to how to process them. I'm assuming you have something in mind...? Anyway, I hope that helps!

catchchaos, 2021-03-18:
Thanks for your quick and detailed response. Yes, I know that the dump contains Bayer data (BGGR pattern for HQ camera) and intend to use my own processing pipeline. I was unaware of the `<stride>` option. It turns out that `<stride>` is slightly larger than `<width>`. I assume this is to gain additional information such as black level. The extracted frames do have the correct spatial structure now. Thanks again!

I noticed something strange when I attempted simple bilinear demosaicing. The means of the individual channels (using simple array splicing) don't make sense for the extracted video frames. For instance, the means of the green channels, G1 and G2 should be very similar, but this is not the case:

```
Processing video directly: test6.raw
mean B: 549.722455798
mean G1: 1840.36185495
mean G2: 784.806562243
mean R: 1531.83185393

Processing image: test6.dng
mean B: 568.142633396
mean G1: 803.677793782
mean G2: 803.734932005
mean R: 463.768095349```

As you can see, when I extract means in the same manner from a `dng` file, everything makes sense. The raw video and dng were captured for the exact same scene using libcamera.

davidplowman, 2021-03-18:
Just to add that the stride often slightly larger than (for 12-bit data) width*12/8 bytes because the hardware needs to start writing rows of bytes at nice aligned addresses (for example, 32-byte aligned). So you get some padding bytes at the end of each row, but there's nothing useful in them!

davidplowman, 2021-03-29:
This probably seems a reasonable point to close this one. If you have any other questions, please feel free to open another issue or ask on the Raspberry Pi Camera Board forum. Thanks!

closed 2021-03-29 by davidplowman