raspberrypi/rpicam-apps #20: Setting camera attributes when using "libcamera-raw"

oyontem, 2021-03-22:
Hi,
I am working with the HQ camera (imx477) on a Raspberry Pi 4 - 4Gb. I could not see any predefined modes in libcamera as in raspivid: https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

===================================================================================
Mode | Size | Aspect Ratio | Frame rates | FOV | Binning/Scaling
-- | -- | -- | -- | -- | --
0 | automatic selection |   |   |   |  
1 | 2028x1080 | 169:90 | 0.1-50fps | Partial | 2x2 binned
2 | 2028x1520 | 4:3 | 0.1-50fps | Full | 2x2 binned
3 | 4056x3040 | 4:3 | 0.005-10fps | Full | None
4 | 1332x990 | 74:55 | 50.1-120fps | Partial | 2x2 binned
===================================================================================

Therefore, I am trying to set the frame rate and frame size when capturing raw video, according to the command list given by: "libcamera-raw -h".

I typed in the following to record a video similar to "mode 4" in raspivid:
"libcamera-raw -v --height 990 --width 1332 --framerate 120 -o test.raw"

and received the following output:
===================================================================================
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 5000
    width: 1332
    height: 990
    output: test.raw
    rawfull: 0
    preview: none
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 120
    denoise: auto
    bitrate: 0
    profile: 
    level:  
    intra: 0
    inline: 0
    save-pts: 
    codec: h264
    quality (for MJPEG): 50
    keypress: 0
    signal: 0
    initial: record
    split: 0
    segment: 0
    circular: 0
Opened NullEncoder
Running without preview window
Opening camera...
[1:08:40.033858188] [2849]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2400-80bebfb6
[1:08:40.049253314] [2852]  WARN CameraSensor camera_sensor.cpp:306 'imx477 10-001a': Recommended V4L2 control 0x009a0922 not supported
[1:08:40.049516670] [2852]  WARN CameraSensor camera_sensor.cpp:358 'imx477 10-001a': The sensor kernel driver needs to be fixed
[1:08:40.049603832] [2852]  WARN CameraSensor camera_sensor.cpp:360 'imx477 10-001a': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[1:08:40.057389214] [2852]  WARN CameraSensor camera_sensor.cpp:472 'imx477 10-001a': Failed to retrieve the camera location, setting to External
Acquired camera /base/soc/i2c0mux/i2c@1/imx477@1a
Configuring video...
[1:08:40.174037490] [2849]  WARN V4L2 v4l2_pixelformat.cpp:180 Unsupported V4L2 pixel format XB24
[1:08:40.174532408] [2849]  INFO RPI raspberrypi.cpp:122 Mode: 1012x760 fmt RG10 Score: 12155.9 (best 12155.9)
[1:08:40.174669327] [2849]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt RG12 Score: 500 (best 500)
[... 150 lines cut ...]
FileOutput: output buffer 0xb1a2b000 size 4669440
Viewfinder frame 37
(failed to queue request)
FileOutput: output buffer 0xb2787000 size 4669440
Camera stopped!
WARNING: could not make request!
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
===================================================================================

For some reason it still records in the default format: "Raw stream: 2028x1520 stride 3072 format SBGGR12_CSI2P".

I noticed two lines in the above output info:
"[1:08:40.178663633] [2849]  INFO Camera camera.cpp:890 configuring streams: (0) 1332x990-YUV420 (1) 2028x1520-SBGGR12_CSI2P"
and,
"[1:08:40.179735002] [2852]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC"
Is it possible to set the output stream to the ones I specified?

I tried different frame rates and sizes similar to raspivid modes but I still have the same fixed output format. Moreover, I cannot even decode those. Also, it does not record in a full frame raw format when I switch "rawfull" on.

Is there a way to get these working? 

Thank you very much.

davidplowman, 2021-03-22:
Hi, yes - so one of the "problems" we have with libcamera is that it expects the camera mode to be selected automatically based on the output resolution, so you can't "just ask for mode 4" (or whichever). In your case it's preferring 12-bit modes over the one that you want. I don't really have a solution for that right now, except to edit the source code. You could, for example, go to [this ](https://git.linuxtv.org/libcamera.git/tree/src/libcamera/pipeline/raspberrypi/raspberrypi.cpp#n77) place in the libcamera code and swap over the 10-bit and 12-bit "penalties".

As regards the "rawfull" problem, it bombs out for me when I try this with a "concurrent streams need matching numbers of buffers message". I assume that's what you're seeing too? Indeed that's a little bug, I'll push an update shortly.

oyontem, 2021-03-22:
Hi, thank you for your quick reply. I will look into your suggestion. 
(Though, I would still expect that the defined attributes would set the output parameters.)

And yes, it returns that message when I choose the "rawfull". Looking forward to getting the updates.

davidplowman, 2021-03-22:
Ah, my colleague seems to have merged that fix already. That was quick! So just update your libcamera-apps and "rawfull" should work as expected (outside chance you might have to increase your cma memory - the first of the "Tips"). As regards the mode selection, that's an on-going discussion.

oyontem, 2021-03-22:
That's great! Thank you.

davidplowman, 2021-06-30:
There were a couple of problems here, one of which has been fixed, and the other (inability to get 10-bit modes) is covered here https://github.com/raspberrypi/libcamera-apps/issues/26 (rest assured we're still thinking about it). I'll close this one in the meantime, but please open a new issue if anything else comes up.

closed 2021-06-30 by davidplowman