raspberrypi/rpicam-apps #22: This will crash if options.nopreview is true

jhihn, 2021-03-26:
https://github.com/raspberrypi/libcamera-apps/blob/4bb40bb0841b62df953577a41865b70e4732b199/libcamera_app.hpp#L149

jhihn, 2021-03-26:
In fact, there are a few scenarios in the preceding `if` block which will set preview_ to null, which will crash at this line.

davidplowman, 2021-03-26:
Hi, thanks for reporting the issue. Unfortunately I haven't quite understood exactly what the problem is - what have you typed at the command line that causes the libcamera-apps to crash? Thanks!

davidplowman, 2021-04-09:
If there's no further information I'm going to propose closing this issue in the next few days. I haven't spotted how the preview_ variable could be null, and the apps seem to work OK when options.nopreview is true. But if there's anything I've overlooked, please post an update. Thanks.

jhihn, 2021-04-09:
I guess I was wrong. Huh.

closed 2021-04-09 by jhihn