raspberrypi/rpicam-apps #23: Select camera resolution (mode)

adgbu, 2021-04-06:
I want to capture the full sensor resolution of the camera at the highest framerate possible. I am using the IMX477 on Raspberry Pi 4 (Raspbian Buster) with all the latest software as of today.

According to [Documentation > Raspbian > Applications > Camera](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md) the maximum is 4056x3040 @ 10fps for the IMX477 and 3280x2464 @ 15 fps for the IMX219. 

Running libcamera-vid without options works with live preview and it appears as if camera mode 2 is selected (2028x1520, 30 fps, 2x2 binned).

```
$ libcamera-vid -t 0
...
[3:01:20.802820052] [7300]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
...
```

Running libcamera-vid at 1080p setting also works. It appears as if camera mode 1 is selected (2028x1080, 50fps, 2x2 binned, 16:9 crop)

``` 
$ libcamera-vid -t 0 --width 1920 --height 1080 --framerate 50
...
[3:12:51.229683989] [7399]  INFO RPI raspberrypi.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1080-pRCC
...
``` 

Any attempt to run libcamera-vid and manually set resolution to 2028x1520 or higher results in error.

```
$ libcamera-vid -t 0 --width 2028 --height 1520 
ERROR: *** failed to start output streaming ***

$ libcamera-vid -t 0 --width 4056 --height 3040 --framerate 10 
ERROR: *** failed to start output streaming ***
```

davidplowman, 2021-04-09:
Hi, sorry I didn't spot this issue come in sooner!

The h.264 encoder is limited to 1920x1080. Actually you can push it a bit further (as you've discovered) but not much. This is a fundamental hardware limitation.

To encode 12MP you would have to use the mjpeg codec (`--codec mjpeg`, though note there's a bug that needs fixing first, [here](https://github.com/raspberrypi/libcamera-apps/pull/24)). I find I need to overclock a bit to get 12MP @ 10fps and disable the preview (the `-n` option), and you might need some cooling.

adgbu, 2021-04-12:
Ok, thank you! That is really useful information that I didn't know and would probably take forever to find.

I would have guessed that h.264 encoder would just be a tradeoff between resolution and framerate, not a hard limit. Good to know! I will try with mjpeg.

I am working on live WebRTC streaming from camera. So far with MMAL based code. I hope we can make it happen with libcamera soon.

davidplowman, 2021-04-12:
That mjpeg encoder fix has been merged now, so you should be ok if you just update your tree again. I'll close this h.264 issue now - obviously if you come across anything else please open another.

closed 2021-04-12 by davidplowman

KansaiUser, 2024-02-20:
Before closing this issue could you give one example of the solution (using codec mjpeg)?

(I am assuming rpicam-vid --codec mjpeg -t 5000 -o hello.<--- what here?

KansaiUser, 2024-02-20:
Also in what version of raspi-vid is this bug fixed? mine is build 9ae39f85ae6b 30-01-2024
v0.1.0+147-057299d0-dirty

davidplowman, 2024-02-20:
Use something like
```
rpicam-vid --codec mjpeg -t 5000 --width 3840 --height 2160 -o file.mjpeg
```
I don't happen to know exactly when this was fixed, but judging by the dates on this thread it was quite probably _years_ ago! The above certainly works for me on a current image.