raspberrypi/rpicam-apps #25: kernel message while running libcamera-raw

richlegrand, 2021-04-09:
I'm noticing  some kernel messages when running libcamera-raw.  I ran with the latest libcamera and libcamera-apps source, and after performing a full-upgrade on  Pi 4, 4GB.  It's produced once every 5 minutes(ish)) while running this (or simlar):

./libcamera-raw -v -o /dev/null -t 500000000 --framerate 300

kernel message:
Apr  9 15:35:59 rpi kernel: [ 8482.526264] ------------[ cut here ]------------
Apr  9 15:35:59 rpi kernel: [ 8482.526302] WARNING: CPU: 0 PID: 0 at drivers/media/common/videobuf2/videobuf2-core.c:997 vb2_buffer_done+0x22c/0x234 [videobuf2_common]
Apr  9 15:35:59 rpi kernel: [ 8482.526309] Modules linked in: rfcomm fuse cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc imx477 v3d vc4 brcmfmac cec brcmutil gpu_sched sha256_generic drm_kms_helper cfg80211 bcm2835_unicam raspberrypi_hwmon v4l2_dv_timings bcm2835_codec(C) bcm2835_v4l2(C) rfkill bcm2835_isp(C) v4l2_fwnode v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_mux_pinctrl i2c_mux drm videodev i2c_bcm2835 vc_sm_cma(C) mc snd_bcm2835(C) drm_panel_orientation_quirks snd_soc_core rpivid_mem snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
Apr  9 15:35:59 rpi kernel: [ 8482.526667] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        5.10.20-v7l+ #1404
Apr  9 15:35:59 rpi kernel: [ 8482.526673] Hardware name: BCM2711
Apr  9 15:35:59 rpi kernel: [ 8482.526680] Backtrace: 
Apr  9 15:35:59 rpi kernel: [ 8482.526707] [<c0b717b0>] (dump_backtrace) from [<c0b71b44>] (show_stack+0x20/0x24)
Apr  9 15:35:59 rpi kernel: [ 8482.526718]  r7:ffffffff r6:00000000 r5:60000193 r4:c12e6a3c
Apr  9 15:35:59 rpi kernel: [ 8482.526729] [<c0b71b24>] (show_stack) from [<c0b75ed4>] (dump_stack+0xcc/0xf8)
Apr  9 15:35:59 rpi kernel: [ 8482.526742] [<c0b75e08>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
Apr  9 15:35:59 rpi kernel: [ 8482.526753]  r10:00000608 r9:00000009 r8:bf1734cc r7:000003e5 r6:00000009 r5:bf1734cc
Apr  9 15:35:59 rpi kernel: [ 8482.526760]  r4:bf178f90 r3:c1205094
Apr  9 15:35:59 rpi kernel: [ 8482.526772] [<c0220ab0>] (__warn) from [<c0b722b4>] (warn_slowpath_fmt+0x70/0xd8)
Apr  9 15:35:59 rpi kernel: [ 8482.526781]  r7:000003e5 r6:bf178f90 r5:c1205048 r4:00000000
Apr  9 15:35:59 rpi kernel: [ 8482.526806] [<c0b72248>] (warn_slowpath_fmt) from [<bf1734cc>] (vb2_buffer_done+0x22c/0x234 [videobuf2_common])
Apr  9 15:35:59 rpi kernel: [ 8482.526816]  r9:c2db9c10 r8:00000002 r7:00000002 r6:00000005 r5:c2b7b000 r4:c2db99b0
Apr  9 15:35:59 rpi kernel: [ 8482.526847] [<bf1732a0>] (vb2_buffer_done [videobuf2_common]) from [<bf23b9c0>] (unicam_isr+0x124/0x3a4 [bcm2835_unicam])
Apr  9 15:35:59 rpi kernel: [ 8482.526856]  r9:c2db9c10 r8:00000002 r7:00000002 r6:c2db9608 r5:c2db9000 r4:bf2422c0
Apr  9 15:35:59 rpi kernel: [ 8482.526876] [<bf23b89c>] (unicam_isr [bcm2835_unicam]) from [<c0287eb0>] (__handle_irq_event_percpu+0xa8/0x268)
Apr  9 15:35:59 rpi kernel: [ 8482.526886]  r10:c1201e08 r9:ffffe000 r8:00000033 r7:c1a8ca00 r6:00000000 r5:c1a8ca6c
Apr  9 15:35:59 rpi kernel: [ 8482.526892]  r4:c36ea340
Apr  9 15:35:59 rpi kernel: [ 8482.526905] [<c0287e08>] (__handle_irq_event_percpu) from [<c02880ac>] (handle_irq_event_percpu+0x3c/0x90)
Apr  9 15:35:59 rpi kernel: [ 8482.526914]  r10:c0e21a40 r9:c1200000 r8:c1810800 r7:00000001 r6:c1a8ca00 r5:c1a8ca6c
Apr  9 15:35:59 rpi kernel: [ 8482.526921]  r4:c1205048
Apr  9 15:35:59 rpi kernel: [ 8482.526933] [<c0288070>] (handle_irq_event_percpu) from [<c0288154>] (handle_irq_event+0x54/0x78)
Apr  9 15:35:59 rpi kernel: [ 8482.526941]  r6:00000000 r5:c1a8ca6c r4:c1a8ca00
Apr  9 15:35:59 rpi kernel: [ 8482.526953] [<c0288100>] (handle_irq_event) from [<c028cef0>] (handle_fasteoi_irq+0xd4/0x18c)
Apr  9 15:35:59 rpi kernel: [ 8482.526961]  r7:00000001 r6:00000000 r5:c1205b7c r4:c1a8ca00
Apr  9 15:35:59 rpi kernel: [ 8482.526972] [<c028ce1c>] (handle_fasteoi_irq) from [<c0286e6c>] (generic_handle_irq+0x44/0x54)
Apr  9 15:35:59 rpi kernel: [ 8482.526979]  r5:00000000 r4:c1094d10
Apr  9 15:35:59 rpi kernel: [ 8482.526991] [<c0286e28>] (generic_handle_irq) from [<c02875cc>] (__handle_domain_irq+0x6c/0xc4)
Apr  9 15:35:59 rpi kernel: [ 8482.527004] [<c0287560>] (__handle_domain_irq) from [<c020135c>] (gic_handle_irq+0x90/0xa4)
Apr  9 15:35:59 rpi kernel: [ 8482.527013]  r9:c1200000 r8:c1094d1c r7:c1201ec8 r6:f081400c r5:f0814000 r4:c1205b7c
Apr  9 15:35:59 rpi kernel: [ 8482.527023] [<c02012cc>] (gic_handle_irq) from [<c0200abc>] (__irq_svc+0x5c/0x7c)
Apr  9 15:35:59 rpi kernel: [ 8482.527030] Exception stack(0xc1201ec8 to 0xc1201f10)
Apr  9 15:35:59 rpi kernel: [ 8482.527039] 1ec0:                   00000000 017768fc eff122c4 c021ac00 ffffe000 c120509c
Apr  9 15:35:59 rpi kernel: [ 8482.527049] 1ee0: c12050e4 00000001 00000001 c133d00f c0e21a40 c1201f24 c1201f28 c1201f18
Apr  9 15:35:59 rpi kernel: [ 8482.527056] 1f00: c02088c0 c02088c4 60000013 ffffffff
Apr  9 15:35:59 rpi kernel: [ 8482.527066]  r9:c1200000 r8:00000001 r7:c1201efc r6:ffffffff r5:60000013 r4:c02088c4
Apr  9 15:35:59 rpi kernel: [ 8482.527080] [<c020887c>] (arch_cpu_idle) from [<c0b817e0>] (default_idle_call+0x4c/0x118)
Apr  9 15:35:59 rpi kernel: [ 8482.527094] [<c0b81794>] (default_idle_call) from [<c02587d4>] (do_idle+0x118/0x168)
Apr  9 15:35:59 rpi kernel: [ 8482.527105] [<c02586bc>] (do_idle) from [<c0258af8>] (cpu_startup_entry+0x28/0x30)
Apr  9 15:35:59 rpi kernel: [ 8482.527115]  r10:00000162 r9:c1053a60 r8:ffffffff r7:c1053a60 r6:c1205040 r5:c1205048
Apr  9 15:35:59 rpi kernel: [ 8482.527122]  r4:000000d8 r3:c108a294
Apr  9 15:35:59 rpi kernel: [ 8482.527133] [<c0258ad0>] (cpu_startup_entry) from [<c0b7a710>] (rest_init+0xbc/0xc4)
Apr  9 15:35:59 rpi kernel: [ 8482.527148] [<c0b7a654>] (rest_init) from [<c1000ab4>] (arch_call_rest_init+0x18/0x1c)
Apr  9 15:35:59 rpi kernel: [ 8482.527155]  r5:c1205048 r4:c1356068
Apr  9 15:35:59 rpi kernel: [ 8482.527168] [<c1000a9c>] (arch_call_rest_init) from [<c1001098>] (start_kernel+0x568/0x59c)
Apr  9 15:35:59 rpi kernel: [ 8482.527178] [<c1000b30>] (start_kernel) from [<00000000>] (0x0)
Apr  9 15:35:59 rpi kernel: [ 8482.527186] ---[ end trace de822ceb1e416dfd ]---

davidplowman, 2021-04-10:
Hi, thanks for reporting this. To be sure we're running the same setup as you, could you please report the output from `uname -a` and `sudo vcdbg version`. I also deduce from the log above that you're using the imx477 (HQ cam).

If you were able and had the time, it might be helpful to execute (as root) `echo 4 > /sys/modules/videobuf2_common/parameters/debug` and then post a couple of screens-worth of messages from dmesg just prior to one of these warnings. Thanks!

richlegrand, 2021-04-10:
Thanks for the quick response!  See below -- happy to try other things.  

I also notice a frame that is coincident with each kernel message has a timestamp of 0.  (I'm looking at frames coming into LibcameraApp::requestComplete.)

I'm not able to run "echo 4 > /sys/module/videobuf2_common/parameters/debug" -- results in "permission denied".

uname -a:
Linux rpi 5.10.20-v7l+ #1404 SMP Thu Mar 4 19:44:07 GMT 2021 armv7l GNU/Linux

sudo vcdbg version:
vcos_build_version     = 8ed5b5711855476273c45bb0754d9e23903a42e7 (clean)
vcos_build_branch      = bcm2711_2
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_application'
vcos_build_application = ?
vcos_build_date        = Mar  5 2021
vcos_build_time        = 12:56:11
vcos_build_user        = dom
vcos_build_hostname    = buildbot
vcos_build_platform    = raspberrypi_linux
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_type'
vcos_build_type        = ?

dmesg:
[72309.657169] ------------[ cut here ]------------
[72309.657197] WARNING: CPU: 0 PID: 0 at drivers/media/common/videobuf2/videobuf2-core.c:997 vb2_buffer_done+0x22c/0x234 [videobuf2_common]
[72309.657204] Modules linked in: rfcomm fuse cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc imx477 v3d vc4 brcmfmac cec brcmutil gpu_sched sha256_generic drm_kms_helper cfg80211 bcm2835_unicam raspberrypi_hwmon v4l2_dv_timings bcm2835_codec(C) bcm2835_v4l2(C) rfkill bcm2835_isp(C) v4l2_fwnode v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_mux_pinctrl i2c_mux drm videodev i2c_bcm2835 vc_sm_cma(C) mc snd_bcm2835(C) drm_panel_orientation_quirks snd_soc_core rpivid_mem snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
[72309.657588] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        5.10.20-v7l+ #1404
[72309.657595] Hardware name: BCM2711
[72309.657602] Backtrace:
[72309.657623] [<c0b717b0>] (dump_backtrace) from [<c0b71b44>] (show_stack+0x20/0x24)
[72309.657633]  r7:ffffffff r6:00000000 r5:60000193 r4:c12e6a3c
[72309.657646] [<c0b71b24>] (show_stack) from [<c0b75ed4>] (dump_stack+0xcc/0xf8)
[72309.657658] [<c0b75e08>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[72309.657670]  r10:00000608 r9:00000009 r8:bf1734cc r7:000003e5 r6:00000009 r5:bf1734cc
[72309.657678]  r4:bf178f90 r3:c1205094
[72309.657691] [<c0220ab0>] (__warn) from [<c0b722b4>] (warn_slowpath_fmt+0x70/0xd8)
[72309.657701]  r7:000003e5 r6:bf178f90 r5:c1205048 r4:00000000
[72309.657729] [<c0b72248>] (warn_slowpath_fmt) from [<bf1734cc>] (vb2_buffer_done+0x22c/0x234 [videobuf2_common])
[72309.657740]  r9:c2db9c10 r8:00000002 r7:00000002 r6:00000005 r5:c2a84c00 r4:c2db99b0
[72309.657772] [<bf1732a0>] (vb2_buffer_done [videobuf2_common]) from [<bf23b9c0>] (unicam_isr+0x124/0x3a4 [bcm2835_unicam])
[72309.657783]  r9:c2db9c10 r8:00000002 r7:00000002 r6:c2db9608 r5:c2db9000 r4:bf2422c0
[72309.657803] [<bf23b89c>] (unicam_isr [bcm2835_unicam]) from [<c0287eb0>] (__handle_irq_event_percpu+0xa8/0x268)
[72309.657814]  r10:c1201e08 r9:ffffe000 r8:00000033 r7:c1a8ca00 r6:00000000 r5:c1a8ca6c
[72309.657822]  r4:c36ea340
[72309.657836] [<c0287e08>] (__handle_irq_event_percpu) from [<c02880ac>] (handle_irq_event_percpu+0x3c/0x90)
[72309.657847]  r10:c0e21a40 r9:c1200000 r8:c1810800 r7:00000001 r6:c1a8ca00 r5:c1a8ca6c
[72309.657854]  r4:c1205048
[72309.657868] [<c0288070>] (handle_irq_event_percpu) from [<c0288154>] (handle_irq_event+0x54/0x78)
[72309.657878]  r6:00000000 r5:c1a8ca6c r4:c1a8ca00
[72309.657891] [<c0288100>] (handle_irq_event) from [<c028cef0>] (handle_fasteoi_irq+0xd4/0x18c)
[72309.657901]  r7:00000001 r6:00000000 r5:c1205b7c r4:c1a8ca00
[72309.657914] [<c028ce1c>] (handle_fasteoi_irq) from [<c0286e6c>] (generic_handle_irq+0x44/0x54)
[72309.657922]  r5:00000000 r4:c1094d10
[72309.657935] [<c0286e28>] (generic_handle_irq) from [<c02875cc>] (__handle_domain_irq+0x6c/0xc4)
[72309.657949] [<c0287560>] (__handle_domain_irq) from [<c020135c>] (gic_handle_irq+0x90/0xa4)
[72309.657960]  r9:c1200000 r8:c1094d1c r7:c1201ec8 r6:f081400c r5:f0814000 r4:c1205b7c
[72309.657971] [<c02012cc>] (gic_handle_irq) from [<c0200abc>] (__irq_svc+0x5c/0x7c)
[72309.657979] Exception stack(0xc1201ec8 to 0xc1201f10)
[72309.657990] 1ec0:                   00000000 0a56050c eff122c4 c021ac00 ffffe000 c120509c
[72309.658001] 1ee0: c12050e4 00000001 00000001 c133d00f c0e21a40 c1201f24 c1201f28 c1201f18
[72309.658011] 1f00: c02088c0 c02088c4 60000013 ffffffff
[72309.658022]  r9:c1200000 r8:00000001 r7:c1201efc r6:ffffffff r5:60000013 r4:c02088c4
[72309.658038] [<c020887c>] (arch_cpu_idle) from [<c0b817e0>] (default_idle_call+0x4c/0x118)
[72309.658053] [<c0b81794>] (default_idle_call) from [<c02587d4>] (do_idle+0x118/0x168)
[72309.658066] [<c02586bc>] (do_idle) from [<c0258af8>] (cpu_startup_entry+0x28/0x30)
[72309.658077]  r10:00000162 r9:c1053a60 r8:ffffffff r7:c1053a60 r6:c1205040 r5:c1205048
[72309.658085]  r4:000000d8 r3:c108a294
[72309.658098] [<c0258ad0>] (cpu_startup_entry) from [<c0b7a710>] (rest_init+0xbc/0xc4)
[72309.658113] [<c0b7a654>] (rest_init) from [<c1000ab4>] (arch_call_rest_init+0x18/0x1c)
[72309.658122]  r5:c1205048 r4:c1356068
[72309.658136] [<c1000a9c>] (arch_call_rest_init) from [<c1001098>] (start_kernel+0x568/0x59c)
[72309.658148] [<c1000b30>] (start_kernel) from [<00000000>] (0x0)
[72309.658157] ---[ end trace de822ceb1e416e6d ]---

davidplowman, 2021-04-11:
I've gone through the rigmarole of flashing a completely new image, updating it, and installing the libcamera bits. For me, uname -a reports Apr 6 and vcdbg version Mar 26, so a bit newer than yours, though no reason to suppose they should behave differently.

Thus far I'm unable to see any of these messages. Is there anything else different about your system, such as edits to the /boot/config.txt, or anything else running?

As regards the debug parameter, did you switch to root before trying it (it would indeed give "permission denied" if you didn't)? I'm a bit suspicious that we might struggle to catch those warnings as the new debug might now swamp the log, but it's worth a try (particularly as I haven't managed to provoke it myself).

richlegrand, 2021-04-11:
When updating the firmware to the March 26 version, the problem went away.  It was my intention to use the latest firmware, but I failed to run rpi-update when updating my pi the latest packages, etc.  Sorry about that, and thanks for taking a look.  :)

davidplowman, 2021-04-11:
No problem, thanks for the update. I'll close this issue then, please post back if you encounter any more problems.

closed 2021-04-11 by davidplowman