raspberrypi/rpicam-apps #26: 10-bit mode

richlegrand, 2021-04-09:
Hello, 
There doesn't seem to be a way to get libcamera to choose a 10-bit mode.  Uf the framerate were considered when choosing the mode, ie a faster framerate would cause the 10-bit penalty to be reduced (perhaps.)

much thanks

davidplowman, 2021-04-10:
Hi, thanks for raising this. We're aware of this as a problem, and indeed I think some other issues have been filed that relate to it. It's a question that we're discussing with the libcamera team, but unfortunately we don't have a plan for how to resolve it just yet. For now, the workaround would be to change the "penalty" values - assume you've found where those are defined. Sorry not to have a better answer at the moment.

dustinkerstein, 2021-11-21:
My personal opinion is that there really isn't a way around just letting the user have direct access to a mode override. There are just too many edge cases for the penalty system to work perfectly. Ie. What about DOL-HDR modes, custom modes that have the same resolution but different FPS output, packed vs. unpacked, etc? 

Ex. I'm doing some testing with libcamera-still and it's picking the 4056x3040-pRCC 12-bit Bayer BGBG/GRGR Packed mode rather than unpacked (what I'd prefer). And I'm seeing some artifacts (black fringing on highlights with aliasing) that I feel could be caused by the packing. While this could certainly be some other ISP issue, being able to discretely select modes will be important for folks who need tight control over the entire capture pipeline.

[image]

naushir, 2021-11-29:
> . And I'm seeing some artifacts (black fringing on highlights with aliasing) that I feel could be caused by the packing.

Packing/unpacking modes would not cause the fringing that you are seeing here.  The pixel data is exactly the same, just that we either represent data as 12-bits directly or 12-bits in a 16-bit word.

As to the cause of the fringing... It may be aggressively sharpening in this scene? Try reducing the sharpening strength a bit (details can be found [here](https://github.com/raspberrypi/documentation/tree/master/linux/software/libcamera/rpi_SOFT_libcamera_1p0.pdf) and [here](https://github.com/raspberrypi/documentation/tree/master/linux/software/libcamera/rpi_SOFT_libcamera_1p0.pdf).

dustinkerstein, 2021-11-29:
> Packing/unpacking modes would not cause the fringing that you are seeing here. The pixel data is exactly the same, just that we either represent data as 12-bits directly or 12-bits in a 16-bit word.

Ok, cool. I'm curious though (sorry for the tangent) - What is the reason to have these packed modes? Does the ISP prefer it?

> As to the cause of the fringing... It may be aggressively sharpening in this scene? Try reducing the sharpening strength a bit (details can be found [...]

Yep, that's what it was. I responded in more detail in the other [issue](https://github.com/raspberrypi/linux/issues/4737).

But back on topic (and apologies for hijacking this thread) - If you're ever looking to expand the discussion about libcamera's design regarding the direct vs. inferred mode access, please let me know. I have plenty of thoughts on the subject :) Boiled down, I think you'll probably need to accommodate both - inferred mode by default (for ease of use), but also provide a way to override this with some user flag that allows the user to directly specify the mode.  Ie. I just tried to capture at 800x600 with the IMX477 (4x3 aspect ratio sensor) and it chose the 2x2 binned 2028x1520 mode, which is not what I would ideally want. I would prefer the full 4056x3040 mode scaled to 800x600 in the ISP. 

Thanks again for the help.

naushir, 2021-11-30:
> Ok, cool. I'm curious though (sorry for the tangent) - What is the reason to have these packed modes? Does the ISP prefer it?

It makes no difference to the ISP, but if the application were to want to peek at the RAW frame buffer, it is much more convenient to have the pixel samples unpacked to 16-bits instead of having to manipulate the packed bitstream.
 
> But back on topic (and apologies for hijacking this thread) - If you're ever looking to expand the discussion about libcamera's design regarding the direct vs. inferred mode access, please let me [...]

The ability to choose sensor modes is in our plans, and we will add support for it in our libcamera-apps in the near future.

dustinkerstein, 2021-11-30:
> It makes no difference to the ISP, but if the application were to want to peek at the RAW frame buffer, it is much more convenient to have the pixel samples unpacked to 16-bits instead of having to manipulate the packed bitstream.

Makes sense. Thanks for the explanation.

> The ability to choose sensor modes is in our plans, and we will add support for it in our libcamera-apps in the near future.

Awesome!

davidplowman, 2021-12-02:
We've finally fixed this here: https://github.com/raspberrypi/libcamera-apps/pull/167

To use it folks would have to build the latest libcamera and libcamera-apps for themselves (currently also requiring an up-to-date kernel using `sudo rpi-update`; caveats apply), alternatively we'll be rolling the feature out shortly as things get updated.

To control the camera mode that gets selected use the `--mode W:H:D` option where W and H are the width and height of the camera mode you wish to use and D is the (bit) depth. libcamera will still try to pick the "closest mode" (in some sense), but if you know what values are available then you can get what you want.

For example, if you want the 10-bit 1332x990 HQ cam mode for a video recording, you could use

```
libcamera-vid -o test.h264 --mode 1332:990:10
```

The `--mode` option works for video and still capture. For preview, `--viewfinder-mode` is equivalent.

There's actually an additional "packing" parameter that determines whether the raw Bayer pixels are packed in memory (occupying only 10 or 12 bits each) or unpacked to 16-bits each (therefore with some wasted memory). So you could use `1332:990:10:U` or `1332:992:10:P`. The default is `P` (packed), though this particular detail is unlikely to be of great interest except to those who wish to write code to manipulate raw frame buffers directly.

Finally, we have quite a few documentation updates pending so we'll get round to working through those, including this feature, and we'll list all the available modes for the official Raspberry Pi cameras.

dustinkerstein, 2021-12-02:
Nice! Looking forward to trying it out. Curious though, how will this affect the behavior described in https://github.com/raspberrypi/libcamera-apps/issues/26#issuecomment-982090956 - Since it's independent of a bit depth, it seems like it might still pick the 2x2 binned mode, right?

davidplowman, 2021-12-02:
No, I think it will be fine. Use `--mode 4056:3040` and you'll get the full resolution (12-bit) mode regardless of the output image size.

closed 2021-12-02 by davidplowman

dustinkerstein, 2021-12-02:
> No, I think it will be fine. Use `--mode 4056:3040` and you'll get the full resolution (12-bit) mode regardless of the output image size.

Ah, yes! It's separate from the --width and --height parameters. Will give it a try now.

dustinkerstein, 2021-12-02:
Just gave it a quick test with a still capture. It looks like it runs its 3A algorithms on the original mode selection, and then switches to the user overridden mode for actual capture. Is that intended? I'll have to think about any ramifications that may have. Here's the short log from this command: `libcamera-still --width 800 --height 600 --mode 4056:3040:12 -o ~/mode.jpg`
```
Preview window unavailable
[0:13:02.677066542] [4944]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3303-b7b72027
[0:13:02.679333024] [4946]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:13:02.679369561] [4946]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:13:02.693031640] [4946]  INFO RPI raspberrypi.cpp:1200 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media0
[0:13:02.693602048] [4944]  INFO Camera camera.cpp:945 configuring streams: (0) 2026x1520-YUV420
[0:13:02.693701881] [4946]  INFO RPI raspberrypi.cpp:164 Format: 1332x990 fmt SBGGR10 Score: 3452.71 (best 3452.71)
[0:13:02.693764215] [4946]  INFO RPI raspberrypi.cpp:164 Format: 2028x1080 fmt SBGGR12 Score: 1084.83 (best 1084.83)
[0:13:02.693784863] [4946]  INFO RPI raspberrypi.cpp:164 Format: 2028x1520 fmt SBGGR12 Score: 0.993421 (best 0.993421)
[0:13:02.693805344] [4946]  INFO RPI raspberrypi.cpp:164 Format: 4056x3040 fmt SBGGR12 Score: 887.993 (best 0.993421)
[0:13:02.693963437] [4946]  INFO RPI raspberrypi.cpp:681 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:13:02.713842815] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:13:02.713930722] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:13:02.713952407] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:13:02.713970130] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:13:02.713989037] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:13:02.714006667] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:13:02.714045685] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:13:02.714064278] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:13:02.714089815] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:13:02.714107815] [4950]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:13:02.714579000] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:02.714605611] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:02.714620204] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:02.714634482] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:02.714648685] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:02.714662871] [4946]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:13:03.246936870] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (6 left)
[0:13:03.280134452] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (5 left)
[0:13:03.313418959] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (4 left)
[0:13:03.346506152] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (3 left)
[0:13:03.379805696] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (2 left)
[0:13:03.413671444] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (1 left)
[0:13:03.446648489] [4946]  INFO RPI raspberrypi.cpp:1738 Dropping frame at the request of the IPA (0 left)
[0:13:08.259223722] [4944]  INFO RPI raspberrypi.cpp:164 Format: 1332x990 fmt SBGGR10 Score: 10552.2 (best 10552.2)
[0:13:08.260304851] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1080 fmt SBGGR12 Score: 8179.84 (best 8179.84)
[0:13:08.260990887] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1520 fmt SBGGR12 Score: 7096 (best 7096)
[0:13:08.261654017] [4944]  INFO RPI raspberrypi.cpp:164 Format: 4056x3040 fmt SBGGR12 Score: 0 (best 0)
[0:13:08.262367479] [4944]  INFO RPI raspberrypi.cpp:164 Format: 1332x990 fmt SBGGR10 Score: 10552.2 (best 10552.2)
[0:13:08.263029220] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1080 fmt SBGGR12 Score: 8179.84 (best 8179.84)
[0:13:08.263401349] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1520 fmt SBGGR12 Score: 7096 (best 7096)
[0:13:08.263763164] [4944]  INFO RPI raspberrypi.cpp:164 Format: 4056x3040 fmt SBGGR12 Score: 0 (best 0)
[0:13:08.264526330] [4944]  INFO RPI raspberrypi.cpp:164 Format: 1332x990 fmt SBGGR10 Score: 10552.2 (best 10552.2)
[0:13:08.265330348] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1080 fmt SBGGR12 Score: 8179.84 (best 8179.84)
[0:13:08.265461330] [4944]  INFO RPI raspberrypi.cpp:164 Format: 2028x1520 fmt SBGGR12 Score: 7096 (best 7096)
[0:13:08.265571644] [4944]  INFO RPI raspberrypi.cpp:164 Format: 4056x3040 fmt SBGGR12 Score: 0 (best 0)
Stream configuration adjusted
[... cut ...]
[0:13:08.336342849] [4974]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:13:08.336716849] [4974]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:13:08.337088552] [4974]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[0:13:08.337467219] [4974]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:13:08.337832515] [4974]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 2
Still capture image received
```
This does output an 800x600 image.

davidplowman, 2021-12-02:
Yes I think that's expected, if you want the preview to run in the full res mode too you'd have to add `--viewfinder-mode 4056:3040:12` (actually you only need `4056:3040`, the bit depth will default to 12).

dustinkerstein, 2021-12-02:
Ah ok. But I'd be curious how this affects the 3A algorithms such as auto-exposure as the 2x2 binning mode will have higher exposure (I believe but need to confirm) than the regular non 2x2 binned modes. If that's true, it would throw off the 3A exposure calcs resulting in an underexposed image, right?

davidplowman, 2021-12-02:
Sensors can behave a bit differently here. Some sensors do exactly what you describe, though most of them (including all the official Raspberry Pi ones) keep the output signal level the same and give you better SNR. Our AEC/AGC algorithm actually deals with this correctly, each different mode of a sensor can have an associated "sensitivity", and when you switch between sensor modes it makes the various necessary adjustments. If you want the full gory details, see here: https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/controller/rpi/agc.cpp#n308

dustinkerstein, 2021-12-02:
Very cool. Thanks for the background detail. Just to play devil's advocate (and let me know if this is getting annoying), what about the following scenario?
```
libcamera-vid --width 1920 --height 1080 --mode 4056:3040:12 -o ~/mode.mp4
```
My intent with those parameters would be to use the full sensor FOV, but have the ISP resize to the max resolution the h264 encoder can handle (with the mp4 source/display aspect ratio metadata fields set). The issue that I see possibly happening, albeit relatively minor, is that the 3A algorithms might get performed on a cropped FOV (assuming there was a 16x9 mode that matched). And if that happens, the algorithms could misjudge the exposure / white balance due to the different crops / FOV. A bit of an edge case, but what do you think? Is my logic correct?

naushir, 2021-12-03:
It is possible that if you use a heavily cropped mode in preview/viewfinder, then it would not reflect the full FoV in the statistics for the 3A.  However, the effect would be marginal at best, given that our sensor crops are pretty modest.

dustinkerstein, 2021-12-03:
> It is possible that if you use a heavily cropped mode in preview/viewfinder, then it would not reflect the full FoV in the statistics for the 3A. However, the effect would be marginal at best, given that our sensor crops are pretty modest.

Kk, sounds reasonable to me.

Bra1nsen, 2022-03-10:
How can I enable the DOL-HDR mode (RP HQ Cam - IMX477)
[image]

thank you

naushir, 2022-03-10:
Sorry, this mode is not supported in either the sensor driver, nor in libcamera.

Bra1nsen, 2022-03-10:
what do you mean with sensor driver?

Bra1nsen, 2022-03-10:
the camera sensor is mounted on a board which does not support the dol-hdr mode?

naushir, 2022-03-10:
That's correct.  There is no functionality in the driver to enable this mode in the hardware at all.  We will be happy to accept pull requests if any members of the public would like to add this features.

However, even if this was available in the sensor driver, it still requires considerable effort to implement in the libcamera stack for the software applications to use.

Bra1nsen, 2022-03-10:
thanks naushir!

what is with the [IMX464](https://www.sony-semicon.co.jp/products/common/pdf/IMX464LQR_LQR1_Flyer.pdf) or [IMX334](https://www.sony-semicon.co.jp/products/common/pdf/IMX334LLR_LQR_Flyer.pdf) with multi exposure HDR mode

is that supported? 

Or better said, how do I find out, if something is supported or not.

6by9, 2022-03-10:
Supported sensors will have a cam_helper file in https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi, and a JSON tuning file in https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/data.

There will also be an associated Linux kernel driver in https://github.com/raspberrypi/linux/tree/rpi-5.15.y/drivers/media/i2c, and a dtoverlay in https://github.com/raspberrypi/linux/tree/rpi-5.15.y/arch/arm/boot/dts/overlays

I've not seen a Linux kernel driver for either IMX464 or IMX334. If you have a vendor of those modules, then ask them for support.

Bra1nsen, 2022-03-10:
that's so enlightening. thanks man!

But what about [this sensor ](https://www.waveshare.com/imx378-190-12.3mp-camera.htm)(IMX387), declared especially for Raspberry Pi, but no support?

6by9, 2022-03-10:
> But what about [this sensor ](https://www.waveshare.com/imx378-190-12.3mp-camera.htm)(IMX387), declared especially for Raspberry Pi, but no support?

Typo - IMX378, not IMX387.

IMX378 is a sibling to IMX477 and uses the same driver.

Bra1nsen, 2022-03-10:
In order to use the new[ IMX678 ](https://www.framos.com/de/produkte/fsm-imx678c-04g-v1a-26024)with all HDR functionalities with raspberry pi, the necessary drivers must be installed. Do you usually get them from the manufacturer? And if not, who does that? Or where do you find someone you can pay to do it?

6by9, 2022-03-10:
The only sensors that are fully supported by Raspberry Pi are the OV5647, IMX219, and IMX477.

All other sensors that have drivers and overlays in our kernel tree for are for modules made by 3rd parties and for which the drivers are either supported by the 3rd party (eg IMX519 and IRS1125), are in the mainline Linux kernel, or have been put together by Raspberry Pi as basic drivers.

If you buy any other sensor from a 3rd party, then you should expect that vendor to also supply drivers.

I've already pointed you at the directories that have the files for the supported sensors. If a sensor is not mentioned in those directories then there is no support for them.