raspberrypi/rpicam-apps #27: Lowmem leak

richlegrand, 2021-04-12:
Hello, 
There seems to be a slow lowmem memory leak while running libcamera-raw.  

Run:
 ./libcamera-raw -v -o /dev/null -t 5000000000 --framerate 300

Testing mem usage:
pi@raspberrypi:~ $ cat /proc/meminfo | grep -i free ;date
MemFree:         3465688 kB
HighFree:        3029572 kB
LowFree:          436116 kB
SwapFree:         102396 kB
CmaFree:          215248 kB
Mon 12 Apr 2021 03:43:18 PM CDT
pi@raspberrypi:~ $ cat /proc/meminfo | grep -i free ;date
MemFree:         3461032 kB
HighFree:        3029572 kB
LowFree:          431460 kB
SwapFree:         102396 kB
CmaFree:          215248 kB
Mon 12 Apr 2021 03:54:43 PM CDT

Notice lowmem (LowFree) decreasing at around 400k/minute.  (Just for illustration, the rate is constant as long as libcamera-raw is running.)  

It's not a huge leak, but it seems to be proportional to framerate, I'm seeing an oom condition after about 8 or so hours of running my code -- libcamera-linked, 90fps.  

This may not be the correct place to report --- happy to report elsewhere, run other tests, just give me the word.  

thanks!   

System info:
Camera: HQ cam
pi@raspberrypi:~/libcamera-apps/build $ uname -a
Linux raspberrypi 5.10.27-v7l+ #1409 SMP Tue Apr 6 18:26:02 BST 2021 armv7l GNU/Linux
pi@raspberrypi:~/libcamera-apps/build $
vcos_build_version     = edf2e9c318863999c97c50cdb74eee235ede3af5 (clean)
vcos_build_branch      = bcm2711_2
vcos_build_application = ?
vcos_build_date        = Mar 26 2021
vcos_build_time        = 16:34:17
vcos_build_user        = dom
vcos_build_hostname    = buildbot
vcos_build_platform    = raspberrypi_linux
vcos_build_type        = ?

davidplowman, 2021-04-13:
Hi, thanks for pointing this out. Actually we fixed this one some time ago on our Linux 5.4 branch but failed to port the patch when we moved to 5.10. This has actually just happened now (https://github.com/raspberrypi/linux/commit/41aaea1d7cb9548646858f8ff8e65ad53b27c66c), so it will appear via rpi-update "in a few days". Sorry about that!

richlegrand, 2021-04-13:
Ah, that's excellent news!  thanks :)

richlegrand, 2021-04-19:
Tested with the latest firmware and memory leak appears to be fixed :)  

thanks!

vcos_build_version     = 0227a0840ee8c94294cf3b2c4b2d42031a58f05a (clean)
vcos_build_branch      = bcm2711_2
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_application'
vcos_build_application = ?
vcos_build_date        = Apr 12 2021
vcos_build_time        = 15:53:21
vcos_build_user        = dom
vcos_build_hostname    = buildbot
vcos_build_platform    = raspberrypi_linux
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_type'
vcos_build_type        = ?

davidplowman, 2021-04-19:
Yes, I noticed that the fix actually became available earlier today so many thanks for the confirmation!

closed 2021-04-19 by davidplowman