raspberrypi/rpicam-apps #30: Preview resolution settings

glddiv, 2021-04-30:
Recently we developed a custom camera module. We encountered a problem. When using libcamera-still, there is some difference between the field of view previewed and the field of view of the final stored image (the default is the maximum resolution). By looking at [the code,](https://github.com/raspberrypi/libcamera-apps/blob/326fdfd120a0f62b7e76f4ee8aa0c82c2bf008e6/libcamera_app.hpp#L189) we found that the current preview resolution is set to 1280x960, but our camera cannot use the binning mode to obtain the maximum field of view at this resolution. Is there any way to solve this problem?Maybe we can manually set the preview resolution?

davidplowman, 2021-04-30:
Hi, thanks for the questions.

You can certainly change the 1280x960 preview resolution to something that works better for you, there should be no problem in doing that. I've sometimes wondering about changing these values, perhaps to half the full sensor width and height so that one would often pick up the 2x2 binning mode, but there's still no absolute guarantee that you would get what you want. Another idea might be to add options so that you can choose on the command line. Would that be helpful?

The slightly longer answer to this is that libcamera doesn't give you terribly good control currently over which mode gets chosen (for example, would the application want a faster framerate, or a greater bit-depth, or a larger field of view?) and this is an on-going discussion!

glddiv, 2021-05-07:
> I've sometimes wondering about changing these values, perhaps to half the full sensor width and height so that one would often pick up the 2x2 binning mode, but there's still no absolute guarantee [...]

We have the same idea. At present, these two methods are effective for us, but specified by command line parameters seems to be more general. Maybe it is possible to combine the two methods?

glddiv, 2021-05-26:
It worked very well for me, thank you very much!

closed 2021-05-26 by glddiv