raspberrypi/rpicam-apps #31: Failed to excute  libcamera-apps

LynnL4, 2021-04-30:
Hi,

HW: Pi4  1GB
OS:  Linux iraspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

When I execute libcamera-hello, I get the following error, but  qcam is working fine

```
terminate called after throwing an instance of 'std::runtime_error' 
   what():  failed to import fd 18 Aborted
```
I located here, 
https://github.com/raspberrypi/libcamera-apps/blob/326fdfd120a0f62b7e76f4ee8aa0c82c2bf008e6/egl_preview.cpp#L317

eglCreateImageKHR seems to be the API provided by
https://github.com/anholt/libepoxy

I'm using the libepoxy from the master branch.
Which version are you using, I can recompile it & try again，

Thanks!

davidplowman, 2021-04-30:
Hi, thanks for the question. I'm just using the default libepoxy branch, which I assume is master, so it's not clear to me why you should get this. I've tried restricting my Pi4 to 1GB and that doesn't seem to cause any problems either. Could you perhaps paste in your /boot/config.txt just to check if there's anything untoward in there? Thanks.

There's always the option of "updating your software to the most bleeding edge version"  (use "sudo rpi-update"), but at this point I can't see any reason why that would make a difference.

GregTheDev, 2021-05-12:
Hi, I'm having a similar problem. I've built the libcamera apps on a Pi Zero W and when I try and execute any of them I get an "Illegal instruction" error. I've followed the instructions in the readme and I'm on the bleeding edge of everything.

I'm a bit stumped by this one, it seems like the application is executing an invalid cpu instruction, but all of the code been compiled on the device that it's executing on? The libcamera "cam" app runs without problems.

My boot/config looks like this:

[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d
gpu_mem=128
dtparam=i2c_vc=on
dtoverlay=imx290,clock-frequency=74250000

start_x=1

davidplowman, 2021-05-12:
Hi, I'm afraid we've only ever run libcamera ourselves on a Pi 3 or 4, my apologies for not making this clearer on the instructions page. libcamera moves a lot of the processing load formerly on the Broadcom GPU onto the ARM cores, so it helps if they are reasonably capable. It may be that it optimises by default for a more powerful type of ARM processor than you have, so you could try deleting and remaking the build directory, and running cmake like this:
`cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..`
Unfortunately, I would expect that you may encounter further problems down the line.

GregTheDev, 2021-05-12:
Thanks, that worked. 

Luckily I'm using the Zero just as an interface to get to the raw data from the sensor (still picture, not video), so for now it looks like I have all I need.

davidplowman, 2021-07-08:
I'm closing this issue because I believe the original problem is duplicated here (https://github.com/raspberrypi/libcamera-apps/issues/46), and in this one we now have unrelated Pi Zero discussions. Please refer to the aforementioned issue for any updates. If the original issue **wasn't** the same (caused by the use of X forwarding), please feel free to re-open a new issue. Thanks!

closed 2021-07-08 by davidplowman