raspberrypi/rpicam-apps #39: Libcam doesn't recognize NoIR camera V2.1

LeonardoLH, 2021-06-11:
Hi, today I tried to install libcam to use in place of raspistill but it didn't recognize the camera. I followed exactly this tutorial (https://www.raspberrypi.org/documentation/linux/software/libcamera/README.md) but when I got to the part of checking the operation (build/src/qcam/qcam) it returned an error. I restarted raspberry and tried again, this time it didn't return an error, but a window appeared to fill in the camera name only.

I proceeded anyway to install libcamera-apps and trying to run the command ./libcamera-hello returned the error "no cameras avaliable".

I use a raspberry Pi 4 with 2GB of RAM

davidplowman, 2021-06-15:
Hi, thanks for your message. libcamera is in fact completely unable to tell the difference between a NoIR and a regular v2.1 camera, so the problem you have would be general to all these modules. So anyway, we'll need to check the usual "camera not found" causes. Could you perhaps:

- Confirm what software versions you have (`uname -a` and `vcgencmd version`).
- Confirm that you've got the correct overlays in your /boot/config.txt (`dtoverlay=imx219` and `dtoverlay=vc4-fkms-v3d`).
- Confirm that you have an official Raspberry Pi camera module, not a clone.
- Confirm that if you remove the `dtoverlay=imx219` line (and reboot), `raspistill` still shows a camera image. If it doesn't, can you report the output of `sudo vcgencmd get_camera`.

Thanks!

LeonardoLH, 2021-06-15:
I decided to recreate the step-by-step guide and annotate the outputs that can help in this problem

`sudo apt update
sudo apt full-upgrade`
All ok

`sudo rpi-update`
All ok

`dtoverlay=imx219`
All ok

`dtoverlay=vc4-fkms-v3d`
All ok, this line already existis on my [config.txt file.](https://pastebin.com/aV6bvyyr)

`sudo apt install libboost-dev
sudo apt install libgnutls28-dev openssl libtiff5-dev
sudo apt install qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
sudo apt install meson
sudo pip3 install pyyaml ply`
All ok

`sudo pip3 install --upgrade meson`
All ok

`git clone git://linuxtv.org/libcamera.git
cd libcamera
meson build
cd build
meson configure -Dpipelines=raspberrypi -Dtest=false
cd ..
ninja -C build
sudo ninja -C build install`
The installation process ocurred with no errors, but when I tested the program with build/src/qcam/qcam its returned an error. ([Link to the error](https://pastebin.com/Nv3cNYEj))([Link to meson output](https://pastebin.com/mADdk5eu))([Link to Ninja output](https://pastebin.com/qLd9m4A5))

`cd
sudo apt install libegl1-mesa-dev
git clone https://github.com/anholt/libepoxy.git
cd libepoxy
mkdir _build
cd _build
meson
ninja
sudo ninja install`
All ok, all meson output cheks is "YES"

`cd
sudo apt install cmake libboost-program-options-dev libdrm-dev libexif-dev
git clone https://github.com/raspberrypi/libcamera-apps.git
cd libcamera-apps
mkdir build
cd build
cmake ..
make -j4`
All ok

When I use ./libcamera-hello, this returns [this error.](https://pastebin.com/5y7whFCn)

LeonardoLH, 2021-06-15:
> Hi, thanks for your message. libcamera is in fact completely unable to tell the difference between a NoIR and a regular v2.1 camera, so the problem you have would be general to all these modules. So [...]

`uname -a`
Linux raspberrypi 5.4.73v64 #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020 aarch64 GNU/Linux

`vcgencmd version`
Jun  8 2021 13:07:19 
Copyright (c) 2012 Broadcom
version 03df0e450c356822f03454c27edea347fd8d211b (clean) (release) (start_x)

> Confirm that you've got the correct overlays in your /boot/config.txt (dtoverlay=imx219 and dtoverlay=vc4-fkms-v3d).
Yes, I put my config.txt file [on this link](https://pastebin.com/aV6bvyyr) if you want to check

>Confirm that you have an official Raspberry Pi camera module, not a clone.
I bought the camera from a reputed seller, and all the informations and contents on the module body, manual and packing are the same as the original

>Confirm that if you remove the dtoverlay=imx219 line (and reboot), raspistill still shows a camera image. If it doesn't, can you >report the output of sudo vcgencmd get_camera.
Yes, the raspistill works properly.

naushir, 2021-06-16:
```
> [0:41:19.885200954] [5183] ERROR IPARPI raspberrypi.cpp:552 Unable to find ISP control 0x009819e9
> [0:41:19.885524837] [5183] ERROR IPARPI raspberrypi.cpp:349 ISP control validation failed.
> [0:41:19.885707649] [5183] ERROR RPI raspberrypi.cpp:1290 IPA configuration failed!
> [0:41:19.885997645] [5183] ERROR RPI raspberrypi.cpp:755 Failed to configure the IPA: -32
```

From your above log snippet, it looks like you might be running an older kernel where the ISP driver slightly out of date.

> 
> uname -a
> Linux raspberrypi 5.4.73v64 #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020 aarch64 GNU/Linux

Yes, you really want to be running the 5.10 kernel.  Instructions to do this are [here](https://www.raspberrypi.org/documentation/linux/kernel/updating.md)

LeonardoLH, 2021-06-17:
> ```
> ```
>
> From your above log snippet, it looks like you might be running an older kernel where the ISP driver slightly out of date.
>
>
> Yes, you really want to be running the 5.10 kernel. Instructions [...]

I tried to update my raspberry pi kernel using `rpi-update` instead of `sudo apt dist-upgrade`, but the kernel stills the 5.4. When I tried to update a message appeared saying I'm using a custom kernel (I'm using only the berryboot for boot manager, but the ISO are original from Raspberry), maybe that's blocking the kernel update?

naushir, 2021-06-17:
I can't say for sure why your kernel update does not work.
Perhaps start with a fresh Raspberry PI OS image and see if you have any issues?

LeonardoLH, 2021-06-22:
Trying to upgrade to kernel 5.10 with berryboot was unsuccessful, but when using the system directly on the SD Card without berryboot I was able to upgrade the kernel and use libcamera without any problems. I can't say if the problem was the size of the boot partition or some other configuration that was preventing the kernel from updating, but in the end it worked.

closed 2021-06-22 by LeonardoLH