raspberrypi/rpicam-apps #41: ERROR: *** no cameras available ***

rlaferla, 2021-06-21:
I'm trying to get libcamera to work on Linux raspberrypi 5.10.44-v8+ #1426 SMP PREEMPT Thu Jun 17 20:20:57 BST 2021 aarch64 GNU/Linux.  However, it can't find any cameras.  I am new to this so perhaps the cable is connected incorrectly or there's an issue with my config.  I added dtoverlay=imx219 to the /boot/config.txt

`pi@raspberrypi:~/libcamera-apps/build $ ./libcamera-hello -v
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 5000
    width: 0
    height: 0
    output:
    rawfull: 0
    preview: default
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
Preview window unavailable
Running without preview window
Opening camera...
[0:08:21.655736517] [821]  INFO Camera camera_manager.cpp:294 libcamera v0.0.0+2687-969da318
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
ERROR: *** no cameras available ***`

davidplowman, 2021-06-22:
Hi, sounds like we need to go through the usual camera-not-found checks! Could you:

- Confirm the firmware version (`sudo vcgencmd version`). Thanks for already including the Linux version.
- Say what kind of Pi you have. I assume it's a Pi 4 of some sort judging by the OS version.
- I notice it's failing to create a preview window - are you running headless? (The camera should still work, though)
- Yes we should double-check the cable connections. On the camera board the blue side of the ribbon cable points backwards, opposite to the camera module. On the Pi the blue side should point towards the USB ports.
- Can you confirm it's an official Raspberry Pi camera module? (clone ones won't work)
- Normally I'd ask you to comment out the `dtoverlay=imx219`, reboot, and try `raspistill`, only that won't work with a 64-bit OS. If you had a 32-bit OS to hand (perhaps on another SD card) it would be worth trying.
- Is this a clean OS install or is there anything else running that might be interfering? Touchscreens are also know to stop the camera from working (to fix that add `disable_touchscreen=1` in your /boot/config.txt
- After booting you could enter `dmesg | grep -i imx219` to see if the kernel driver is failing to load.

Thanks!

naushir, 2021-08-13:
Closing due to lack of activity.

closed 2021-08-13 by naushir

samikha, 2021-08-31:
I get the same error on Ubuntu 20.10 64-bit with kernel 5.8.0-1032-raspi
The version firmware reported by `vcgencmd` is `564e5f9b852b23a330b1764bcf0b2d022a20afd0`
Has anyone managed to get libcamera-apps to run on Ubuntu 20.10 or 21.04 ?

naushir, 2021-08-31:
The Ubuntu distribution is entirely maintained by Canocial. You should raise this issue through their support channels.