raspberrypi/rpicam-apps #46: Error: failed to import fd 18 Aborted

BrendanHenryGithub, 2021-07-04:
Env: 
- 4B (with 4G memory), 64bit Pi OS
- uanme -a:   
[image]
- /boot/config.txt:
[image]

Problem:
[image]

Remarks:
- I try rerun cmake with this in this project build dir:  cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..      (but have the same problem.)

davidplowman, 2021-07-04:
Hi, thanks for this report. Indeed I've seen this error reported before (on a 1GB Pi 4 running a 32-bit OS) but the thread rather dried up. But I'd like to get to the bottom of this.

Firstly, I've been running a 64-bit build with exactly the same data (via `uname -a`) as yours, and I've copied your /boot/config.txt verbatim, and (you've guessed it) it works fine. So:

1. First let's double-check the firmware version (type `vcgencmd version`). I have:
May 27 2021 14:03:06 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)
2. Have you installed anything extra in your build, and are you running this straight after booting up with nothing else going on?
3. Are you running headless, or logged in via ssh, or anything like that? I'm assuming you're logged directly in the Pi with a local screen attached. (BTW, what resolution is the screen? I think `tvservice -s` would tell you)
4. The failure is clearly [here](https://github.com/raspberrypi/libcamera-apps/blob/main/egl_preview.cpp#L318). Would you be able to fish out the actual EGL error? So you'd have to add something like `+ " " + std::to_string(eglGetError())` to the end of that error runtime error message, and re-run make.
5. In the meantime I'll consult with some folks who know rather about all this, perhaps there's some more debug we can enable or look at. (I don't suppose `dmesg` reports anything that looks related?)

Thanks!

BrendanHenryGithub, 2021-07-04:
Thanks for your help!
1.  My vcgencmd version is the same as yours.
2.  First time it prompt: can not find libGLESv2.so,  and i google and run : apt install libgles2-mesa-dev libglfw3-dev.
3.  I  logged in via ssh (Mobaxterm tool with X server).  tverservice output:
[image]
4.  I try like this in egl_preview.cpp:
[image]
rerun output:
[image]
I google for this error code. Look like this:
[image]

BrendanHenryGithub, 2021-07-04:
Some info maybe useful:
[image]

[image]

davidplowman, 2021-07-04:
Hi again

Firstly, It sounds to me like there might be something wrong about your installation. The fact that you've had missing libraries and installed some others is clearly not right - the correct procedure is [here](https://github.com/raspberrypi/libcamera-apps#readme)

Having said that, I assume from your comments that you're running headless. Are you by any chance trying to use X forwarding? Our EGL-based preview shares dmabufs from the camera with the GPU on the Pi and I have no idea whether that's even possible with X forwarding. It's certainly not supported at the moment, but I'll have to take advice on that. In the meantime, you might find that qcam with its Qt copying-based preview would work.

BrendanHenryGithub, 2021-07-04:
Thanks a lot. 
Yes,  according to your suggestion, i found the core of problem is about X forwarding. When i turn off X Server and  hook up the monitor everything is ok.

davidplowman, 2021-08-24:
We've pushed a change which provides another implementation of the preview window that doesn't use zero-copy buffer sharing or GPU acceleration. It's implemented using Qt and can be enabled using (for example) `libcamera-hello --qt-preview`. Note that it's very slow compared with the other preview implementations (so we never use it unless it's requested explicitly), and it would be best to avoid large images. But it does work with X forwarding. I think that's enough to close this thread - but please file a new issue if anything crops up.

closed 2021-08-24 by davidplowman

Petros626, 2023-02-24:
@BrendanHenryGithub 

Hey there,

I got the same problem like you but in my case it was `what(): failed to import fd19`. So with the installation from here (https://github.com/raspberrypi/libcamera-apps#readme) it should work ?

Would appreciate it, if you could answer this.