raspberrypi/rpicam-apps #60: raspiraw-like app

charlieharley2, 2021-07-23:
Hi, thanks for sharing these useful libcamera-apps.

I was wondering if it could be possible to make a [raspiraw](https://github.com/raspberrypi/raspiraw)-like app using libcamera library as well. If that was possible, how would one go about building such an app. I will appreciate any guidelines you could provide. 

I have been reading the code for libcamera-apps and raspiraw and, as mentioned in your README, unlike raspiraw, libcamera does not allow to directly set "modes" (only indirectly by the setting a frame size). Also, raspiraw allows you to set several sensor registers directly, whereas I could not find register sets in any of the libcamera-apps (maybe that is done in libcamera??). Please correct me if I am wrong.

My interest in a raspiraw-like app based on libcamera is the [array of apps](https://stamm-wilbrandt.de/en/Raspberry_camera.html) tha can be developed with raspiraw, but MMAL and  this and the older apps (raspiraw, raspivid, raspistill, etc) are now considered legacy and libcamera is the way forward.

Thank you!

davidplowman, 2021-07-24:
Hi, thanks for your question. You're correct that libcamera gives you only limited control over sensor modes currently, and it certainly doesn't allow applications to define register sets to send to the camera. Those are regarded as the business of the kernel sensor driver (e.g. https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/media/i2c/imx477.c). You're also free to change this for your own purposes so as to have the modes you want, though there's more effort involved now that it's in the kernel. We are looking to provide better "hints" to libcamera so as to choose more appropriate modes in general, but I think the issue where the sensor is driven only by its kernel driver is more fundamental.

naushir, 2021-08-13:
As @davidplowman has mentioned, you are free to update register sets directly in the kernel driver.  The functionality provided by raspiraw is not really suited for the libcamera API, but we do have libcamera-raw that does save raw "video" streams.

closed 2021-08-13 by naushir