raspberrypi/rpicam-apps #66: The preview image is wrong when the preview width is greater than 2048 using Pi3.

glddiv, 2021-08-13:
When using libcamera-still to set the width of the preview resolution (--viewfinder-width) greater than 2048 on Pi3, the image is not displayed correctly. (The saved picture is normal)
The effect is as follows:
[image]

There is no problem using the same SD Card on Pi4.

kernel version:
`Linux raspberrypi 5.10.52-v7+ #1444 SMP Thu Aug 12 20:22:57 BST 2021 armv7l GNU/Linux`

firmware version:
```
Aug 11 2021 14:01:51 
Copyright (c) 2012 Broadcom
version 5c8758dbac5b27911d2384c5dcd5db843528b606 (clean) (release) (start_x)
```

naushir, 2021-08-13:
Could you provide the exact command line you are using to get this corruption?  Additionally, what sensor is this on?
Providing any log output from libcamera would also be helpfuly.

glddiv, 2021-08-16:
Sorry for that. I thought this might be a known issue or limitation.
libcamera-still:
```
pi@raspberrypi:~/libcamera-apps/build $ ./libcamera-still -t 0 --viewfinder-width=2592 --viewfinder-height=1944
[0:02:52.656226158] [1281]  INFO Camera camera_manager.cpp:294 libcamera v0.0.0+2880-9947b4cf
[0:02:53.057193702] [1281]  INFO Camera camera.cpp:870 configuring streams: (0) 2592x1944-YUV420
[0:02:53.058737348] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt RG10 Score: 1808.49 (best 1808.49)
[0:02:53.059017396] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt RG10 Score: 4738.67 (best 1808.49)
[0:02:53.059145647] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt RG10 Score: 4834.49 (best 1808.49)
[0:02:53.059266501] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt RG10 Score: 8332 (best 1808.49)
[0:02:53.059395116] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt pRAA Score: 1308.49 (best 1308.49)
[0:02:53.059518679] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt pRAA Score: 4238.67 (best 1308.49)
[0:02:53.059635886] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt pRAA Score: 4334.49 (best 1308.49)
[0:02:53.059752781] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt pRAA Score: 7832 (best 1308.49)
[0:02:53.059911454] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt RGGB Score: 2808.49 (best 1308.49)
[0:02:53.060074606] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt RGGB Score: 5738.67 (best 1308.49)
[0:02:53.060210671] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt RGGB Score: 5834.49 (best 1308.49)
[0:02:53.060334181] [1283]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt RGGB Score: 9332 (best 1308.49)
[0:02:53.060977051] [1283]  INFO RPI raspberrypi.cpp:619 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 3280x2464-pRAA
[0:02:53.175439092] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Sharpness = 1.000000
[0:02:53.175806394] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Saturation = 1.000000
[0:02:53.175929488] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Contrast = 1.000000
[0:02:53.176050237] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Brightness = 0.000000
[0:02:53.176146035] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:02:53.176362426] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: NoiseReductionMode = 3
[0:02:53.176664612] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AeMeteringMode = 0
[0:02:53.176985292] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AeExposureMode = 0
[0:02:53.177102186] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: ExposureValue = 0.000000
[0:02:53.177235594] [1286]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AwbMode = 0
[0:02:53.181024882] [1283]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:53.181199026] [1283]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:53.181285551] [1283]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:53.327134627] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (6 left)
[0:02:53.381911769] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (5 left)
[0:02:53.436913636] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (4 left)
[0:02:53.490836362] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (3 left)
[0:02:53.544127766] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (2 left)
[0:02:53.597330616] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (1 left)
[0:02:53.650282226] [1283]  INFO RPI raspberrypi.cpp:1633 Dropping frame at the request of the IPA (0 left)

```

libcamera-vid:
```
pi@raspberrypi:~/libcamera-apps/build $ ./libcamera-vid -t 0 --codec yuv420 --width 3280 --height 2464
[0:04:10.429087767] [1376]  INFO Camera camera_manager.cpp:294 libcamera v0.0.0+2880-9947b4cf
[0:04:10.858163566] [1376]  INFO Camera camera.cpp:870 configuring streams: (0) 3280x2464-YUV420
[0:04:10.859806455] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt RG10 Score: 1500 (best 1500)
[0:04:10.860095635] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt RG10 Score: 7155.48 (best 1500)
[0:04:10.860192097] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt RG10 Score: 7244 (best 1500)
[0:04:10.860274758] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt RG10 Score: 10748.8 (best 1500)
[0:04:10.860435234] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt pRAA Score: 1000 (best 1000)
[0:04:10.860589876] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt pRAA Score: 6655.48 (best 1000)
[0:04:10.860870306] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt pRAA Score: 6744 (best 1000)
[0:04:10.861048752] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt pRAA Score: 10248.8 (best 1000)
[0:04:10.861197404] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 3280x2464 fmt RGGB Score: 2500 (best 1000)
[0:04:10.861379027] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1920x1080 fmt RGGB Score: 8155.48 (best 1000)
[0:04:10.861471688] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 1640x1232 fmt RGGB Score: 8244 (best 1000)
[0:04:10.861718678] [1382]  INFO RPI raspberrypi.cpp:121 Mode: 640x480 fmt RGGB Score: 11748.8 (best 1000)
[... cut ...]
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d,cma-512
dtoverlay=imx219

```

I tested IMX219 and IMX477 and they have the same problem, it seems that this problem has nothing to do with the camera and only has to do with the resolution.

naushir, 2021-08-16:
@gdldiv, thank you for the info.  I am away for a week, so will not get a chance to look at this for a few days.  However, could I ask you to try using the `--denoise=cdn_off` option with libcamera-vid to see if the corruption still occurs?

davidplowman, 2021-08-16:
Thanks for the information. I've had a look at this today and indeed it's easy to reproduce the problem. I think the root cause is that the maximum texture size is more restricted on the Pi 3. Unsurprisingly given the symptoms, GL_MAX_TEXTURE_SIZE appears to be set at 2048.

On the one hand I don't think there's much we can do about that, although I wonder whether the real issue is not so much that you want huge images, it's more that you perhaps want to force the camera into full read-out mode? In general we still have some issues around getting libcamera to use the correct camera mode, but there might also be some other workarounds we could look at - for example, libcamera can give us a "lower resolution stream" alongside the main one, which we might be able to use instead.

Can you explain anytthing more about the specific use-cases you have in mind, then we could give these questions some thought? Thanks!

glddiv, 2021-08-17:
Hi @davidplowman 

Thanks for your reply,
The main problem I encountered at the moment is that we have a camera with a larger resolution, and libcamera-still will choose a resolution with a width greater than 2048 as the preview resolution.

As you said, our main purpose is not to preview the large resolution, but whether the preview mode selected by libcamera-still is correct.

davidplowman, 2021-08-17:
@glddiv Thanks for the reply.

I think the most obvious idea would simply be to figure out the maximum allowed image size and be sure not to exceed it (well, certainly not by default!). That wouldn't quite fix everything, though:
- There's the risk that it might make libcamera jump to a camera mode that you don't want (for example, something cropped rather than full FoV). As I think I said, we still have problems with libcamera in this area. This would be more likely to hit Pi 3s than Pi 4s.
- It wouldn't help if you wanted to record 4K MJPEG video. I suppose we could consider setting the low resolution stream to half resolution and displaying that instead, though one could start to have problems if something else wants to use the low resolution stream (image analysis using Neural Networks, for example). And it starts producing generally twistier code paths, which I never much like!

I probably need to sleep on that a bit.

glddiv, 2021-08-17:
Does the current ISP support image resize operation? 
Maybe it is a good choice if you can use ISP for resizing?

davidplowman, 2021-08-24:
We've pushed a modification which detects the maximum allowed image size and uses this to limit what it asks for from the camera. This should largely fix this issue. If anyone runs into the remaining awkward case of trying to record (for example) 4K MJPEG on a Pi 3 with a preview, please submit a new report and we'll think some more on that one. Thanks!

closed 2021-08-24 by davidplowman

glddiv, 2021-08-24:
Thank you very much, I tested it and it works great!