raspberrypi/rpicam-apps #77: libcamera still: keypress / signal mode

StevePitt, 2021-09-09:
Hi! 

I am trying to use the signal mode triggered from a python script on a headless Raspi.
The idea is to warm up the camera and then call for capture at the desired time.

Right now I am simply calling libcamera-sill via a subprocess from the mentioned python script. 
How do I send a signal from the control script to libcamera-still? 
How is the signal defined? 
Do I need to send it via a socket?

davidplowman, 2021-09-09:
Hi, you should only need to send it a signal, that is, the equivalent of `kill -SIGUSR1 $LIBCAMERA_STILL_PID`.  You can check this works initially in a terminal window: first enter `libcamera-still -t 0 -s -o test.jpg &` and then type `kill -SIGUSR1 $!`  Hope that helps!

StevePitt, 2021-09-10:
Thanks! I was able to reproduce the terminal part. The control script part ... not so much. 
When calling `subprocess.popen ('libcamera-still', '-t 0'. '-o test.jpg', '-s'` via UDP, libcam blocks the UDP / the control script. 

I experimented with listener and camera functionality running in separate threads, passing the output to queues, but nothing helped so far. 
Any hint where I should look at?
Thanks!

StevePitt, 2021-09-10:
Nevermind... I joined the threads...
Thanks for your time!

closed 2021-09-10 by StevePitt