raspberrypi/rpicam-apps #80: Libtiff segmentation fault on TIFFTAG_CFAPATTERN

Silvanosky, 2021-09-14:
Hello,

Using the latest version of libtiff (4.3) the api changed for the TIFFTAG_CFAPATTERN here :

https://github.com/raspberrypi/libcamera-apps/blob/4dca4c95742a75e2bda8197539a921dc00dfadac/image/dng.cpp#L285

Causing a segmentation fault at runtime.

It should be replaced with :

```
TIFFSetField(tif, TIFFTAG_CFAPATTERN, 4, bayer_format.order);
```

I don't know if making a pull request is relevant if it works on Raspberrypi OS.

But on my yocto build I use the latest version of libtiff from git so I write here to help other people (I did not found a lot of doc on this).

Hope it is helpful

davidplowman, 2021-09-14:
Hi, thanks for the info. I don't suppose we'll do anything if it works on Raspberry Pi OS but I'll be sure to check the behaviour of the up-coming version, it's possible that things might have changed there.

davidplowman, 2021-09-15:
Indeed there is a problem in our candidate Bullseye images, so thanks for posting this. Hopefully that fix (https://github.com/raspberrypi/libcamera-apps/pull/81) will help you out too.

Silvanosky, 2021-09-15:
Nice ! thanks for the PR and happy to help

davidplowman, 2021-09-20:
I think this should be fixed now for anyone who updates their code from the repo, certainly our Bullseye images are fine if folks "sudo apt update libcamera-apps". So I'm going to close this one - but please open another report if anything else comes to light. Thanks!

closed 2021-09-20 by davidplowman