raspberrypi/rpicam-apps #83: Custom lens shading table. Where to begin?

Schaggo, 2021-09-20:
Hi, 

I am using Arducam's IMX230 and adapted an MFT lens to it. The images are crisp and feature little to no aberrations but feature a green smudge in the image's center. I am pretty sure the lens shading table that comes with Arducam's precompiled version of libcamera is to blame. 

How do I start calibrating the sensor for the new lens? 

Thanks for your time!

davidplowman, 2021-09-20:
Hi, I don't think there's anything I can do to help you with Arducam's precompiled binaries, you will have to go to them for support. Sorry I can't be more helpful.

I would of course encourage Arducam to contribute to the public libcamera project at which point it would be possible for me to support their camera modules.

Schaggo, 2021-09-20:
Thank you for your reply! 
Yeah, the guys over at Arducam seem a little ... odd to say the least. 

I will go bug them a little.
Thanks!

Schaggo, 2021-09-21:
Sorry to bug you again, Arducam seems uncooperative.
I went and compiled the libcam-apps from this repo, but still have no clue where to start tuning/calibrating for my lens.

Am I right to believe that this section in the readme is what I am looking for?

``To try your own camera tuning, make a raspberrypi subfolder in a folder of your choice (let us say, /home/pi). Copy the JSON file for the camera to the raspberrypi subfolder, without changing its name, where you can edit it. Setting the following environment variable will cause libcamera to load your tuning file in preference, for example run LIBCAMERA_IPA_CONFIG_PATH=/home/pi ./libcamera-hello.``

If yes, how does the mentioned JSON structure look like and how do I get the values?

davidplowman, 2021-09-21:
Hi again

Our approach with libcamera has been to use and distribute open source camera drivers and to provide a completely open system for re-tuning our pipeline and ISP for different image sensors, and this is the document that you have found. If you look in libcamera/src/ipa/raspberrypi/data you'll see a number of JSON files for different kinds of sensor, including the official Raspberry Pi ones but also 3rd party ones where an open source driver was available or has been provided.

Unfortunately, in that folder, you'll see nothing for the imx230. That's because, as far as I can tell, Arducam are using proprietary drivers and software which bypasses all our published tuning methods. I don't even know if they're using the Pi's ISP at all, or if there's an ISP on the camera board. As regards libcamera-apps they're allowed to take our code and do pretty much whatever they want with it though I also suspect what they're doing may contravene the libcamera license (IANAL, of course).

But the long and short of it is, there's nothing we can do to help you with Arducam cameras, you will have to persuade Arducam to support you. Or buy a non-Arducam sensor, of course! Sorry not to have better news for you.

Schaggo, 2021-09-21:
Ah man... a bummer...
To clarify: yes, they seem to have an STM32 for control on their board. 

Thank you very much for your time anyway!

closed 2021-09-21 by Schaggo

kbingham, 2021-09-21:
If Arducam are packaging libcamera and releasing binaries, and you have purchased a device and received those binaries, you can request access to the source code for libcamera. They are legally obliged to provide any changes they have made to libcamera, but as David suggests, if they have modified the IPA component from Raspberry Pi, that has a separate license.

Schaggo, 2021-09-21:
They use your repo as the source for libcam-apps but include their own [IPA ](https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-access-and-program-the-pivariety-camera-using-command-line-python-opencv-and-gstreamer/)-  

My application is somewhat dependent on the resolution of the IMX230. I'd be glad to buy cams/chips directly from you, but the HQ cam has too little resolution.

Schaggo, 2021-09-22:
Alright. Arducam agreed to send me a compiled IPA once I send them a tuning file. 
To produce the tuning file I want to follow this [guide ](https://datasheets.raspberrypi.org/camera/raspberry-pi-camera-guide.pdf).

As I understand it I need a colorimeter and an X-Rite ColorChecker card. I take a bunch of pictures of the card and of a neural background in different light color temps, whoop the dataset in the utility app and receive a tuning file.
Two questions beforehand: 

Does it need to be the exact x-rite color checker (24 squares)? - I have a spyder color checker with 24 squares and an x-rite with 120. Would it work with either of them or does the app try to find colors based on relative position?

How accurate does the colorimeter have to be? I have the opportunity to borrow one from a university but could/would also buy something like [this](https://www.opplelighting.de/de/product/indoor/light-master/light-master-g3/light-master-iii) 

Thanks for your time!

davidplowman, 2021-09-22:
Yes, that's the guide to follow. I would recommend getting the x-rite color checker. The spyder ones I've seen online tend to have the squares in a different order which would mess things up. 

That colorimeter looks OK, you need lux and colour temperature (kelvin) measurements.

In your case the problem is mostly the lens shading. In order to capture flat field pictures I would suggest buying a video light, one of those ones made up of lots of LEDs and a diffuser. Get one that can output a couple of different colour temps. They're far from perfect, of course, but even the cheap ones are less frustrating than trying to take pictures of walls and/or ceilings, in my experience. (Professional equipment like integrating spheres cost a fortune, but most folks really don't need that - it depends on what you're hoping to accomplish.)

Obviously you need to be able to capture raw images. I assume Arducam's libcamera and libcamera-apps build will support this...

Schaggo, 2021-11-08:
Idk if you're interested in this, but I managed to calibrate the lens sensor combination. 
For some idiotic reason I "needed" to build my own integrating sphere, barium sulfate and all... 
Anyway, thanks again for your help (or sorry to bother you).

kbingham, 2021-11-08:
Hi @Schaggo,

Quite interested in how you built your sphere, and how well it worked. Tempted to build something like that here someday too.
Have you got any pictures? notes? suggestions now that you've done one?

Tempted to try to 3d print a couple of half sphere shells and piece something together sometime ...

Schaggo, 2021-11-08:
Including the Bosch spray gadget the total cost for materials was around 200€ -deducting the red primer it should be around 150€.

kbingham, 2021-11-08:
Nice, thank you for the details. It's bigger than I imagined ;-)

kbingham, 2021-11-08:
> Arducam's IMX230

Is it worth asking Arducam to consider supporting this with upstream drivers ? They've recently submitted the IMX519 support after all...

Schaggo, 2021-11-08:
One of the main selling points of the product they are offering with the imx230 is utterly useless: the lens package, including the motorized focus. All the gains of high res were made useless because of it. 

The “glass” is terrible with uneven sharpness and really strong aberrations all over the place. The autofocus has a very bad success rate focussing. So I went and bugged you about lens shading... 

The only way the sensor makes sense for broad adoption is with a different lens package or no lens at all. 

Sadly, their implementation of libcam isn't great either. This led me to the idea of writing something that grabs pure sensor data (ignoring the ISP), and then process it on a desktop. But no idea if it's even possible to get high-quality images that way.