raspberrypi/rpicam-apps #90: libcamera-jpeg SIGABRT @libcamera::CameraConfiguration::at

kbingham, 2021-09-28:
From: Damian Wrobel 2021-09-21 19:01:03 CEST : https://bugs.libcamera.org/show_bug.cgi?id=78

I'm observing a SIGABRT @libcamera::CameraConfiguration::at when using libcamera-jpeg application as follows:

$ libcamera-jpeg -o test.jpg
Preview window unavailable
[144:40:33.264346482] [28470]  INFO Camera camera_manager.cpp:294 libcamera v0.0.0+1-dab03790
[144:40:33.335326696] [28470]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[144:40:33.335481490] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RG10 Score: 2218 (best 2218)
[144:40:33.335544174] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 2041.48 (best 2041.48)
[144:40:33.335579655] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 1500)
[144:40:33.335612118] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RG10 Score: 5004.81 (best 1500)
[144:40:33.335666376] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pRAA Score: 1718 (best 1500)
[144:40:33.335696728] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1541.48 (best 1500)
[144:40:33.335728542] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 1000)
[144:40:33.335768856] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pRAA Score: 4504.81 (best 1000)
[144:40:33.335805430] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RGGB Score: 3218 (best 1000)
[144:40:33.335834615] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RGGB Score: 3041.48 (best 1000)
[144:40:33.335877725] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGGB Score: 2500 (best 1000)
[144:40:33.336105166] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RGGB Score: 6004.81 (best 1000)
[144:40:33.336404236] [28474]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pRAA
[144:40:33.367341751] [28474]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[144:40:33.367382084] [28474]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[144:40:33.367407139] [28474]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[144:40:33.478414460] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (6 left)
[144:40:33.509496510] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (5 left)
[144:40:33.544003622] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (4 left)
[144:40:33.578627770] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (3 left)
[144:40:33.609094866] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (2 left)
[144:40:33.642823804] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (1 left)
[144:40:33.675786142] [28474]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (0 left)
[144:40:38.517760186] [28470]  INFO Camera camera.cpp:937 configuring streams: (0) 3280x2464-YUV420
[144:40:38.517915609] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt BG10 Score: 1500 (best 1500)
[144:40:38.517977572] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 7155.48 (best 1500)
[144:40:38.518018478] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG10 Score: 7244 (best 1500)
[144:40:38.518054182] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 10748.8 (best 1500)
[144:40:38.518114921] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt BA81 Score: 2500 (best 1500)
[144:40:38.518153069] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BA81 Score: 8155.48 (best 1500)
[144:40:38.518202383] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA81 Score: 8244 (best 1500)
[144:40:38.518238383] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 11748.8 (best 1500)
[144:40:38.518275401] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pBAA Score: 1000 (best 1000)
[144:40:38.518321196] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 6655.48 (best 1000)
[144:40:38.518600914] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBAA Score: 6744 (best 1000)
[144:40:38.518732801] [28474]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 10248.8 (best 1000)
[144:40:38.518926521] [28474]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 3280x2464-pBAA
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = libcamera::StreamConfiguration; _Alloc = std::allocator<libcamera::StreamConfiguration>; std::vector<_Tp, _Alloc>::reference = libcamera::StreamConfiguration&; std::vector<_Tp, _Alloc>::size_type = unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

Environment:

- RPi4 (4GB) with:
 -- dtoverlay=imx219 and camera connected,
 -- kernel 5.10.60: e4cb65cf23c78b4912786aaf8467aa215d4e1d88,

- libcamera-apps: https://github.com/raspberrypi/libcamera-apps/commit/4868f6922c99669d2b36ffe0b0b35e4cd5b0ea53

- libcamera: https://github.com/libcamera-org/libcamera/commit/a881b90b5d2a51e196137221a7487055c03926cd

Backtrace:

$ gdb --args libcamera-jpeg -o test.jpg
GNU gdb (GDB) Fedora 10.2-3.fc34
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "armv7hl-redhat-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
[... 51 lines cut ...]
[144:42:15.519191418] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BA81 Score: 8155.48 (best 1500)
[144:42:15.519230473] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA81 Score: 8244 (best 1500)
[144:42:15.519267362] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 11748.8 (best 1500)
[144:42:15.519307213] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pBAA Score: 1000 (best 1000)
[144:42:15.519345583] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 6655.48 (best 1000)
[144:42:15.519384804] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBAA Score: 6744 (best 1000)
[144:42:15.519422767] [28520]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 10248.8 (best 1000)
[144:42:15.519609486] [28520]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 3280x2464-pBAA
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = libcamera::StreamConfiguration; _Alloc = std::allocator<libcamera::StreamConfiguration>; std::vector<_Tp, _Alloc>::reference = libcamera::StreamConfiguration&; std::vector<_Tp, _Alloc>::size_type = unsigned int]: Assertion '__n < this->size()' failed.

Thread 1 "libcamera-jpeg" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
49	  return ret;
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0xb6a76ffc in __GI_abort () at abort.c:79
#2  0xb6e763d4 in std::__replacement_assert (__condition=0xb6f17e9c "__n < this->size()", 
    __function=0xb6f17d24 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = libcamera::StreamConfiguration; _Alloc = std::allocator<libcamera::StreamConfi"..., 
    __line=1045, __file=0xb6f17e74 "/usr/include/c++/11/bits/stl_vector.h") at /usr/include/c++/11/armv7hl-redhat-linux-gnueabi/bits/c++config.h:514
#3  std::vector<libcamera::StreamConfiguration, std::allocator<libcamera::StreamConfiguration> >::operator[] (this=<optimized out>, __n=1) at /usr/include/c++/11/bits/stl_vector.h:1045
#4  std::vector<libcamera::StreamConfiguration, std::allocator<libcamera::StreamConfiguration> >::operator[] (this=<optimized out>, this=<optimized out>, __n=1) at /usr/include/c++/11/bits/stl_vector.h:1043
#5  libcamera::CameraConfiguration::at (this=<optimized out>, index=index@entry=1) at ../src/libcamera/camera.cpp:215
#6  0xb6f82824 in LibcameraApp::ConfigureStill (this=0xb238b010, flags=<optimized out>) at /usr/include/c++/11/bits/unique_ptr.h:173
#7  0x00407d6c in event_loop (app=...) at /usr/src/debug/libcamera-apps-0.0.0-0.2.20210915git4868f69.fc34.arm/apps/libcamera_jpeg.cpp:63
#8  main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/libcamera-apps-0.0.0-0.2.20210915git4868f69.fc34.arm/apps/libcamera_jpeg.cpp:103

kbingham, 2021-09-28:
@naushir @davidplowman, This has been reported on our issue tracker but a quick glance shows that it may be the libcamera-jpeg app accessing an invalid StreamConfiguration offset.

Is this an issue you're aware of or can replicate?
It may be that we need to add some more defensive support in the libcamera StreamConfiguration API here of course ...

davidplowman, 2021-09-28:
@kbingham Hi Kieran, thanks for this. I wasn't aware of it (or it would have been fixed!) but there does seem to be an obvious clanger in ConfigureStill. Let me fix that and then we'll see how things are.

kbingham, 2021-09-28:
@dwrobel Could you retest this with the latest code please?

dwrobel, 2021-10-01:
I updated to the following versions:

- [libcamera](https://github.com/libcamera-org/libcamera/commit/c8c1d07cea2d8ef63905bf4f0b5266a5e6fd749f),
- [libcamera-apps](https://github.com/raspberrypi/libcamera-apps/commit/8b232ca68b805f1531f623519e8431aa17f487bb)

and libcamera-jpeg seems to get stuck as follows:
```
$  libcamera-jpeg -o test.jpg
Preview window unavailable
[0:01:20.210404229] [867]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+1-24db7f96
[0:01:20.310786412] [867]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[0:01:20.311015649] [879]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RG10 Score: 2218 (best 2218)
[0:01:20.311120444] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 2041.48 (best 2041.48)
[0:01:20.311158314] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 1500)
[0:01:20.311189943] [879]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RG10 Score: 5004.81 (best 1500)
[0:01:20.311222627] [879]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pRAA Score: 1718 (best 1500)
[0:01:20.311252386] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1541.48 (best 1500)
[0:01:20.311283034] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 1000)
[0:01:20.311312700] [879]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pRAA Score: 4504.81 (best 1000)
[0:01:20.311350681] [879]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RGGB Score: 3218 (best 1000)
[0:01:20.311382051] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RGGB Score: 3041.48 (best 1000)
[0:01:20.311413736] [879]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGGB Score: 2500 (best 1000)
[0:01:20.311444198] [879]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RGGB Score: 6004.81 (best 1000)
[0:01:20.311606677] [879]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pRAA
[0:01:22.385439814] [879] ERROR IPCPipe ipc_pipe_unixsocket.cpp:132 Call timeout!
[0:01:22.385505924] [879] ERROR IPCPipe ipc_pipe_unixsocket.cpp:64 Failed to call sync
[0:01:22.385530349] [879] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:234 Failed to call start
```

backtrace:
```
$ gdb --args  libcamera-jpeg -o test.jpg
GNU gdb (GDB) Fedora 10.2-3.fc34
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "armv7hl-redhat-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from libcamera-jpeg...
Reading symbols from /usr/lib/debug/usr/bin/libcamera-jpeg-0.0.0-0.4.20210930git8b232ca.fc34.arm.debug...
(gdb) r
Starting program: /usr/bin/libcamera-jpeg -o test.jpg
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
[New Thread 0xb2386020 (LWP 1494)]
[New Thread 0xb1b85020 (LWP 1495)]
[New Thread 0xb1384020 (LWP 1496)]
Preview window unavailable
[0:04:56.343366707] [1491]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+1-24db7f96
[New Thread 0xb0b83020 (LWP 1497)]
[Detaching after fork from child process 1498]
[0:04:56.417113659] [1491]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[0:04:56.417261138] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RG10 Score: 2218 (best 2218)
[0:04:56.417332396] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 2041.48 (best 2041.48)
[0:04:56.417367581] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 1500)
[0:04:56.417400025] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RG10 Score: 5004.81 (best 1500)
[0:04:56.417433043] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pRAA Score: 1718 (best 1500)
[0:04:56.417464932] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1541.48 (best 1500)
[0:04:56.417497505] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 1000)
[0:04:56.417528486] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pRAA Score: 4504.81 (best 1000)
[0:04:56.417563449] [1497]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RGGB Score: 3218 (best 1000)
[... 83 lines cut ...]
#3  __pthread_cond_wait (cond=0xbeffecd8, mutex=0xbeffecc0) at pthread_cond_wait.c:619
#4  0xb6c8ec40 in __gthread_cond_wait (__mutex=<optimized out>, __cond=__cond@entry=0xbeffecd8) at /usr/src/debug/gcc-11.2.1-1.fc34.arm/obj-armv7hl-redhat-linux-gnueabi/armv7hl-redhat-linux-gnueabi/libstdc++-v3/include/armv7hl-redhat-linux-gnueabi/bits/gthr-default.h:865
#5  std::__condvar::wait (__m=..., this=this@entry=0xbeffecd8) at /usr/src/debug/gcc-11.2.1-1.fc34.arm/obj-armv7hl-redhat-linux-gnueabi/armv7hl-redhat-linux-gnueabi/libstdc++-v3/include/bits/std_mutex.h:155
#6  std::condition_variable::wait (this=this@entry=0xbeffecd8, __lock=...) at ../../../../../libstdc++-v3/src/c++11/condition_variable.cc:41
[... cut ...]
P.S.: Building libcamera-apps produces quite a lot of warnings:
```
$ curl -s https://download.copr.fedorainfracloud.org/results/dwrobel/libcamera-apps/fedora-rawhide-armhfp/02869106-libcamera-apps/builder-live.log.gz | gzip -d | grep -e '/builddir.*warning:' | wc -l
46
```
Please consider to look at them and fix some of them e.g.:
```
frame_info.hpp:45:33: note: use reference type to prevent copying
   45 |                 for (auto const t : tokens)
      |                                 ^
      |                                 &
```

davidplowman, 2021-10-01:
@dwrobel  Thanks for the extra info. Couple of things:

1. It turns out we thought we were compiling with warnings enabled but weren't. Oops. We'll fix that!

2. It seems to be trying to run the Raspberry Pi IPA through the IPC sandbox mechanism, which isn't the normal procedure. I guess there's a question as to why it would be trying to do this, and also why it wouldn't "just work" when it's fine without the IPC wrapper. @kbingham Can you help us with any suggestions about this?

kbingham, 2021-10-01:
Hard to know off the bat why it's running an isolated IPA - but if the modules are not signed, or the libraries aren't available to check the signatrues then  it will likely cause it to get isolated.

Check that the packages:
        libgnutls28-dev openssl
are installed.

We should add that to the build summary to make it easy to spot if it gets disabled.

That said, things should function with an isolated IPA - but of course it probably hasn't been tested on the RPi.
I'm away on holiday for a week now, so I'm sorry in advance for any delay in further responses.

davidplowman, 2021-10-04:
I've investigated this at fairly great length. It turns out that passing ControlLists across libcamera's pipeine handler / IPA interface is fussier when running in the "isolation" mode. This patch https://github.com/raspberrypi/libcamera-apps/pull/109 fixes the libcamera-apps end of the problem, but anything that does a camera mode switch will need a patch in libcamera itself too.

davidplowman, 2021-10-07:
The necessary patch seems just to have gone into libcamera as well now, so I _believe_ everything here should be good. Please let us know if it isn't otherwise I'll proceed to close this issue shortly.