raspberrypi/rpicam-apps #112: Building Failed on RPI CM4 64-bit OS

YusongLeng, 2021-10-06:
Hi,

Truly appreciate your excellent works! 

But I encountered a failure during building. Could you please take a look? Thank you!

I'm using RPI CM4 with 64-bit OS:
```
$ uname -a
Linux raspberrypi 5.10.63-v8+ #1457 SMP PREEMPT Tue Sep 28 11:27:02 BST 2021 aarch64 GNU/Linux
```

I once (August 30th) built `libcamera` and `libcamera-apps` following this [guide](https://www.raspberrypi.org/documentation/accessories/camera.html#libcamera-and-libcamera-apps-installation), and it was all successful at that time.

However, when I try to rebuild `libcamera` and `libcamera-apps` following the same guide today (October 6th), `libcamera` was built successful as previous, but there was an error that happened around the end of `libcamera-apps`'s compiling.

In particular, the terminal out is as below:
```
Cloning into 'libcamera-apps'...
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No previous build - default to Release build
-- Platform: arm64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libcamera'
--   Found libcamera, version 1.0
-- libcamera library found:
--     version: 1.0
--     libraries: /usr/local/lib/aarch64-linux-gnu/libcamera.so;/usr/local/lib/aarch64-linux-gnu/libcamera-base.so
--     include path: /usr/local/include/libcamera
-- Found Boost: /usr/include (found version "1.67.0") found components: program_options 
-- LIBDRM_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libdrm.so
-- X11_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libX11.so
-- EPOXY_LINK_LIBRARIES=/usr/local/lib/aarch64-linux-gnu/libepoxy.so
-- QTCORE_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libQt5Core.so
-- QTCORE_INCLUDE_DIRS=/usr/include/aarch64-linux-gnu/qt5/QtCore;/usr/include/aarch64-linux-gnu/qt5
-- Checking for OpenCV
-- OpenCV library found:
--     version: 4.5.3
--     libraries: -lopencv_core;-lopencv_imgproc;-lopencv_objdetect
--     include path: /usr/local/include/opencv4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcamera-apps/build
Scanning dependencies of target VersionCpp
Scanning dependencies of target images
[  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
Scanning dependencies of target preview
Scanning dependencies of target post_processing_stages
[  4%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
[  6%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/negate_stage.cpp.o
Generating version string: ce96233ec506 06-10-2021 (16:54:35)
[  6%] Built target VersionCpp
Scanning dependencies of target encoders
[  8%] Building CXX object encoder/CMakeFiles/encoders.dir/encoder.cpp.o
[ 10%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
[ 12%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
[ 14%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/hdr_stage.cpp.o
[... 71 lines cut ...]
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::CascadeClassifier::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::Mat::clone() const'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::rectangle(cv::_InputOutputArray const&, cv::Rect_<int>, cv::Scalar_<double> const&, int, int, int)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::putText(cv::_InputOutputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/libcamera-raw.dir/build.make:93: libcamera-raw] Error 1
make[1]: *** [CMakeFiles/Makefile2:488: apps/CMakeFiles/libcamera-raw.dir/all] Error 2
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::Mat::Mat()'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::CascadeClassifier::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::Mat::clone() const'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::rectangle(cv::_InputOutputArray const&, cv::Rect_<int>, cv::Scalar_<double> const&, int, int, int)'
/usr/bin/ld: ../post_processing_stages/libpost_processing_stages.so: undefined reference to `cv::putText(cv::_InputOutputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/libcamera-vid.dir/build.make:93: libcamera-vid] Error 1
make[1]: *** [CMakeFiles/Makefile2:426: apps/CMakeFiles/libcamera-vid.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

A difference I noticed is that in my previous building, there was no checking for `OpenCV`. It seems that the issue was brought by `OpenCV`.

Could you kindly take a look and let me know Did I miss something? Or is there a way that I could circumvent this issue?

Thank you very much!

YusongLeng, 2021-10-06:
Thank you so much for the workaround!

Yes, I build from scratch following the procedures from [here](https://www.raspberrypi.org/documentation/accessories/camera.html#libcamera-and-libcamera-apps-installation)

Specifically, the commands I used are as below:

```
# Build libcamera
sudo apt install -y meson \
                    libboost-dev libgnutls28-dev openssl libtiff5-dev \
                    qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5 qttools5-dev-tools

cd ~
sudo rm -rf libcamera*
git clone git://linuxtv.org/libcamera.git
cd libcamera

meson build
cd build
meson configure -Dpipelines=raspberrypi -Dtest=false
cd ..
ninja -C build
sudo ninja -C build install

# Build libepoxy
sudo apt install -y libegl1-mesa-dev

cd ~
git clone https://github.com/anholt/libepoxy.git
cd libepoxy

mkdir _build && cd _build
meson
ninja
sudo ninja install

# Build libcamera-apps
sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev

cd ~
git clone https://github.com/raspberrypi/libcamera-apps.git
cd libcamera-apps

mkdir build && cd build
cmake ..
make -j4
```

If I'm not wrong, I think I'm using commit `ce96233ec506ba712547b56cd5eed1d8f7d95f2c` (I cloned the repo 4 hours ago, and this commit was the most recent commit at that time).

davidplowman, 2021-10-06:
Thanks for the information. As you may have noticed, there a bit of churn going on as we're getting ready to release some new images, so I suppose it's possible something bad has happened. However, I can't really see what's gone wrong in your case - evidently you have OpenCV installed, which it has detected, but then the linker doesn't seem to have included the libraries for it. I'll try doing some installs of the latest code on some fresh images and see what happens.

davidplowman, 2021-10-07:
Glad the workaround is OK for you. I did try building on clean 64 bit Buster and Bullseye images after installing OpenCV but couldn't get it to fail. Also I notice that even Bullseye only comes with OpenCV 4.5.1. Can you confirm what OS you're using and where your OpenCV installation is coming from? Thanks!

closed 2021-10-20 by naushir