raspberrypi/rpicam-apps #125: libcamera-apps not working on RPi 3 Model B+

babinos87, 2021-10-12:
Hi,

I am trying to run `libcamera-apps` in my Raspberry Pi 3 Model B+. I get the below error:

```
$ libcamera-jpeg -o ~/Desktop/test6.jpg
[0:26:34.032643212] [2313]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3097-fa7a95bf
ERROR: *** no cameras available ***
```

but my camera is detected:
```
$ sudo vcgencmd get_camera
supported=1 detected=1
```

same for 

```
$ sudo libcamera-hello 
[0:49:18.747128742] [3096]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3097-fa7a95bf
ERROR: *** no cameras available ***
```

and, if I use `raspistill`, the camera seems to work fine:
```
$ raspistill -o ~/Desktop/test.jpg
```

works great. I can see a pop up window with video, image is saved.

I am following the guide from Raspberry Pi site: https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps

I have built by specifying `-Dv4l2=true` to the `meson configure` command and without. The result is the same. I also tried to run with `-n`, still same error.

I am clueless to what is happening. The camera seems to be installed correctly and works fine, so the problem in my case seems to be with the `libcamera-apps`. Any ideas what could be missing here, or at least some guidance to debug this?

Some info of my setup:

```
$ sudo vcgencmd version
Sep 28 2021 11:33:44 
Copyright (c) 2012 Broadcom
version 778b6a4f3c7d8d48bb63c02c47bcfbac79417bea (clean) (release) (start_x)
```

```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

`/boot/config.txt` file:
```
$ cat /boot/config.txt 
disable_overscan=1

# I have tried both setting the below to on and commenting out, same result
dtparam=i2c_arm=on
dtparam=audio=on

[pi4]
max_framebuffers=2

[all]
# imx219 did not work at all - not even raspistil.
# raspistill works with vc4-fkms-v3d
dtoverlay=vc4-fkms-v3d
#dtoverlay=imx219
start_x=1
gpu_mem=128
```

Let me know if I can provide further info.

Many Thanks,
Babis

babinos87, 2021-10-13:
Hi @davidplowman , thanks for your reply. I have already tried this and it didn't help. In fact, when I set `dtoverlay=imx219`, then the camera is not detected at all:

```
$ sudo vcgencmd get_camera
supported=1 detected=0
```

babinos87, 2021-10-13:
OK, I got a fresh Raspbian Buster, ran all the steps listed in official RPI site regarding the build of `libcamera-apps`.

Something works, but something is wrong.

My config

```
$ cat /boot/config.txt 
# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=imx219
start_x=1
gpu_mem=128
```

Camera seems not detected:
```
$ sudo vcgencmd get_camera
supported=1 detected=0
```

`libcamera-hello` - NOT working
```
$ libcamera-hello
libEGL warning: DRI2: failed to authenticate
[0:19:58.125501550] [2636]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3100-0ce723fc
[0:19:58.184663587] [2636]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[0:19:58.185014046] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RG10 Score: 2218 (best 2218)
[0:19:58.185154250] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 2041.48 (best 2041.48)
[0:19:58.185228050] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 1500)
[0:19:58.185290601] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RG10 Score: 5004.81 (best 1500)
[0:19:58.185356016] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt pRAA Score: 1718 (best 1500)
[0:19:58.185416326] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 1541.48 (best 1500)
[0:19:58.185478095] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 1000)
[0:19:58.185538302] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pRAA Score: 4504.81 (best 1000)
[0:19:58.185604759] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 3280x2464 fmt RGGB Score: 3218 (best 1000)
[0:19:58.185664028] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RGGB Score: 3041.48 (best 1000)
[0:19:58.185724964] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGGB Score: 2500 (best 1000)
[0:19:58.185785326] [2647]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RGGB Score: 6004.81 (best 1000)
[0:19:58.186086203] [2647]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pRAA
[0:19:58.325198965] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:19:58.325416199] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:19:58.325475781] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:19:58.325537341] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:19:58.325588955] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:19:58.325667129] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:19:58.325730148] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:19:58.325785668] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:19:58.325836187] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:19:58.325924362] [2651]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:19:58.327021831] [2647]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:19:58.327078704] [2647]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:19:58.327122193] [2647]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:19:58.412352092] [2647]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:19:58.443061647] [2647]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:19:58.479978059] [2647]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:19:58.513771068] [2647]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:19:58.546534472] [2647]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[... 174 lines cut ...]
[0:22:53.492038086] [2756]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt RGGB Score: 6004.81 (best 1000)
[0:22:53.492324748] [2756]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pRAA
[0:22:53.653029868] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:22:53.653868551] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:22:53.654310887] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:22:53.654478645] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:22:53.655653312] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:22:53.656126377] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:22:53.656503037] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:22:53.656767356] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:22:53.656880531] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:22:53.657064538] [2760]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[... cut ...]
[0:22:53.881427060] [2756]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:22:53.912288200] [2756]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:22:53.945657890] [2756]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```

So it seems that using `-n` works ok, so does the `qt-preview` as @davidplowman suggested. But without `-n`, none of those apps seem to work. Seems that this relates to qt somehow?

babinos87, 2021-10-16:
OK, hold on a second because I've just realized something crucial; when you wrote _uncomment the dtoverlay=vc4-fkms-v3d line in the /boot/config.txt_, you literally meant to just uncomment (AND keep the other dtoverlay setting - `imx219` in my case)?

What I was trying was to keep one but not another, so I was trying with EITHER `dtoverlay=imx219` or `dtoverlay=vc4-fkms-v3d` - and I was getting the above mentioned errors in both cases.

Stupid me, not very clear from the other hand when I re-read the instructions (for someone doing this for their very first time).

So, when I keep both in my `/boot/config.txt`, ie:

```
...
dtoverlay=vc4-fkms-v3d
dtoverlay=imx219
...
```

all commands work great (both with `-n` and without)!!

Thanks @davidplowman  :)

closed 2021-10-16 by babinos87

davidplowman, 2021-10-16:
Glad you've figured it out. Sorry if the instructions aren't always totally clear, the new documentation website is still WIP to some extent. And life will get much better once the new Bullseye OS is released soon, stuff will just work "out of the box".

harshalak77, 2022-04-03:
Hi @babinos87 and @davidplowman,
Here is my config file for RPI4

[pi4]
dtoverlay=vc4-fkms-v3d
dtoverlay=imx219
# Run as fast as firmware / board allows
arm_boost=1

and 

Here is libcamera-hello

libcamera-hello
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[0:32:00.312563346] [2941]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
ERROR: *** no cameras available ***

I have both 
dtoverlay=vc4-fkms-v3d
dtoverlay=imx219
enabled(uncommented) but still camera doesn't work please help......

kevleyski, 2022-05-26:
Also having problems with dtoverlay=vc4-fkms-v3d (Firmware/Fake)KMS vs Full KMS dtoverlay=vc4-kms-v3d (which is ok but then the touchscreen doesn't work)

dhaval-khatri1996, 2022-08-28:
@harshalak77 were you able to solve your issue?

MightyMirko, 2022-10-26:
Hello,

please make sure your driver config is in all section, not in pi4

Petros626, 2023-02-24:
> Hi @babinos87 and @davidplowman, Here is my config file for RPI4
> 
> [pi4] dtoverlay=vc4-fkms-v3d dtoverlay=imx219 arm_boost=1
> 
> and
> 
> Here is libcamera-hello command output
> 
> libcamera-hello libEGL warning: DRI2: failed to authenticate Made X/EGL preview window [0:32:00.312563346] [2941] INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e ERROR: *** no cameras available ***
> 
> I have both dtoverlay=vc4-fkms-v3d dtoverlay=imx219 enabled(uncommented) but still camera doesn't work please help......

I have the same issue since days. I Installed a new image (Bullseye) and tried again, but the preview always fails. I need this preview for my work so the `--qt-preview` works, but isn't the solution for it. Did you solve it @harshalak77 ?

Petros626, 2023-02-24:
> 

what you mean by that?

Petros626, 2023-02-24:
> Generally my advice is to start by connecting a keyboard and HDMI display directly to your Pi, downloading the latest image, touching _absolutely nothing else in the image_, and seeing if `libcamera-hello` works.
> 
> If that works as it should then proceed to change just one thing at a time until you reach a point where it doesn't. If it doesn't "just work" at the outset, then something is either not plugged in properly or actually broken.
> 
> If you haven't been able to get things working by the above steps, then please file a new issue. Thanks!

Thanks I'll try it