raspberrypi/rpicam-apps #137: Suggestion: libcamera-apps as a dev library

pablofrommars, 2021-11-09:
As stated in the documentation:

> Raspberry Pi’s libcamera-apps are not only command line applications that make it easy to capture images and video from the camera, they are also examples of how users can create their own libcamera-based applications with custom functionality to suit their own requirements.

I believe it would be very useful if the core part of libcamera-apps could compile to a library (libcamera-apps-dev) to facilitate the creation of custom functionality while retaining upstream compatibility and bug fixes.

I would be interested in helping out as much as I can if others find the suggestion valuable.

Thanks to anyone involved.

kbingham, 2021-11-09:
@pablofrommars What functionality and features would you like to see in the library?

pablofrommars, 2021-11-09:
@kbingham I am in the process of porting an app from the "legacy" system to libcamera, it uses libavformat to process h264 frames. My plan is to derive from the Output class and modify libcamera_vid to get the job done.

This would add a bunch of dependencies to this project. Having a "dev" library, could really speed up development and more importantly maintenance.

kbingham, 2021-11-10:
So is this more about encoding (and presumably specifically on the RPi), than capturing images?

pablofrommars, 2021-11-10:
That's correct and being able to embed it.

naushir, 2021-11-12:
Could you not link against the libcamera_app.so library that is created with the build?

pablofrommars, 2021-11-12:
@naushir Fair point ;) I appreciate the suggestion.

closed 2021-11-12 by pablofrommars