raspberrypi/rpicam-apps #138: Stream over TCP has severe latency and visual artefacts

laukejas, 2021-11-09:
I am trying to get a OV5647 camera stream to work with Raspberry Pi 3B+. Running Raspberry PI OS, x32, version 11 (5.10.63-v7+ armv7l). Pi is connected over WiFi to home router. I am connecting to Pi via SSH from my computer (cable connection to the router).

Video recording works fine - high quality, no artefacts. However, video streaming over TCP connection introduces huge latency (around 5 seconds during the start of the stream, but after a few minutes, latency grows to 30 seconds or so). Also, there are visual artefacts when objects move fast in the frame. In the picture below, I am moving my hand in front of the camera at moderate speed, and resulting pixelation looks like some kind of compression issue.

[image]

I am running the stream on the Pi with:

`pi@raspberrypi:~ $ libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:5000`

And then connecting to it on my computer via VLC with:

`tcp/h264://192.168.1.170:5000`

When I start the stream, I also get this in the console:

```
pi@raspberrypi:~ $ libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:5000
Preview window unavailable
[0:12:24.525446840] [1515]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:12:24.684323433] [1515]  INFO Camera camera.cpp:937 configuring streams: (0) 640x480-YUV420
[0:12:24.684526295] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2354 (best 2354)
[0:12:24.684639367] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 2136.67 (best 2136.67)
[0:12:24.684696658] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1787 (best 1787)
[0:12:24.684750824] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 1500 (best 1500)
[0:12:24.684807334] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 2500 (best 1500)
[0:12:24.684861344] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1854 (best 1500)
[0:12:24.684912646] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1636.67 (best 1500)
[0:12:24.684965249] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1287 (best 1287)
[0:12:24.685017072] [1519]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 1000 (best 1000)
[0:12:24.686911325] [1519]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 640x480-pBAA
[0:12:24.709137003] [1519]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:12:24.709200961] [1519]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:12:24.775540655] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:12:24.835211337] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:12:24.866674212] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:12:24.898431356] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:12:24.929476631] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:12:24.961015234] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:12:24.992405767] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:12:25.023586040] [1519]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```

I have tried various settings, such as reducing framerate and resolution, but it doesn't help - in fact, limiting framerate also introduces some kind of buffering freeze - feed often stops for around 10 seconds, and then resumes with a fast-forward, the freezes again.

WiFi connection is strong. I get no such delay with old stack (raspivid) with the same hardware and WiFi.

Any suggestions?

davidplowman, 2021-11-10:
Hi, I had a look at this and tried exactly the same `libcamera-vid` command as you. I think we may be having some issues related to vlc and the framerate at which it runs. It does seem to have some options to adjust this but they didn't seem to work for me. I had more luck with ffplay:
```
ffplay tcp://<id-addr>:<port> -vf "setpts=N/30"
```
Of course, over  a longer time span the clocks may still likely to drift.

As regards why raspivid seemed better, I'm not sure. I tried it too and it still seemed to jump and glitch in vlc. Maybe it has a different default resolution, bitrate or framerate which changes the symptoms?

For interest, I captured the encoded stream on the Pi (`libcamera-vid -t 0 --inline -o - | tee check.h264 | nc -l 5000`), and received it to a file on the PC (`nc <ip-addr> <port> > test.h264`). I could verify that the bytes sent and received were identical. Bizarrely, vlc played the stream from file perfectly on the Pi but not on the PC. Puzzling. ffplay also had no trouble on either device.

laukejas, 2021-11-10:
Hi, @davidplowman , thank you for the reply. How exactly did you get this running? I tried running that line on my Pi:
`ffplay tcp://0.0.0.0:5000 -vf "setpts=N/30"` 

And I got:
```
Could not initialize SDL - displayIndex must be in the range 0 - -1
(Did you set the DISPLAY variable?)
```

As for the PC, I assume I would still be connecting via VLC with **tcp/h264://192.168.1.170:5000**, correct?

davidplowman, 2021-11-10:
I ran the `ffplay tcp://...` command on my PC.  Possibly I installed ffplay or ffmpeg at some time in the past, can't remember! On the Pi I only used `ffplay <filename>` to check if the captured bitstream played.

laukejas, 2021-11-10:
@davidplowman , awesome! Thank you for that clarification. This is working beautifully. Super high quality, very little latency. I guess it was VLC to blame, after all.

davidplowman, 2021-11-11:
Thanks for that improved command line. We'll make a note of it to help other users!

closed 2021-11-11 by davidplowman