raspberrypi/rpicam-apps #139: libcamera-still -o - sends output to a file named -

knutegit, 2021-11-13:
libcamera-still -o - sends the image data to a file name - instead of stdout.

Pi4, 2GB, RP camera, PiOS Bullseye 32bit up to date.

davidplowman, 2021-11-13:
Never crossed my mind that writing to stdout is a thing for libcamera-still / libcamera-jpeg. I'll put it on the list of little things to tidy up. Thanks for pointing it out!

davidplowman, 2021-11-15:
This fix has been merged now. You can get it either by building from source, or we'll be updating the apt packages in the coming days.

naushir, 2021-11-18:
Closing, as this has now been fixed.  If there are still any further issue, feel free to re-open.

closed 2021-11-18 by naushir

aiot-maker, 2022-01-19:
I wonder if this issue has been fixed indeed. I jut downloaded the latest Raspberry Pi OS Lite (as per imager tool the release date is 2021-1030). I event tried to update/upgrade Raspberry PI OS afterwards (sudo apt update, sudo apt full-upgrade). Camera capture works perfectly except I am unable to redirect the image as a buffer to stdout using libcamera-still -n -o -

naushir, 2022-01-20:
The updated libraries are not yet available in either the OS image or in apt.  This will be happening imminently.  In the meantime, if you need this functionality, you will have to rebuilt the libcamera-apps and libcamera libraries manually.  Instructions can be found [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps) to do this.

aiot-maker, 2022-01-20:
Fully understood naushir. Thank you so much for your quick reply. I really appreciated.