raspberrypi/rpicam-apps #140: How to restore "raspicam" apps?

SuhaylZhao, 2021-11-14:
Hi,
You know, before the bullseye system, the camera applications were raspistill,raspivid, but after the bullseye was released, the original camera applications raspistill,raspivid all failed, and picamera, a python library, could not be used. I want to restore the original camera application so that the picamera library can continue to be used. What should I do? Any help would be greatly appreciated!

davidplowman, 2021-11-14:
Hi, there is a forum post here: https://forums.raspberrypi.com/viewtopic.php?t=323390&sid=65b91e146d233b5282246a4e917fe5f1 or alternatively users can continue to use Buster.

SuhaylZhao, 2021-11-14:
Hi，Thank you very much!!! I have succeeded!

davidplowman, 2021-11-14:
Great, thanks for letting us know.

closed 2021-11-14 by davidplowman