raspberrypi/rpicam-apps #141: App stop whit multiple img

ml50us, 2021-11-14:
I need to use -s or -k option to take multiple pict, as i do with raspistill:

libcamera-still -t 0 -k -n -o img%05d.jpg 

But app stop after the first img. It works with timelaps option.
Tks

davidplowman, 2021-11-14:
Ah, I hadn't realised that was the expect behaviour, but it makes sense. Thanks for pointing it out!

ml50us, 2021-11-14:
Thanks a lot, its really important for my project. Is it possible to go back to raspistill with bullseye?

davidplowman, 2021-11-14:
Hi, check out https://forums.raspberrypi.com/viewtopic.php?t=323390 or you can of course stay with Buster.

In general image quality and control of the camera should be better with the new stack, but of course if there are particular issues preventing people from using it (other than ones we already know about!) then we'd appreciate hearing about it.

ml50us, 2021-11-14:
Perfect! Thanks for your time and for  this great project.

davidplowman, 2021-11-15:
This fix has been merged now. You can get it either by building from source, or we'll be updating the apt packages in the coming days.

naushir, 2021-11-18:
Closing, as this has now been fixed.  If there are still any further issue, feel free to re-open.

closed 2021-11-18 by naushir

ml50us, 2021-11-27:
it works now with keypress -k, but with signal -s , it starts well for the first pict and then the process never stop. 

libcamera-still --version
libcamera-apps build: 6145daf735fc-intree 27-11-2021 (07:06:36)
libcamera build: v0.0.0+3156-f4070274