raspberrypi/rpicam-apps #144: Not enough buffers provided by V4L2VideoDevice on RPi3B

katyaxm, 2021-11-15:
Hey
I've used libcamera-apps on RaspberryPi4B with no problems, but now that I am trying to run the same thing on RaspberryPi3B I'm getting the following output which says there's not enough memory provided by V4L2VideoDevice:
```./libcamera-vid -t 500000 -o x.h264 --width 1920 --height 1080
Preview window unavailable
[0:00:38.313611546] [574]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:00:39.112628265] [574]  INFO Camera camera.cpp:937 configuring streams: (0) 1920x1080-YUV420
[0:00:39.113816546] [579]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 3217.33 (best 3217.33)
[0:00:39.114026234] [579]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1500 (best 1500)
[0:00:39.114364359] [579]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 4297.33 (best 1500)
[0:00:39.114585348] [579]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 6593.33 (best 1500)
[0:00:39.114927275] [579]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 7593.33 (best 1500)
[0:00:39.115175921] [579]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 2717.33 (best 1500)
[0:00:39.115281754] [579]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1000 (best 1000)
[0:00:39.115495348] [579]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 3797.33 (best 1000)
[0:00:39.115815869] [579]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 6093.33 (best 1000)
[0:00:39.116914671] [579]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1920x1080-pBAA
[0:00:39.174797848] [579] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video14[20:cap]: Not enough buffers provided by V4L2VideoDevice
[0:00:39.179206129] [579] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers
```

 If I lower the width and height to 1280x720 It'll work with the following output: 
```./libcamera-vid -t 5000 -o x.h264 --width 1280 --height 720
Preview window unavailable
[0:05:32.120688361] [597]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:05:32.592355757] [597]  INFO Camera camera.cpp:937 configuring streams: (0) 1280x720-YUV420
[0:05:32.593463465] [602]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 3467.33 (best 3467.33)
[0:05:32.593667892] [602]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1750 (best 1750)
[0:05:32.593766486] [602]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 2900.33 (best 1750)
[0:05:32.593862632] [602]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 4593.33 (best 1750)
[0:05:32.593968465] [602]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 5593.33 (best 1750)
[0:05:32.594068049] [602]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 2967.33 (best 1750)
[0:05:32.594163049] [602]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1250 (best 1250)
[0:05:32.594256017] [602]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 2400.33 (best 1250)
[0:05:32.594349611] [602]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 4093.33 (best 1250)
[0:05:32.595439819] [602]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1920x1080-pBAA
[0:05:32.652724819] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Sharpness = 1.000000
[0:05:32.653003153] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Saturation = 1.000000
[0:05:32.653121278] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Contrast = 1.000000
[0:05:32.653257163] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: Brightness = 0.000000
[0:05:32.653363153] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:05:32.653985340] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: NoiseReductionMode = 1
[0:05:32.655017840] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AeMeteringMode = 0
[0:05:32.656151694] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AeExposureMode = 0
[0:05:32.656324194] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: ExposureValue = 0.000000
[0:05:32.656595757] [607]  INFO IPARPI raspberrypi.cpp:621 Request ctrl: AwbMode = 0
[0:05:32.664247997] [602]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:05:32.664372372] [602]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:05:32.772272007] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (7 left)
[0:05:32.819013934] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (6 left)
[0:05:32.860382580] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (5 left)
[0:05:32.887909871] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (4 left)
[0:05:32.919877424] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (3 left)
[0:05:32.953200444] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (2 left)
[0:05:32.987983778] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (1 left)
[0:05:33.021247267] [602]  INFO RPI raspberrypi.cpp:1635 Dropping frame at the request of the IPA (0 left)
```

This problem did not occur in RaspberryPi4. I have dedicated 128M to gpu_mem. Also there's no problem when using raspivid, it only occurs when using libcamera-vid and camera resolution of 1920x1080.

davidplowman, 2021-11-15:
Hi, I've just tried this on a Pi 3 and it seemed to work OK. Can you perhaps say:

* What OS you've installed (`uname -a` and `vcgencmd version` might be helpful). I used the latest version of Bullseye from the Raspberry Pi Imager tool.
* What version does `libcamera-hello --version` report?
* Have you changed anything in your /boot/config.txt? I believe the default gpu_mem is only 76MB when running Bullseye on a Pi 3 (the system needs less because the camera stack no longer uses it), Though I tried bumping it all the way up to 256 and it still works.
* I notice it reports "Preview window unavailable". Is it running headless, or over ssh or is there anything else that would help me to do the same thing as you?
* Giving the full commands you're using would also be helpful. Ah, I see you've reported the same thing on the forum with the commands given. No problem, then!

Thanks!

Rakhmanov, 2021-11-21:
Same issue attaching versions

```
uname -a
Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

```

```
libcamera-vid  --inline --listen -o tcp://0.0.0.0:8554
[0:46:31.874985905] [7107]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:46:32.026030499] [7107]  INFO Camera camera.cpp:937 configuring streams: (0) 640x480-YUV420
[0:46:32.026309457] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2354 (best 2354)
[0:46:32.026453154] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 2136.67 (best 2136.67)
[0:46:32.026541019] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1787 (best 1787)
[0:46:32.026622894] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 1500 (best 1500)
[0:46:32.026704717] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 2500 (best 1500)
[0:46:32.026784144] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1854 (best 1500)
[0:46:32.026862373] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1636.67 (best 1500)
[0:46:32.026941435] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1287 (best 1287)
[0:46:32.027021435] [7112]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 1000 (best 1000)
[0:46:32.029519922] [7112]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 640x480-pBAA
[0:46:32.056812500] [7112]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:46:32.056879427] [7112]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:46:32.122507488] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:46:32.184187637] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:46:32.214363545] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:46:32.246249868] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:46:32.277151193] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:46:32.308807725] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:46:32.340154569] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:46:32.371539487] [7112]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```

```
libcamera-hello --version
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:17)
libcamera build: v0.0.0+3156-f4070274

```

```
vcgencmd version
Oct 29 2021 10:50:02
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start_x)

```

davidplowman, 2021-11-22:
@Rakhmanov Thanks for attaching this extra information. I can't see where you are getting a "not enough V4L2 buffers" error like the OP, so it's not immediately clear to me whether this is the same problem or something else. Can you perhaps clarify? Thanks!

miztroh, 2021-11-25:
@davidplowman This is what I get using an RPi 3B with the camera module attached when attempting to run libcamera-hello:

[image]

Seems to be along the lines of what the OP is describing.  This is with a fresh install of RPi OS Bullseye.

davidplowman, 2021-11-25:
@miztroh Thanks for this information. I'm struggling to reproduce the same thing, here's what I've tried on a Pi 3B:

* Installed a fresh Bullseye (full, not Lite) using the Raspberry Pi imager tool. I let it run through its updates when it started.
* Re-enabled Glamor using `sudo raspi-config` and then let it reboot.
* Once it had restarted I ran `libcamera-hello` which worked fine.

Again, I'm not convinced this is the same problem as the OP, after all, the error message is different, and I'm guessing you don't have the same symptom that it works with smaller frame sizes but not larger ones? If you're still seeing this problem, could I perhaps ask you to post a new bug report? Could I also please ask for the following information in the new bug report:

* Report the output of `uname -a`.
* Report the output of `vcgencmd version`.
* Report the output of `libcamera-hello --version`.
* Say which camera you are using. Is it an official Raspberry Pi module or if not where did it come from? Do you have a link?
* Give the full command you are running which fails.
* Include the full console output of that command.
* Say whether you are logged directly into the Pi with an attached keyboard or screen, or using ssh. If you are using ssh, are you using X forwarding?
* Have you changed anything in the /boot/config.txt file, and if so, what?
* Does the end of `dmesg` show any obvious errors?

Thanks very much!

lff5, 2021-11-25:
I am experiencing the same error.
official RPi HQ camera on Pi 4b running Buster 64-bit version.

in **/boot/config.txt** I replaced dtoverlay. Before that it did not detect camera. 
dtoverlay=imx477
#dtoverlay=vc4-kms-v3d
After changing that USB keyboard does not work any more. I saw it on dmesg though.

GPU memory is set to 256

**libcamera-hello --version**
```
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:45:05)
libcamera build: v0.0.0+3156-f4070274
```

**uname -a**
`Linux raspberry 5.10.52-v8+ #1440 SMP PREEMPT Tue Jul 27 09:56:40 BST 2021 aarch64 GNU/Linux
`
f
**vcgencmd version**
```
Jul 21 2021 16:22:36 
Copyright (c) 2012 Broadcom
version 6a796bb0062a6c75191c57cba1c13f9300076d02 (clean) (release) (start_x)
```

**v4l2-ctl --list-devices**
```
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/video1
```

Running locally, display attached:
**libcamera-still**
```
libEGL warning: DRI2: failed to authenticate
[0:16:22.662715342] [2155]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:16:22.680540714] [2165]  WARN CameraSensor camera_sensor.cpp:197 'imx477 10-001a': Recommended V4L2 control 0x009a0922 not supported
[0:16:22.680628509] [2165]  WARN CameraSensor camera_sensor.cpp:249 'imx477 10-001a': The sensor kernel driver needs to be fixed
[0:16:22.680661953] [2165]  WARN CameraSensor camera_sensor.cpp:251 'imx477 10-001a': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:16:22.681476124] [2165]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:16:22.681548123] [2165]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:16:22.681577974] [2165]  WARN CameraSensor camera_sensor.cpp:414 'imx477 10-001a': Failed to retrieve the camera location
[0:16:22.716744355] [2155]  INFO Camera camera.cpp:937 configuring streams: (0) 2028x1520-YUV420
[0:16:22.716908000] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt RG10 Score: 3956.22 (best 3956.22)
[0:16:22.716988665] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt RG12 Score: 1387 (best 1387)
[0:16:22.717022183] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt RG12 Score: 500 (best 500)
[0:16:22.717053516] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt RG12 Score: 1583.84 (best 500)
[0:16:22.717084886] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt pRAA Score: 3456.22 (best 500)
[0:16:22.717116108] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt pRCC Score: 887 (best 500)
[0:16:22.717146181] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt pRCC Score: 0 (best 0)
[0:16:22.717176218] [2165]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt pRCC Score: 1083.84 (best 0)
[0:16:22.717298105] [2165]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
[0:16:22.754567080] [2165] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video14[18:cap]: Not enough buffers provided by V4L2VideoDevice
[0:16:22.758382795] [2165] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers
```

[... 20 lines cut ...]
[0:04:45.550889449] [1678]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt RG12 Score: 1583.84 (best 500)
[0:04:45.550921207] [1678]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt pRAA Score: 3456.22 (best 500)
[0:04:45.550954002] [1678]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt pRCC Score: 887 (best 500)
[0:04:45.550984612] [1678]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt pRCC Score: 0 (best 0)
[0:04:45.551015352] [1678]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt pRCC Score: 1083.84 (best 0)
[0:04:45.551153550] [1678]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
Camera streams configured
Buffers allocated and mapped
Viewfinder setup complete
Requests created
[0:04:45.590440952] [1678] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video14[17:cap]: Not enough buffers provided by V4L2VideoDevice
[0:04:45.594731076] [1678] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
```

dmesg:
```
[  475.687471] cma: cma_alloc: alloc failed, req-size: 1140 pages, ret: -12
[  475.687506] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 4669440 failed
[  590.817398] cma: cma_alloc: alloc failed, req-size: 1140 pages, ret: -12
[  590.817434] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 4669440 failed
```

davidplowman, 2021-11-25:
Thanks very much for this information. This does look like the same kind of problem as the OP, and it certainly seems like you're running out of CMA memory. A few things to check:

* Are you doing anything in your `/boot/config.txt` or `/boot/cmdline.txt` to alter the amount of CMA allocation?
* Could you post the contents of `/proc/meminfo`? In particular the amount of CMA (at the bottom) might be illuminating.
* The GPU memory of 256MB seems larger than you should need when using libcamera. On Bullseye we set this as low as 76MB by default.

Thanks!

lff5, 2021-11-25:
I got it to work! Almost entire day wasted because of no option to fall back to picamera...

@davidplowman you suggested to re-enable Glamor.

When I did, this added following line to /boot/config.txt:
`dtoverlay=vc4-fkms-v3d`
I had removed it earlier when adding dtoverlay=imx477 because i thought there cant be two parameters with identical name.

I tested with original dtoverlay
`dtoverlay=vc4-kms-v3d`
and it worked also.

# Conclusion:
to solve `libcamera ERROR: *** no cameras available ***` add camera dtoverlay, e.g. `dtoverlay=imx477` to /boot/config.txt

If there is no `dtoverlay=vc4-kms-v3d` or  `dtoverlay=vc4-fkms-v3d` defined in the same file, there will be` Not enough buffers provided by V4L2VideoDevice` error.

For some reason usb keyboard takes ages to start working (it started occuring when i added dtoverlay=imx477).

davidplowman, 2021-12-01:
Hi everyone, I'm going to close this bug because there doesn't appear to be any further information on the original problem, and various other bugs have crept in, at least some of which appear to be quite different and I'm afraid that I'm struggling a bit to understand what are still live issues and what are not.

But I'd like to encourage everyone who is still experiencing a problem to please file a new bug report of your own describing your issue. Please include all the information mentioned here: https://github.com/raspberrypi/libcamera-apps/issues/144#issuecomment-979076466 . Also the result of `grep -i cma /proc/meminfo` would be helpful if there appear to be buffer or memory problems.

Thanks very much!

closed 2021-12-01 by davidplowman

lff5, 2021-12-02:
Followup

I removed dtoverlay=vc4-kms-v3d again to reproduce the error
```
[0:16:22.754567080] [2165] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video14[18:cap]: Not enough buffers provided by V4L2VideoDevice
[0:16:22.758382795] [2165] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers
```

Note. GPU memory is still set to 256 MB.
/proc/meminfo :
```
MemTotal:        3832956 kB
MemFree:         3210736 kB
MemAvailable:    3473668 kB
Buffers:           37328 kB
Cached:           278020 kB
SwapCached:            0 kB
Active:           100976 kB
Inactive:         431372 kB
Active(anon):        932 kB
Inactive(anon):   225672 kB
Active(file):     100044 kB
Inactive(file):   205700 kB
Unevictable:          60 kB
Mlocked:              60 kB
SwapTotal:        102396 kB
SwapFree:         102396 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:        217208 kB
Mapped:           136128 kB
Shmem:              9604 kB
KReclaimable:      27876 kB
Slab:              53452 kB
SReclaimable:      27876 kB
SUnreclaim:        25576 kB
KernelStack:        4192 kB
PageTables:         5816 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     2018872 kB
Committed_AS:    1254292 kB
VmallocTotal:   262930368 kB
VmallocUsed:        9284 kB
VmallocChunk:          0 kB
Percpu:              768 kB
CmaTotal:          65536 kB
CmaFree:           54920 kB
```

davidplowman, 2021-12-02:
Hi, thanks for this information. It's clear that this configuration has insufficient CMA memory so libcamera is going to struggle. You will need to put back either `dtoverlay=vc4-kms-v3d` or `dtoverlay=vc4-fkms-v3d`. Setting gpu_mem does not help since libcamera does not use it, in fact it is almost certainly harmful so I would strongly recommend removing it. Hope that helps!

PinkFreud, 2022-01-17:
I received the same buffer error as @katyaxm on a Zero W running bullseye, which led me to this bug report.

I had initially set gpu_mem to 256M while troubleshooting a camera I was having some difficulty with (yep, half the Zero W's memory, ouch!).  Now that I have the camera recognized, and verified it was able to produce an image on buster, switching to an sd card with bullseye installed and libcamera gave me the buffer error.

At @davidplowman's suggestion to check /proc/meminfo for Cma memory:
```
CmaTotal:           8192 kB
CmaFree:            3676 kB
```

... which seemed a bit low.  Since libcamera doesn't use the gpu for the cam, I dropped gpu_memory back to the default of 76M and rebooted.  Checking /proc/meminfo a second time:
```
CmaTotal:         262144 kB
CmaFree:          153460 kB
```

... and sure enough, libcamera-hello now functions - even the preview window works w/ a monitor attached.

It seems like this error could be triggered by merely having too much memory allocated to the gpu, leaving too little free for Cma.

kevleyski, 2022-05-26:
Hi I've got a related problem when using IMX477 and touch screen at same time
Seem if I dtoverlay=vc4-kms-v3d then the camera works, but then the touch screen wont
If I remove that then the touchscreen works but its the no cameras available issue

addohm, 2023-05-20:
> Hi, if useful to others - I had a related problem when using IMX477 and touch screen at same time Seem if I dtoverlay=vc4-kms-v3d then the camera works, but then the touch screen wont If I remove that then the touchscreen works but its the no cameras available issue
> 
> I can however use dtoverlay=vc4-fkms-v3d and get both - but then you see
> 
> [0:00:35.383195367] [2235] INFO Camera camera.cpp:1029 configuring streams: (0) 1280x720-YUV420 [0:00:35.385228182] [2269] INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC [0:00:36.862222014] [2269] ERROR V4L2 v4l2_videodevice.cpp:1852 /dev/video1[20:cap]: Failed to start streaming: Input/output error ERROR: *** failed to start camera ***

I’m stuck at the same point.  Any solutions yet?

steverinard, 2023-06-02:
I am having the same problem with touchscreen and camera.

With dtoverlay=vc4-kms-v3d, the camera works (imx219) but no touchscreen. 
With dtoverlay=vc4-fkms-v3d the touchscreen works but no camera.

zinnakhan, 2023-06-29:
C