raspberrypi/rpicam-apps #149: Vertical or horizontal flip or rotation transform has no effect on Raspberry Pi Zero

pastcompute, 2021-11-18:
When I run libcamera-still (or libcamera-vid) the vf and hf and rot options have no effect.
Example:
```
libcamera-still -o test.jpg -t 1000 --vf -v -n
```
Expected result: my image is flipped
Actual result: image is not flipped
Running same command with no vf option results in same (unflipped) picture.
If I remove width, height, etc. the size adjust but otherwise same unflipped result.

I have a raspberry Pi zero with the following versions installed:
```
ii  libcamera-apps                       0~git20211021+2a38ae9-1~bpo10~0         armhf        libcamera-apps
ii  libcamera0:armhf                     0~git20211020+f4070274-1~bpo10~0        armhf        complex camera support library
```

Output collected because of `-v` option:
```
$ libcamera-still -o test.jpg -t 1000 --width 1920 --height 1080 --vf -v -n
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 1000
    width: 1920
    height: 1080
    output: test.jpg
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: vflip
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 0
    framestart: 0
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 320
    thumbnail height: 240
    thumbnail quality: 70
    latest:
    immediate 0
Running without preview window
Opening camera...
[0:38:23.358660844] [5413]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
Viewfinder size chosen is 1296x728
Stream configuration adjusted
[0:38:23.657444953] [5413]  INFO Camera camera.cpp:937 configuring streams: (0) 1296x728-YUV420
[0:38:23.660357925] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 3468.66 (best 3468.66)
[0:38:23.663865891] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1751.33 (best 1751.33)
[0:38:23.665637873] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 2901.66 (best 1751.33)
[0:38:23.667379857] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 4648.66 (best 1751.33)
[0:38:23.670980822] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 5648.66 (best 1751.33)
[0:38:23.672533807] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 2968.66 (best 1751.33)
[0:38:23.672861804] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1251.33 (best 1251.33)
[0:38:23.675272780] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 2401.66 (best 1251.33)
[0:38:23.675619777] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 4148.66 (best 1251.33)
[0:38:23.678467749] [5415]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1920x1080-pBAA
Camera streams configured
Buffers allocated and mapped
Viewfinder setup complete
Requests created
[0:38:23.812868449] [5415]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:38:23.814194437] [5415]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:38:23.815254426] [5415]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:38:23.816092418] [5415]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
Camera started!
[0:38:23.925463360] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:38:23.981983814] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:38:24.021464432] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:38:24.064321017] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:38:24.096936702] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:38:24.127681404] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:38:24.167842016] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:38:24.212142587] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
Viewfinder frame 0
Viewfinder frame 1
Viewfinder frame 2
Viewfinder frame 3
Viewfinder frame 4
Viewfinder frame 5
Viewfinder frame 6
Viewfinder frame 7
Viewfinder frame 8
Viewfinder frame 9
Viewfinder frame 10
Viewfinder frame 11
Viewfinder frame 12
Viewfinder frame 13
Viewfinder frame 14
Viewfinder frame 15
Viewfinder frame 16
Viewfinder frame 17
Viewfinder frame 18
Viewfinder frame 19
Camera stopped!
Tearing down requests, buffers and configuration
Configuring still capture...
Stream configuration adjusted
[0:38:24.958539370] [5413]  INFO Camera camera.cpp:937 configuring streams: (0) 1920x1080-YUV420
[0:38:24.960728349] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 3217.33 (best 3217.33)
[0:38:24.965646302] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1500 (best 1500)
[0:38:24.967967279] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 4297.33 (best 1500)
[0:38:24.971583244] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 6593.33 (best 1500)
[0:38:24.973324227] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 7593.33 (best 1500)
[0:38:24.975067211] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 2717.33 (best 1500)
[0:38:24.976853193] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1000 (best 1000)
[0:38:24.978614176] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 3797.33 (best 1000)
[0:38:24.981492148] [5415]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 6093.33 (best 1000)
[0:38:24.986980095] [5415]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1920x1080-pBAA
Camera streams configured
Buffers allocated and mapped
Still capture setup complete
Requests created
[0:38:25.093138069] [5415]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
Camera started!
[0:38:25.235386694] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:38:25.383238265] [5415]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
Camera stopped!
Still capture image received
Exposure time: 49789
Ag 4 Dg 1 Total 4
Thumbnail dimensions are 320 x 240
Thumbnail size 10222
JPEG size is 413084
EXIF data len 266
Saved image 1920 x 1080 to file test.jpg
Closing Libcamera application(frames displayed 19, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
```

Please let me know what additional info might be needed

davidplowman, 2021-11-18:
Thanks for reporting this. It seems the Linux kernel ov5647 driver simply doesn't support mirrors and flips, so we'll have to look into that.

pastcompute, 2021-11-20:
One interesting thing about this is that when I had the older raspberry pi OS with `raspivid`, I didn't need to flip anything, it only is required with bullseye and libcamera, so I guess the drivers must be different

davidplowman, 2021-11-20:
Yes, it is a completely different driver. The new one is an open source driver in the Linux kernel: https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/media/i2c/ov5647.c
The old driver is Broadcom's proprietary driver that runs on the GPU. You would need an NDA with Broadcom to have any chance of seeing it!

davidplowman, 2021-11-23:
We've fixed this in that driver file now so it should be available in the next release.

davidplowman, 2021-11-25:
I think there's nothing else to do here now, the next release will contain the fix, so closing.

closed 2021-11-25 by davidplowman

Hsad1644, 2022-01-19:
@davidplowman is there a way to implement the driver that you've [linked](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/media/i2c/ov5647.c) to? Waiting on the next buster release too!

6by9, 2022-01-19:
You can use ```sudo rpi-update``` to grab the latest kernel which includes the fix. There is a change with regard the use of the Media Controller API though, so you either need to rebuild libcamera and libcamera-apps, or edit /boot/config.txt to:
- remove the line ```camera_auto_detect=1```
- add the line ```dtoverlay=ov5647,media-controller=0```

The Bullseye repos should be updated very soon with an updated kernel and libcamera that include all the latest changes.