raspberrypi/rpicam-apps #151: libcamera-vid yuv420 output corrupt and missing frames

dustinkerstein, 2021-11-21:
When outputting raw YUV via the following command:

`libcamera-vid --width 4056 --height 3040 -t 10000 --gain 1 --awbgains 1,1 --shutter 20000 --denoise cdn_fast --save-pts timestamps.txt --codec yuv420 -o /dev/shm/out.yuv`

The output YUV file is partially corrupt, and prevents me from fully encoding ffmpeg: 

h264_v4l2m2m
```
Truncating packet of size 18495360 to 18057601
[rawvideo @ 0x1b40d90] Packet corrupt (stream = 0, dts = 99).
/dev/shm/out.yuv: corrupt input packet in stream 0
[rawvideo @ 0x1b4b720] Invalid buffer size, packet size 18057600 < expected frame_size 18495360
Error while decoding stream #0:0: Invalid argument
No more output streams to write to, finishing.
[h264_v4l2m2m @ 0x1b4d0f0] ff_v4l2_buffer_buf_to_avpkt
[h264_v4l2m2m @ 0x1b4d0f0] Dq event 2
Not writing 'clli' atom. No content light level info.
Not writing 'mdcv' atom. Missing mastering metadata.
frame=   99 fps= 15 q=-0.0 Lsize=    2140kB time=00:00:09.80 bitrate=1788.5kbits/s speed=1.45x
video:2138kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.058731%
Input file #0 (/dev/shm/out.yuv):
  Input stream #0:0 (video): 100 packets read (1849098240 bytes); 99 frames decoded;
  Total: 100 packets (1849098240 bytes) demuxed
Output file #0 (/dev/shm/out_1.mp4):
  Output stream #0:0 (video): 99 frames encoded; 100 packets muxed (2189660 bytes);
  Total: 100 packets (2189660 bytes) muxed
[AVIOContext @ 0x1b4ddd0] Statistics: 2 seeks, 12 writeouts
[AVIOContext @ 0x1b49870] Statistics: 1849098240 bytes read, 0 seeks
```
libx264 (note different number of frames/packets encoded/muxed)
```
Truncating packet of size 18495360 to 18057601
[rawvideo @ 0x1c52d90] Packet corrupt (stream = 0, dts = 99).
/dev/shm/out.yuv: corrupt input packet in stream 0
[rawvideo @ 0x1c5d720] Invalid buffer size, packet size 18057600 < expected frame_size 18495360
Error while decoding stream #0:0: Invalid argument
No more output streams to write to, finishing.
Not writing 'clli' atom. No content light level info.
Not writing 'mdcv' atom. Missing mastering metadata.
frame=   99 fps=4.1 q=-1.0 Lsize=    6102kB time=00:00:09.60 bitrate=5206.9kbits/s speed= 0.4x
video:6100kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.023308%
Input file #0 (/dev/shm/out.yuv):
  Input stream #0:0 (video): 100 packets read (1849098240 bytes); 99 frames decoded;
  Total: 100 packets (1849098240 bytes) demuxed
Output file #0 (/dev/shm/out_1.mp4):
  Output stream #0:0 (video): 99 frames encoded; 99 packets muxed (6246831 bytes);
  Total: 99 packets (6246831 bytes) muxed
[AVIOContext @ 0x1c606f0] Statistics: 2 seeks, 27 writeouts
```

Interestingly if I use auto exposure (or use awb auto): 

`libcamera-vid --width 4056 --height 3040 -t 10000 --gain 1 --awbgains 1,1 --denoise cdn_fast --save-pts timestamps.txt --codec yuv420 -o /dev/shm/out.yuv`

The corruption happens earlier (maybe due to the skipped frames from the ISP doing its auto exposure calcs). When encoding with libx264 ffmpeg you can see the corruption starts at dts = 93:

```
Truncating packet of size 18495360 to 16963201e=00:00:03.80 bitrate=7726.3kbits/s speed=0.265x
[rawvideo @ 0x120cd70] Packet corrupt (stream = 0, dts = 93).
/dev/shm/out.yuv: corrupt input packet in stream 0
[rawvideo @ 0x1217700] Invalid buffer size, packet size 16963200 < expected frame_size 18495360
Error while decoding stream #0:0: Invalid argument
```

Let me know if I can get any further debug.

davidplowman, 2021-11-21:
Hi, so there are a couple of things to check here.

Firstly, there are certain alignment constraints in the hardware that produces the YUV420 output image. The Y rows must be aligned to multiples of 64 bytes, U and V to multiples of 32. In practice this means your image will _appear_  to have the size 4096x3040, though there will be 40 pixels of padding down the right hand edge.

Thus I would expect that each frame should be 4096x3040x3/2 = 18677760 bytes in size. It would be interesting to check if your output file is a multiple of this number. My recollection is that a frame is written out by a single call to `fwrite`, so apart from the output filesystem filling up it shouldn't really be possible to get "corruption" as such, only frames that are entirely dropped. The timestamps.txt file should reveal exactly which frames are missing. Note that you can also use the `--segment 1` argument to write each frame to separate file, which should make the size of each frame very obvious.

If things seem not to be as I have described, or if we can provide other assistance, please let us know.

dustinkerstein, 2021-11-22:
Thanks for the quick reply and the background info. When I add `--segment` and run:

`libcamera-vid --width 4056 --height 3040 -t 1000 --segment 1 --gain 1 --awbgains 1,1 --shutter 20000 --denoise cdn_fast --save-pts timestamps.txt --codec yuv420 -o /dev/shm/camera%04d.yuv`

9 frames are captured:
```
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0000.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0001.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0002.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0003.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0004.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0005.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0006.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0007.yuv
-rw-r--r--  1 pi   pi   18677760 Nov 22 00:12 camera0008.yuv
```
FFmpeg doesn't seem to support encoding a video from individual yuv files (unlike jpeg/tiff/etc) so I tried again without `--segment` and confirmed that the non-segmented output file byte size divided by 18677760 = 9 frames. So I then ran `ffmpeg -y -f rawvideo -pix_fmt yuv420p -s:v 4096x3040 -r 10 -i /dev/shm/camera.yuv -filter:v scale=1920:1080 -c:v h264_v4l2m2m -b:v 25M -v verbose ~/camera.mp4` (ie. 4096 rather than 4056) and that encoded without error. So that seems to have been the issue. Though, I did again notice that discrepancy between the encoded/muxed frames in the FFmpeg output:
```
Input file #0 (/dev/shm/camera.yuv):
  Input stream #0:0 (video): 9 packets read (168099840 bytes); 9 frames decoded;
  Total: 9 packets (168099840 bytes) demuxed
Output file #0 (/home/pi/camera.mp4):
  Output stream #0:0 (video): 9 frames encoded; 10 packets muxed (29386 bytes);
  Total: 10 packets (29386 bytes) muxed
```
While I'm able to play this video in VLC, when I attempt to extract the frames with FFmpeg, something seems a bit wrong:
```
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x558bf63fe0] st: 0 edit list: 1 Missing key frame while searching for timestamp: 0
[h264 @ 0x558bf65180] no frame!
[h264 @ 0x558bf65180] Reinit context to 1920x1088, pix_fmt: yuv420p
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'camera.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
  Duration: 00:00:00.90, start: 0.000000, bitrate: 269 kb/s
    Stream #0:0(und): Video: h264 (High), 1 reference frame (avc1 / 0x31637661), yuv420p(left), 1920x1080 (1920x1088), 261 kb/s, 11.11 fps, 10 tbr, 10240 tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandler
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> png (native))
Press [q] to stop, [?] for help
[h264 @ 0x558bf6c180] no frame!
[h264 @ 0x558bf9e2a0] Reinit context to 1920x1088, pix_fmt: yuv420p
Error while decoding stream #0:0: Invalid data found when processing input
[graph 0 input from stream 0:0 @ 0x558bf6c6a0] w:1920 h:1080 pixfmt:yuv420p tb:1/10240 fr:10/1 sar:0/1
[auto_scaler_0 @ 0x558c34a4c0] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x558c348fb0] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
[swscaler @ 0x558c34b260] No accelerated colorspace conversion found from yuv420p to rgb24.
[auto_scaler_0 @ 0x558c34a4c0] w:1920 h:1080 fmt:yuv420p sar:0/1 -> w:1920 h:1080 fmt:rgb24 sar:0/1 flags:0x4
Output #0, image2, to 'camera_%04d.png':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
    Stream #0:0(und): Video: png, 1 reference frame, rgb24(left), 1920x1080, q=2-31, 200 kb/s, 10 fps, 10 tbn, 10 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      encoder         : Lavc58.91.100 png
[AVIOContext @ 0x558c370090] Statistics: 0 seeks, 1 writeouts
[AVIOContext @ 0x558bfcb670] Statistics: 0 seeks, 1 writeouts
    Last message repeated 3 times
No more output streams to write to, finishing.
[AVIOContext @ 0x558bfcb670] Statistics: 0 seeks, 1 writeouts
    Last message repeated 3 times
frame=    9 fps=0.0 q=-0.0 Lsize=N/A time=00:00:00.90 bitrate=N/A speed=2.41x
video:399kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Input file #0 (camera.mp4):
  Input stream #0:0 (video): 10 packets read (29386 bytes); 9 frames decoded;
  Total: 10 packets (29386 bytes) demuxed
Output file #0 (camera_%04d.png):
  Output stream #0:0 (video): 9 frames encoded; 9 packets muxed (409083 bytes);
  Total: 9 packets (409083 bytes) muxed
[AVIOContext @ 0x558bf19d50] Statistics: 30276 bytes read, 0 seeks
```
I think this may actually be a `h264_v4l2m2m` bug, as when I encode the `camera.yuv` file with libx264, I do not see this discrepancy in the encoded/muxed frames (see the log in the first post), and I am able to then extract the frames without error from the encoded file.

So, 4096 is the answer for the missing frame. I'm a little curious if that's why the Nvidia/RidgeRun folks decided to use 4032 in their IMX477 driver, as 4032 is divisible by 32. I can definitely deal with cropping those padding pixels, but maybe libcamera-vid could output some kind of debug message that warns about the added padding for folks not accustomed with working with yuv data.

As for the encoded/muxed frames issue with `h264_v4l2m2m`, should I create a new issue for that? Are you able to replicate?

Thanks again for your help.

dustinkerstein, 2021-11-22:
And one more thing, shouldn't `-t 1000` at 10fps create 10 frames rather than 9?

davidplowman, 2021-11-22:
Probably best to post a new bug at https://github.com/raspberrypi/linux - I don't thing anyone who monitors this list is likely to know very much about it.

As regards the number of frames, it times from the moment the "pushes" the camera's start button and it's quite usual for a camera to take "a moment or two" to get going. So I think some variability here is to be expected.

dustinkerstein, 2021-11-23:
kk, I'll post a new issue over there later today.

Regarding the missing frame - it's very consistently 1 frame short. Not too big of a deal. But it might be nice to consider adding a `num-buffers` parameter for folks who need to control the exact number of frames. That would be useful for libcamerasrc for Gstreamer as well.

dustinkerstein, 2021-11-23:
Feel free to close this ticket, or keep it open if you want to use it to track anything (maybe that warning about the pixel padding).

closed 2021-11-23 by davidplowman

dustinkerstein, 2021-11-25:
I added an issue for the muxing/encoder issue here - https://github.com/raspberrypi/linux/issues/4734