raspberrypi/rpicam-apps #152: libcamera-still signal and keypress modes don't allow for capturing multiple frames like raspistill

dustinkerstein, 2021-11-21:
We had an app built around raspistill's ability to capture frames quickly when SIGUSR1 was received. This worked great. It seems like libcamera-still has similar functionality, but it quits upon the first received signal (or keypress). I'm sure I could hack the source, but I was just curious, is that the intended behavior or should it have parity with raspistill?

davidplowman, 2021-11-21:
Hi, thanks for reporting this. Actually we have already fixed this problem here: https://github.com/raspberrypi/libcamera-apps/pull/143
We'll make sure it gets into the next update of our apt repositories, probably "in a few days".

dustinkerstein, 2021-11-21:
Awesome! Thanks. I'll go ahead and close this issue.

closed 2021-11-21 by dustinkerstein

byWulf, 2022-01-16:
@davidplowman Hey David, can you aproximately say how long it will take until this fix is in the apt repository available? I am still getting the version vom 2021-10-21 there :/ Thanks!

davidplowman, 2022-01-16:
Hi again, yes I'm sorry this has been taking so long. We would have updated the apt repositories ages ago but for the problem with the Bullseye kernel being incompatible with recent libcamera code (and in turn libcamera-apps).

However, we are currently testing our Bullseye update image which brings everything back in sync and will resolve all these problems. My expectation is quite strongly that it is likely to happen this week, though of course it's not finally my decision exactly when it gets the final go-ahead.