raspberrypi/rpicam-apps #158: CMA memory usage of libcamera-raw vs. libcamera-vid vs. v4l2-ctl

dustinkerstein, 2021-11-27:
I've been doing some stress testing in preparation for getting a 4 lane CSI driver for the IMX477 and ran across some interesting memory behavior. I had previously run into some CMA issues, which lead me to my current config: cma=512M@128M and gpu_mem=256, and since then I've been paying closer attention to CMA usage. I ran the following three commands:
```
v4l2-ctl -d /dev/video0 --stream-mmap --stream-count=100 --stream-to=/dev/shm/test.raw
```
```
libcamera-raw --width 4056 --height 3040 -t 10000 -o /dev/shm/test.raw
```
```
libcamera-vid --width 4096 --height 3040 -t 10000 --gain 1 --awbgains 1,1 --shutter 100000 --denoise cdn_fast --codec yuv420 -o /dev/shm/test.yuv
```
And came up with the below CMA memory usage:
```
v4l2-ctl - 84MB
libcamera-raw - 457MB
libcamera-vid - 366MB
```
The libcamera-apps seem to be using more buffers than v4l2-ctl, but libcamera-raw sticks out to me. Does that memory usage seem correct?

I am testing with an Arducam Mini IMX477 module + Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.63-v8+ Kernel running headless over ssh. Full system info here - https://pastebin.com/654vBepL
```
$ libcamera-hello --version
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:53:29)
libcamera build: v0.0.0+3156-f4070274
$ vcgencmd version
Oct 29 2021 10:47:33
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```

davidplowman, 2021-11-27:
Hi, an interesting question. There are some general issues about the system allocating buffers that can be returned to the application, but also having to allocate "hidden" buffers in case the user does not supply any. This does lead to over-allocation, but my colleague knows more about that so I would do better to leave that to him!

On the subject of `libcamera-raw`, all you really care about is the size of the raw buffers, not the output of the ISP. So you would probably get better behaviour with this:
```
libcamera-raw --rawfull -t 10000 -o /dev/shm/test.raw
```
It's forcing the system to allocate full size raw buffers (as before), but the ISP outputs will be small.

I guess `libcamera-vid` is better because it doesn't have to allocate those very large raw buffers for returning to the application, it only needs its internal "hidden" ones.

dustinkerstein, 2021-11-27:
`--rawfull` is indeed interesting. It uses 235MB CMA vs. 457MB (from above). 

Not sure I understand the difference, but obviously there is a different path being taken. Ah, maybe it's allocating fewer buffers - https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L234 - I haven't followed the whole path, but that line seems intriguing. 

Still, those CMA memory numbers are quite high, and pushing the limits even with the max CMA of 512MB. I'd love to find a way to set higher CMA, but I've come up empty on that front. Let me know if you have any thoughts there.

Feel free to close this issue if you don't feel there's anything actionable.

davidplowman, 2021-11-27:
The difference with `--rawfull` is that both cases will allocate the same large raw buffers which double as the output from the camera and the input to the ISP. The difference is that previously you were getting 12MP output buffers from the ISP whereas now you're getting (I think) just VGA.

Actually there has been some work in the buffer allocation area recently, so when we next update our apt packages you should see some improvement.

I'll probably close this now as I don't think there's anything we would plan to do here, but get back to us if there are more questions or problems. Thanks!

closed 2021-11-27 by davidplowman