raspberrypi/rpicam-apps #159: problem in signal mode

ml50us, 2021-11-27:
i try to run libcamera-still with -s option but the process run continuously until i send SIGUSR2 signal .
the process don't wait after the first image. 
is it normal ? 

libcamera-still --version
libcamera-apps build: 6145daf-intree 27-11-2021 (07:06:36)
libcamera build: v0.0.0+3156-f4070274

davidplowman, 2021-11-27:
Thanks for reporting this. I'll check it out as soon as I can.

davidplowman, 2021-11-28:
Yes, sorry about that. I've just pushed a fix here: https://github.com/raspberrypi/libcamera-apps/pull/160

davidplowman, 2021-11-29:
This has been merged now so I'll close this issue. We'll get it into our apt repository shortly, if the feature is more urgent then libcamera-apps can be built from source.

closed 2021-11-29 by davidplowman

ml50us, 2021-11-29:
Yes it works! Thanks