raspberrypi/rpicam-apps #161: Libcamera issue

thepahema, 2021-11-28:
Hello. I have a two yers old picamera Noir. In Raspbian Bullseye, when I run this command:

libcamera-still

I've got this:

[0:31:07.755058346] [6199]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:31:07.830120601] [6201] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0901 but control is not exposed by device /dev/video0
[0:31:07.830186640] [6201] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x00980911 but control is not exposed by device /dev/video0
[0:31:07.830217234] [6201] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0903 but control is not exposed by device /dev/video0
Stream configuration adjusted
[0:31:07.831536808] [6199]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[0:31:07.831989580] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y10  Score: 500 (best 500)
[0:31:07.832076490] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y12  Score: 500 (best 500)
[0:31:07.832117861] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y14  Score: 500 (best 500)
[0:31:07.832155270] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA10 Score: 1500 (best 500)
[0:31:07.832192141] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB10 Score: 1500 (best 500)
[0:31:07.832230476] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG10 Score: 1500 (best 500)
[0:31:07.832265754] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 500)
[0:31:07.832301459] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA81 Score: 2500 (best 500)
[0:31:07.832337627] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA12 Score: 500 (best 500)
[0:31:07.832373850] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB12 Score: 500 (best 500)
[0:31:07.832408573] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG12 Score: 500 (best 500)
[0:31:07.832443537] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG12 Score: 500 (best 500)
[0:31:07.832479279] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGB3 Score: 500 (best 500)
[0:31:07.832514928] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BGR3 Score: 500 (best 500)
[0:31:07.832550670] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB14 Score: 500 (best 500)
[0:31:07.832586819] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG14 Score: 500 (best 500)
[0:31:07.832622320] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG14 Score: 500 (best 500)
[0:31:07.832657154] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GR14 Score: 500 (best 500)
[0:31:07.832692507] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGB4 Score: 500 (best 500)
[0:31:07.832726768] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBAA Score: 1000 (best 500)
[0:31:07.832761546] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGAA Score: 1000 (best 500)
[0:31:07.832796084] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 500)
[0:31:07.832830622] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgAA Score: 1000 (best 500)
[0:31:07.832866197] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGGB Score: 2500 (best 500)
[0:31:07.832900828] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBCC Score: 0 (best 0)
[0:31:07.832936440] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGCC Score: 0 (best 0)
[0:31:07.832971386] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRCC Score: 0 (best 0)
[0:31:07.833005942] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgCC Score: 0 (best 0)
[0:31:07.833042110] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBEE Score: 500 (best 0)
[0:31:07.833078574] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGEE Score: 500 (best 0)
[0:31:07.833113964] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pREE Score: 500 (best 0)
[0:31:07.833149465] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgEE Score: 500 (best 0)
[0:31:07.833184188] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GRBG Score: 2500 (best 0)
[0:31:07.833219467] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GBRG Score: 2500 (best 0)
[0:31:07.833254875] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBO Score: 500 (best 0)
[0:31:07.833290543] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y10P Score: 500 (best 0)
[0:31:07.833326025] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y12P Score: 500 (best 0)
[0:31:07.833361360] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y14P Score: 500 (best 0)
[0:31:07.833397712] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBP Score: 500 (best 0)
[0:31:07.833433325] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBQ Score: 500 (best 0)
[0:31:07.833468418] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBR Score: 500 (best 0)
[0:31:07.833503345] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt YVYU Score: 500 (best 0)
[0:31:07.833538698] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt YUYV Score: 500 (best 0)
[0:31:07.833573217] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GREY Score: 2500 (best 0)
[0:31:07.833608607] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt VYUY Score: 500 (best 0)
[0:31:07.833643905] [6201]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt UYVY Score: 500 (best 0)
[0:31:07.833806168] [6201]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pgCC
[0:31:07.835583830] [6201] FATAL Serializer control_serializer.cpp:598 A list of V4L2 controls requires an ControlInfoMap
Backtrace:

Aborted

Anyone knows what is the problem?

davidplowman, 2021-11-28:
Hi, that all looks very strange. From the "control not exposed" to all the YUYV type formats, these are not messages we would expect to see at all and suggest the system is misconfigured. Could you perhaps provide some more information, including:
* Output of `uname -a`
* Output of `vcgencmd version`
* Make and model of camera. Is it an official Raspberry Pi modle? If not, do you have a link?
* Type of Raspberry Pi e.g. Pi 4 model B?
* Any changes to your `/boot/config.txt`? Perhaps post the contents, please.
* Have you done an `rpi-update` or rebuilt `libcamera`?
* Have you made any other obvious camera-related changes, such as rebuilding the legacy camera stack?

Thanks!

thepahema, 2021-11-28:
Yeah, of course:
     - Output of uname -a: 
   
Linux raspberrypi 5.10.81-v7l+ #1492 SMP Fri Nov 26 21:42:52 GMT 2021 armv7l GNU                                                                                                                   /Linux

     - Output of vcgencmd version:

Nov 16 2021 17:38:45
Copyright (c) 2012 Broadcom
version 99c5aad7da0d1e3b2c28669a3941db77734b9f2d (clean) (release) (start)

     -Make and model of the camera: 

     -Type of Raspberry Pi: https://www.amazon.es/gp/product/B07TC2BK1X/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&th=1

     -No changes in /boot/config.txt.

     -Yes, I've done sudo rpi-update after installing Raspbian.

     -I've not touched anything of the camera.

davidplowman, 2021-11-28:
The first problem is going to be that the `rpi-update` will have left you with a kernel that isn't compatible with the `libcamera` version in our repositories. We're going to sync them up again properly soon, but for now you would either have to re-install the OS (and not do an `rpi-update`) or check out and build `libcamera` and `libcamera-apps` from scratch.

Secondly, all those YUYV modes are completely wrong. It suggests some other driver (perhaps a legacy one?) has got hold of your /dev/video0, and that's never going to work.

The easiest thing, if it's possible for you, is going to be to re-install the OS and try again - it should simply work without touching anything else at all.

thepahema, 2021-11-28:
I've reinstalled the OS and now it works. Thank you so much!!
One last question: how should I use rpi-update?

davidplowman, 2021-11-28:
`rpi-update` is never a "safe" thing to do - it always gives you the most bleeding edge of everything and there may be issues. It's a bit unlucky at the moment that the latest kernel doesn't work with the version of libcamera in our apt repository, but it does work with the latest libcamera if you build it yourself (and vice versa). But I hope we'll get beyond this difficulty with the next set of updates.

davidplowman, 2021-11-29:
I'll close this issue now. Please open a new one if you encounter any more problems. Thanks!

closed 2021-11-29 by davidplowman