raspberrypi/rpicam-apps #162: libcamera-still has no --thumb none option

knutegit, 2021-11-28:
Unlike raspistill, libcamera-still has no --thumb none option.  I have an application that reads image data from raspistill using the stdout stream and detecting the end of the image when FFD9 is read.  For images from libcamera-still I can look for two sets of FFD9 bytes.  Since I don't use the thumbnail in my application I don't need to read the extra data.  Any chance you could put a --thumb none option into libcamera-still?  It would more closely replicate raspistill that way.

davidplowman, 2021-11-28:
Hi, thanks for reporting this. I've submitted a fix here: https://github.com/raspberrypi/libcamera-apps/pull/164 so we'll try and get that merged shortly.

davidplowman, 2021-11-29:
This has been merged now so I'll close this issue. We'll get it into our apt repository shortly, if the feature is more urgent then libcamera-apps can be built from source.

closed 2021-11-29 by davidplowman

knutegit, 2021-11-30:
Thanks very much for the fixes!