raspberrypi/rpicam-apps #163: [BUG] libcamera-jpeg and libcamera-still SIGSEGV segmentation fault

janvrska, 2021-11-28:
**Describe the bug**
When I want to capture photo with libcamera-jpeg or libcamera-still  in 90% of attempts it show preview window successfully and then ends with sigsegv segmentation fault and image is not saved. It's interesting that in libcamera-vid saving video works fine.

**To Reproduce**
With backtrace:
```
pi@raspberrypi:~/camera $ gdb libcamera-jpeg
GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from libcamera-jpeg...
(No debugging symbols found in libcamera-jpeg)
(gdb) bt
No stack.
(gdb) run -o test.jpg
Starting program: /usr/local/bin/libcamera-jpeg -o test.jpg
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb232f1c0 (LWP 4797)]
[2:07:59.479617365] [4790]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3266-cb1de399
[New Thread 0xaaa961c0 (LWP 4805)]
[2:07:59.486130512] [4805]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[2:07:59.486203326] [4805]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[New Thread 0xaa0ff1c0 (LWP 4806)]
[New Thread 0xa98fe1c0 (LWP 4807)]
[2:07:59.626312973] [4790]  INFO Camera camera.cpp:945 configuring streams: (0) 2028x1520-YUV420
[2:07:59.627812653] [4805]  INFO RPI raspberrypi.cpp:168 Format: 1332x990 fmt SBGGR10 Score: 3456.22 (best 3456.22)
[2:07:59.627904486] [4805]  INFO RPI raspberrypi.cpp:168 Format: 2028x1080 fmt SBGGR12 Score: 1083.84 (best 1083.84)
[2:07:59.627945301] [4805]  INFO RPI raspberrypi.cpp:168 Format: 2028x1520 fmt SBGGR12 Score: 0 (best 0)
[2:07:59.630589960] [4805]  INFO RPI raspberrypi.cpp:168 Format: 4056x3040 fmt SBGGR12 Score: 887 (best 0)
[2:07:59.631276124] [4805]  INFO RPI raspberrypi.cpp:679 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[New Thread 0xa7f2d1c0 (LWP 4808)]
[New Thread 0xa772c1c0 (LWP 4809)]
[2:07:59.725662373] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[2:07:59.725763688] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[2:07:59.725801225] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[2:07:59.725833743] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[2:07:59.725864780] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[2:07:59.725896799] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[2:07:59.725932132] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[2:07:59.725960669] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[2:07:59.726036483] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[2:07:59.726065428] [4809]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[2:07:59.726789888] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:07:59.726841888] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:07:59.726869722] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:07:59.726896258] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:07:59.726921777] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:07:59.726946740] [4805]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[2:08:00.276912173] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (6 left)
[2:08:00.311809029] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (5 left)
[2:08:00.343940005] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (4 left)
[2:08:00.373520563] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (3 left)
[2:08:00.407112998] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (2 left)
[2:08:00.441068468] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (1 left)
[2:08:00.474271570] [4805]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (0 left)
[Thread 0xa772c1c0 (LWP 4809) exited]
[Thread 0xa7f2d1c0 (LWP 4808) exited]
[2:08:05.357694947] [4790]  INFO Camera camera.cpp:945 configuring streams: (0) 4056x3040-YUV420
[2:08:05.357864279] [4805]  INFO RPI raspberrypi.cpp:168 Format: 1332x990 fmt SBGGR10 Score: 10552.2 (best 10552.2)
[2:08:05.358006612] [4805]  INFO RPI raspberrypi.cpp:168 Format: 2028x1080 fmt SBGGR12 Score: 8179.84 (best 8179.84)
[2:08:05.358057890] [4805]  INFO RPI raspberrypi.cpp:168 Format: 2028x1520 fmt SBGGR12 Score: 7096 (best 7096)
[2:08:05.358098797] [4805]  INFO RPI raspberrypi.cpp:168 Format: 4056x3040 fmt SBGGR12 Score: 0 (best 0)
[2:08:05.358368482] [4805]  INFO RPI raspberrypi.cpp:679 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[New Thread 0xa7f2d1c0 (LWP 4832)]
[New Thread 0xa772c1c0 (LWP 4833)]
[2:08:05.411113802] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[2:08:05.411284524] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[2:08:05.411403135] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[2:08:05.411518264] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[2:08:05.411626912] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[2:08:05.411739485] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[2:08:05.411850337] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[2:08:05.411956466] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[2:08:05.412070873] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[2:08:05.412182040] [4833]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 2

Thread 2 "libcamera-jpeg" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb232f1c0 (LWP 4797)]
0xb6f5ff40 in LibcameraApp::previewThread() ()
   from /usr/local/lib/libcamera_app.so
(gdb) bt
#0  0xb6f5ff40 in LibcameraApp::previewThread() ()
   from /usr/local/lib/libcamera_app.so
#1  0xbeffed32 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

**Expected behaviour**
Capture image and save to file successfully.

**Hardware Environment:**
 - Raspberry Pi 4 Model B
 - Raspberry Pi HQ Camera - IMX477

**Software Environment:**
```
pi@raspberrypi:~/camera $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

Linux raspberrypi 5.10.81-v7l+ #1492 SMP Fri Nov 26 21:42:52 GMT 2021 armv7l GNU/Linux

libcamera-apps build: 6145daf735fc-intree 27-11-2021 (10:55:17)
libcamera build: v0.0.0+3266-cb1de399

it didn't work well even before I did the build

```
pi@raspberrypi:~/camera $ vcgencmd version
Nov 16 2021 17:38:45 
Copyright (c) 2012 Broadcom
version 99c5aad7da0d1e3b2c28669a3941db77734b9f2d (clean) (release) (start)

```

davidplowman, 2021-11-29:
Hi, it looks like you've done an `rpi-update` and checked out and built the latest libcanera and libcamera-apps, is that correct? A couple of suggestions here.

Firstly, did it work with a vanilla Bullseye installation? It should do. Then make your changes one at a time until we can spot what caused it to break. Note that once you do an `rpi-update` you will _have_ to both check out and build the latest libcamera, and you will then have to rebuild libcamera-apps too (checking out the latest is optional here).

Alternatively, it might help to enable debug in libcamera-apps so that we can see a more meaningful stack trace. Add `-DCMAKE_BUILD_TYPE=Debug` to the cmake command and rebuild.

(Just as an aside, the `cma-512` looks a bit unusual to me with the imx477 overlay, it would normally go with the V3D driver I believe, but I don't know that this would make any difference.)

janvrska, 2021-11-29:
Yes exactly. But it didn't work well even before my build on fresh install of bullseye. But now I found out that if I run it from ssh it works perfect, so it's not problem after all for me. So there is only some issue with running through LXTerminal through VNC.

janvrska, 2021-12-05:
I pulled last libcamera and libcamera-apps but when i tried build libcamera-apps it failed

`cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0 -DCMAKE_BUILD_TYPE=Debug`

```
root@raspberrypi:~/libcamera-apps/build# make -j4
[  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
[  4%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
[  6%] Building CXX object encoder/CMakeFiles/encoders.dir/encoder.cpp.o
Generating version string: 19190467dbd2-intree 05-12-2021 (19:04:15)
[  6%] Built target VersionCpp
[  9%] Building CXX object output/CMakeFiles/outputs.dir/output.cpp.o
[ 11%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
[ 13%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
[ 15%] Building CXX object encoder/CMakeFiles/encoders.dir/null_encoder.cpp.o
[ 18%] Building CXX object output/CMakeFiles/outputs.dir/file_output.cpp.o
[ 20%] Building CXX object preview/CMakeFiles/preview.dir/drm_preview.cpp.o
[ 22%] Building CXX object output/CMakeFiles/outputs.dir/net_output.cpp.o
[ 25%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
[ 27%] Building CXX object encoder/CMakeFiles/encoders.dir/h264_encoder.cpp.o
[ 29%] Building CXX object preview/CMakeFiles/preview.dir/egl_preview.cpp.o
[ 31%] Building CXX object output/CMakeFiles/outputs.dir/circular_output.cpp.o
[ 34%] Building CXX object image/CMakeFiles/images.dir/png.cpp.o
[ 36%] Building CXX object encoder/CMakeFiles/encoders.dir/mjpeg_encoder.cpp.o
[ 38%] Linking CXX shared library liboutputs.so
[ 38%] Built target outputs
[ 40%] Building CXX object preview/CMakeFiles/preview.dir/qt_preview.cpp.o
[ 43%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
[ 45%] Linking CXX shared library libencoders.so
[ 45%] Built target encoders
[ 47%] Linking CXX shared library libimages.so
[ 47%] Built target images
[ 50%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/negate_stage.cpp.o
[ 52%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/post_processing_stage.cpp.o
[ 54%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/hdr_stage.cpp.o
[ 56%] Linking CXX shared library libpreview.so
[ 56%] Built target preview
[ 59%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/pwl.cpp.o
[ 61%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/histogram.cpp.o
[ 63%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/motion_detect_stage.cpp.o
[ 65%] Linking CXX shared library libpost_processing_stages.so
[ 65%] Built target post_processing_stages
Scanning dependencies of target libcamera_app
[ 68%] Building CXX object core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o
[ 70%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
[ 72%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
[ 75%] Building CXX object core/CMakeFiles/libcamera_app.dir/options.cpp.o
/root/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
/root/libcamera-apps/core/libcamera_app.cpp:674:54: error: ‘const class libcamera::SharedFD’ has no member named ‘fd’; did you mean ‘std::shared_ptr<libcamera::SharedFD::Descriptor> libcamera::SharedFD::fd_’? (not accessible from this context)
  674 |     if (i == buffer->planes().size() - 1 || plane.fd.fd() != buffer->planes()[i + 1].fd.fd())
      |                                                      ^~
compilation terminated due to -Wfatal-errors.
make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Chyba 1
make[2]: *** Čeká se na nedokončené úlohy…
make[1]: *** [CMakeFiles/Makefile2:272: core/CMakeFiles/libcamera_app.dir/all] Chyba 2
make: *** [Makefile:171: all] Chyba 2
```

naushir, 2021-12-06:
This seems to be an API breakage with the latest source of libcamera.  I have a fix for this at https://github.com/raspberrypi/libcamera-apps/pull/181

janvrska, 2021-12-06:
OK, successfully compiled latest libcamera and libcamera-apps with debug and here is stack trace of sigsegv:
```
Thread 2 "libcamera-still" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb21a61c0 (LWP 22828)]
0xb6f1e2fc in LibcameraApp::previewThread (this=0xbefff270)
    at /root/libcamera-apps/core/libcamera_app.cpp:779
779			libcamera::Span span = Mmap(buffer)[0];
```

davidplowman, 2021-12-07:
Thanks for the backtrace. I'm still rather at a loss to understand what's happening. Presumably `LibcameraApp::Mmap` is returning an empty vector. Perhaps `buffer` is NULL? None of this should be possible...

Ideally I'd still like to reproduce this problem for myself. Could you perhaps talk me through again exactly everything you've done from the installation of the fresh image? Did that fresh image already have problems or have they only become apparent when using VNC? Sorry if this is making you repeat stuff, but if you could be as clear as possible about every step, then I'll have another go at seeing if I can provoke this problem too.

janvrska, 2021-12-07:
I did a clean installation and documented everything for you. As I look at it now, the problem probably arises after disconnecting the HDMI display and changing the headless resolution when using VNC. As you can see in the 3rd video

[Video](https://www.youtube.com/watch?v=8whg3NHQXEU&list=PLZc0H9XiFAqLpSNJPWAS4uxYkQ1l239TK&index=3)

davidplowman, 2021-12-08:
Thanks for the careful demonstration using the video! It seems that unplugging the HDMI cable was key, and was the thing we haven't been doing. But as soon as I did that, even without rebooting, I found the preview in the VNC window became really laggy and libcamera-still would fail quite often.

Anyway, it should all be fixed now. If you're in a position to rebuild the latest code please give it a try, otherwise we'll be rolling out all our fixes and updates "shortly". Thanks again!

naushir, 2021-12-09:
I'll close this issue down as the bug has been fixed.  Please re-open if you think this is still occurring with an updated build.

closed 2021-12-09 by naushir

ganey, 2021-12-13:
@naushir I'm struggling to build this on a 3b+ headless

I had the older version working a month or so ago? But I wanted to update to get the datetime/timestamp outputs.

I've run `apt-get update` and `apt-get upgrade`, also ran `git pull` on this on the `main` branch, and reset the build folder.

For cmake i run this:
`cmake .. -DENABLE_DRM=1 -DENABLE_X11=0 -DENABLE_QT=0 -DENABLE_OPENCV=0 -DENABLE_TFLITE=1`

I then run `make -j2` and get the following error:
```
Scanning dependencies of target libcamera_app
[ 69%] Building CXX object core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o
[ 69%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
/home/pi/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
/home/pi/libcamera-apps/core/libcamera_app.cpp:669:54: error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’
  669 |     if (i == buffer->planes().size() - 1 || plane.fd.get() != buffer->planes()[i + 1].fd.get())
      |                                                      ^~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:274: core/CMakeFiles/libcamera_app.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

Any suggestions on what to try next?

naushir, 2021-12-13:
The head been a recent API change in the libcamera library.

If your building install libcamera, the above build failure will be fixed.

ganey, 2021-12-13:
> The head been a recent API change in the libcamera library.
> 
> If your building install libcamera, the above build failure will be fixed.

Thanks, building libcamera seems to have fixed that, I also had to run `sudo rpi-update` to get the latest kernel. All seems to be working fine again now thanks!