raspberrypi/rpicam-apps #168: ERROR: *** no cameras available ***

Friday8229, 2021-12-01:
I'm trying to get the raspberry pi camera module v1 working on my raspberry pi 4 but I get this error whenever I run the command below

```
pi@raspberrypi:~ $ libcamera-hello -t 0
libEGL warning: DRI2: failed to authenticate
[0:06:37.643508933] [1155]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***
```

Things i tried:

Checking the camera and camera cable, I know the camera works because it used to work with buster and my other setup (raspberry pi 4, an HDMI capture card, and the same cable I'm using with the camera ) works flawlessly, so it can't be the hardware I guess. 

It is a clean install so I don't think I messed up the settings.

after following [this ](https://www.raspberrypi.com/documentation/accessories/camera.html#getting-started) guide, 

My config.txt looks like this:

```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
disable_touchscreen=1

# Enable DRM VC4 V3D drive
dtoverlay=ov5647
max_framebuffers=2
arm_64bit=1
start_x=1
gpu_mem=128
```

At first, I was doing ssh but later I switched to HDMI display. when I run " libcamera-hello -t 0" a little black window pops up for less than a second and vanishes away.

I also tried adding disable_touchscreen=1 in /boot/config.txt but no luck.

output of "sudo vcgencmd version":

```
pi@raspberrypi:~ $ sudo vcgencmd version
Oct 29 2021 10:48:24 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start_x)
```

It's most probably an official camera module. I am using raspberry pi 4 4Gb model with "2021-05-07-raspios-buster-arm64" OS

Can you tell me what am I doing wrong?  

When I use the same SD card with my RPI3 B+ the red light of the camera lights up at boot but I see nothing with Pi 4.

Friday8229, 2021-12-01:
UPDATE

I flashed my other sd card with bullseye. The behavior of RPI4 is the same but with RPI3 the camera light flashes once while booting then it stays on, what am I missing?

My config file now:

```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
dtoverlay=ov5647
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=0

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

```

Friday8229, 2021-12-01:
ok so now there's a new issue:

```
pi@raspberrypi:~ $ libcamera-still -e png -o test.png
libEGL warning: DRI2: failed to authenticate
[0:00:29.413902869] [1076]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:00:29.477313332] [1076]  INFO Camera camera.cpp:937 configuring streams: (0) 1296x972-YUV420
[0:00:29.477469666] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2067 (best 2067)
[0:00:29.477522814] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1849.67 (best 1849.67)
[0:00:29.477550554] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1500 (best 1500)
[0:00:29.477575943] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 3796 (best 1500)
[0:00:29.477603943] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 4796 (best 1500)
[0:00:29.477628888] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1567 (best 1500)
[0:00:29.477653610] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1349.67 (best 1349.67)
[0:00:29.477678591] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1000 (best 1000)
[0:00:29.477702999] [1090]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 3296 (best 1000)
[0:00:29.477800462] [1090]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[0:00:29.502436462] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:00:29.502533443] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:00:29.502559869] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:00:29.502583980] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:00:29.502607869] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:00:29.502631795] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:00:29.502656277] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:00:29.502678443] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:00:29.502705703] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:00:29.502727184] [1094]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:00:29.504377925] [1090]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:00:29.504420684] [1090]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:00:29.504439128] [1090]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:00:29.504456536] [1090]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:00:29.576412184] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:00:29.621891888] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:00:29.655425295] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:00:29.689159814] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:00:29.721822758] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:00:29.755262258] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:00:29.788660425] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:00:29.822052795] [1090]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 19
Aborted
pi@raspberrypi:~ $ 

```

davidplowman, 2021-12-01:
Hi, let's start with the first problem.

From what you've said I've understood that that you're using a 64-bit Buster image. The date seems quite old, but maybe there isn't a newer one. Is it fully up to date?

The firmware seems to date from October which is much newer. Have you done a `sudo rpi-update`? Except I expect that would have updated the kernel too, so I'm not really understanding what you have.

Could you also please report the output of `libcamera-hello --version`. Where did you get libcamera and libcamera-apps from, the apt repositories or have you built them for yourself? That actually makes quite a big difference - the latest versions of the libcamera source requires a more up to date kernel that you have, and vice versa.

Now a couple of other things that might (or might not) help:

1. You don't seem to specify a display driver. Usually you need once of `dtoverlay=vc4-fkms-v3d` or `dtoverlay=vc4-kms-v3d`. I'm not sure what happens if you don't specify one. Maybe you get the full legacy driver which won't work.
2. When using ssh with X-forwarding (`ssh -X`) the preview windows don't work because they use hardware acceleration which isn't supported across a network. But you can add the `--qt-preview` to get an unaccelerated preview which does work. You can also always use the `-n` option which suppresses the preview window altogether.
3. On the Pi 3 you need to re-enable Glamor. Please check here: https://forums.raspberrypi.com/viewtopic.php?t=323547

Friday8229, 2021-12-04:
ok so I switched to Bullseye OS, and yes I tried `sudo rpi-update` both in Buster and Bullseye 

To keep it clear I'm using Raspberry pi 4 with Bullseye OS

The libcamera was already there in Bullseye so this time I didn't did anything, previously I used [this ](https://www.raspberrypi.com/documentation/accessories/camera.html#getting-started)link to build it 

Here is the output:

```
pi@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:53:32)
libcamera build: v0.0.0+3156-f4070274
```

Things I did:

added `dtoverlay=vc4-kms-v3d` in config.txt

ran `sudo rpi-update` again

Other details that might help:

```
pi@raspberrypi:~ $ cat /etc/debian_version
11.1
```

```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.82-v8+ #1497 SMP PREEMPT Fri Dec 3 16:30:35 GMT 2021 aarch64 GNU/Linux
```

The error is still there:

```
pi@raspberrypi:~ $ libcamera-still -e png -o test.png
Preview window unavailable
[0:02:00.257211009] [1558]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***
```

```
pi@raspberrypi:~ $ libcamera-still -e png -n -o test.png
[0:02:18.830181788] [1563]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***
```

davidplowman, 2021-12-04:
So one thing you need to be aware of is that, rather unfortunately, the latest kernel (which you get when you do `sudo rpi-update`) has moved on from the libcamera and libcamera-apps apt packages. So if you do an rpi-update then you _must_ checkout out the latest libcamera code and build it from source. We're working hard on an update the last release which will bring all this stuff back together again, but does it sound like this is what has "bitten" you?

Shraimsami, 2021-12-19:
Yes. I have the same problem. I am using camera module v 1.3. and up to date version bullseye and the same message i get from the libcamera which is ERROR: *** no cameras available ***

davidplowman, 2021-12-20:
@Shraimsami Thanks for reporting this. Could you perhaps file a new bug report as there can be many causes of this sort of error, and it could quite easily be different in your case. Please be sure to include all the helpful information that you can:

* The exact model of Pi that you are using.
* I see that you're using a v1.3 camera. Is it an official module, or do you have a link where it came from?
* The output of `uname -a`.
* The output of `vcgencmd version`.
* The output of `libcamera-hello --version`.
* Any modifications you have made to your /boot/config.txt file.
* The exact command line that you are running.
* The console output produced by that command line.
* Have you done a `sudo rpi-update` at any point?
* Have you checked out and rebuilt either libcamera or libcamera-apps?
* Are there any error messages relating to the camera in `dmesg`?
* Can you say if you're logged in directly to the Pi with a monitor and keyboard, or are you using ssh?

Thanks!

Shraimsami, 2021-12-20:
HiThank you for your fast respond. I find the problem. I was controlling raspberry pi through ethernet and using putty. I fixed the problem by downloading buster OS into my raspberry pi. I have a question. What is an alternative way to controll the camera module beside picamera library. Is there another library  that I can use Which can be compatible with libcamera.I want to make server and clint to stream video can you give a link to an example. I am using python.Thanks very much again.

davidplowman, 2021-12-20:
@Shraimsami There is a Picamera2 library in development, unfortunately it is not yet available.

kbingham, 2021-12-21:
> HiThank you for your fast respond. I find the problem. I was controlling raspberry pi through ethernet and using putty. I fixed the problem by downloading buster OS into my raspberry pi. I have a [...]

If you are looking to be able to stream the output of the camera, you probably want to use features from something like gstreamer as well to provide your network streaming. If that's the case, then gstreamer already has python bindings, and an element that can use libcamera -- so you should be able to construct a pipeline that uses a libcamerasrc element, and connect it to your network streaming pipeline, all controlled from python, using gstreamer already.

There are still some missing features in the gstreamer element, so you likely have to manually specify the framerate and colorimetry caps, with a caps filter.

Shraimsami, 2021-12-25:
HelloYes thanks for the tips. I have spent alot of time trying  to download gstreamer to windows. I have managed to download it  to raspberry  pi.  Can you help me installing gstreamer on python  on windows. I have downloaded gstreamer from the official website but couldnot import gi to python. couldnot put the path for  environmental variables. I have python 3.9. And I want to  capture the stream from raspberry pi to a PC windows OS.i want to use h246 format because of high framerate. Best regards Sami shraim

davidplowman, 2022-01-04:
There are quite a lot of different questions in this issue now, which I think have mostly been answered? So I'm going to close this one, but if there's still anything outstanding please feel free to open a new "clean" issue instead. Thanks!

closed 2022-01-04 by davidplowman