raspberrypi/rpicam-apps #172: libcamera-apps crash on Rpi 3b+

PeteHemery, 2021-12-02:
I see crashes when running any of the libcamera-apps, without the `--qt-preview` option.
I've found recently closed similar issues, that seems to point towards what I'm seeing: [31](https://github.com/raspberrypi/libcamera-apps/issues/31), [46](https://github.com/raspberrypi/libcamera-apps/issues/46), [125](https://github.com/raspberrypi/libcamera-apps/issues/125])
I've tried a lot of avenues of the advice I found and still having issues; I reached out to kbingham, who suggested I open this issue.

Setup:
Raspberry Pi 3b+, Raspbian OS (bullseye) with desktop, latest version from last week. libcamera and libcamera-apps updates from Monday (29th Nov 2021)
Working with an IR camera setup from eBay.
[raspinfo.txt](https://github.com/raspberrypi/libcamera-apps/files/7642251/raspinfo.txt)
[config.txt](https://github.com/raspberrypi/libcamera-apps/files/7642249/config.txt)

I tried with `gpu_mem=128` set, but as kbingham described it, that is taking memory away from the desired area for libcamera.
Allocating extra memory for the GPU reduces the memory available for the kernel (citation needed on if this correct interpretation of libcamera implementation).
Didn't seem to make a difference either way.

Things that do work:
* --qt-preview
* -n (no preview when taking pictures or recording raw h264)

Seems like there's an issue with the result returned from the call to [eglCreateImageKHR](https://github.com/raspberrypi/libcamera-apps/search?q=eglCreateImageKHR)

[libcamera-hello_backtrace.txt](https://github.com/raspberrypi/libcamera-apps/files/7642250/libcamera-hello_backtrace.txt)

Any help I can give to solving this do let me know.
Many thanks!

davidplowman, 2021-12-02:
Hi, perhaps we can start by checking a few things.

1. Can you report the output of `uname -a`.
2. Also `vcgencmd version`.
3. And `libcamera-hello --version`.
4. Can you confirm that you have re-enabled Glamor (see [here](https://forums.raspberrypi.com/viewtopic.php?t=323547)).
5. Could you try it without the config.txt changes (specifically, `i2c_arm` and those `ir` ones). Is that any different?
6. Are you running with an attached display or are you logged in via ssh? Are you using X-forwarding?

Thanks!

PeteHemery, 2021-12-04:
Hi David,
Thank you for the link recommending enable Glamor. Enabled it and it's all working. Boy is my face red (bathed in IR light ;P)

For others who follow due to SEO:
> A workaround is to open a terminal, run "sudo raspi-config", navigate to "Advanced Options" and enable "Glamor" graphic acceleration. Then reboot your Pi.
>
> We're still investigating a longer term solution, as well as other problems that have been reported.

For completeness, running with display attached. Here's the info you asked for 👍 
```
$ uname -a
Linux raspberrypi3b 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux

$ vcgencmd version
Nov 18 2021 16:19:03 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start_x)

$ libcamera-hello --version
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:21)
libcamera build: v0.0.0+3156-f4070274
```

When running over ssh with X-forwarding it failed with the following output:
```
$ libcamera-hello 
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  155 ()
  Minor opcode of failed request:  1
  Serial number of failed request:  16
  Current serial number in output stream:  16
```
But I'm happy. Thanks again.

closed 2021-12-04 by PeteHemery