raspberrypi/rpicam-apps #173: [BUG]  Internal error: Oops: 817 [#1] SMP ARM

flatsiedatsie, 2021-12-02:
**Describe the bug**
I was calling `libcamera-jpeg` from a python3 program on the Pi Zero 2 W and saw this error. Since the "oops" made me laugh I thought I'd share.

**Bug report**
`camera-bug-report -t 1 -o "libcamera_bug.txt" -c "libcamera-jpeg -o /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480"`
gave:
-bash: camera-bug-report: command not found

```
get_snapshot_command = libcamera-jpeg -o /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611084] Internal error: Oops: 817 [#1] SMP ARM

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611461] Process libcamera-jpeg (pid: 2370, stack limit = 0xc4a7d842)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611469] Stack: (0x86f05d30 to 0x86f06000)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611479] 5d20:                                     8010cb68 82d54bb8 86f05e3c 82d54bc0

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611490] 5d40: 00000000 86f05e3c 00000014 c27e41dc 00000014 82d54bb4 c0145608 80f05008

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611502] 5d60: 82aab900 82d54bb4 00000008 7f334874 86f05d8c 86f05d80 809fd0d4 809fcae8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611514] 5d80: 86f05da4 86f05d90 809fd134 809fd0c4 82d54848 c0145608 86f05e14 86f05da8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611525] 5da0: 7f1e1fe4 809fd0e4 00000000 00000000 81401d80 8020001f 837e3a00 82aab900

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611536] 5dc0: 86f05e3c 00000000 00000000 837e3a00 86f05df4 8020001f 809fffd4 80229a74

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611548] 5de0: 837e3400 c27e41dc 86f05e3c c0145608 c0145608 00000000 80f05008 00000000

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611559] 5e00: 86f05e3c 00000000 86f05eec 86f05e18 7f1e2b38 7f1e1eec 86f05e5c 00000000

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611570] 5e20: 00000000 00000000 00000000 7f1e1ee0 82aab900 71a97974 001987f4 00000000

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611581] 5e40: 00000001 00000001 00000000 00000000 86f05e84 86f05e60 80229ac0 8022acec

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611593] 5e60: 9bf192c0 9bf192c0 40000013 82aab3c0 9bf192fc 80f76240 86f05ebc 86f05e88

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611604] 5e80: 801987f4 80229a74 ffffffff 00000000 81fd50d0 82aab3c0 83381428 000e001f

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611615] 5ea0: 86f05ed4 86f05eb0 8010bc24 8010cae8 80f05008 00000000 80100204 c27e41dc

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611627] 5ec0: 5ac3c35a 7f1e2ff0 80f05008 82aab901 82aab900 00000000 86f04000 0000000e

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611638] 5ee0: 86f05efc 86f05ef0 7f1e3010 7f1e2a0c 86f05f14 86f05f00 7f1db1f4 7f1e2ffc

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611649] 5f00: c0145608 80f05008 86f05fa4 86f05f18 8034a468 7f1db1b4 86f05f4c 86f05f28

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611660] 5f20: 8010bc24 8010cae8 80f05008 71a97974 711114d0 00000000 80100204 86f04000

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611672] 5f40: 86f05f84 86f05f50 8010cb68 8010bc00 86f05f7c 00000000 00000000 80e8fc10

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611683] 5f60: 801000a8 711114d0 00000000 80100204 86f04000 c27e41dc 86f05fac 711129a8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611694] 5f80: 711129a8 00000000 00000036 80100204 86f04000 00000036 00000000 86f05fa8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611706] 5fa0: 80100040 8034a2a0 711129a8 711129a8 0000000e c0145608 71a97974 00000001

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611717] 5fc0: 711129a8 711129a8 00000000 00000036 00000001 00000000 7115f290 71a97cfc

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611728] 5fe0: 76eebff4 71a9795c 76e8fbf4 76b3b1ac 60000010 0000000e 00000000 00000000

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611735] Backtrace: 

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611751] [<809fcadc>] (__mutex_lock.constprop.9) from [<809fd0d4>] (__mutex_lock_interruptible_slowpath+0x1c/0x20)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611765]  r10:7f334874 r9:00000008 r8:82d54bb4 r7:82aab900 r6:80f05008 r5:c0145608

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611772]  r4:82d54bb4

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611784] [<809fd0b8>] (__mutex_lock_interruptible_slowpath) from [<809fd134>] (mutex_lock_interruptible+0x5c/0x60)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611877] [<809fd0d8>] (mutex_lock_interruptible) from [<7f1e1fe4>] (__video_do_ioctl+0x104/0x460 [videodev])

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.611887]  r5:c0145608 r4:82d54848

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612025] [<7f1e1ee0>] (__video_do_ioctl [videodev]) from [<7f1e2b38>] (video_usercopy+0x138/0x5f0 [videodev])

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612038]  r10:00000000 r9:86f05e3c r8:00000000 r7:80f05008 r6:00000000 r5:c0145608

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612045]  r4:c0145608

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612183] [<7f1e2a00>] (video_usercopy [videodev]) from [<7f1e3010>] (video_ioctl2+0x20/0x24 [videodev])

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612196]  r10:0000000e r9:86f04000 r8:00000000 r7:82aab900 r6:82aab901 r5:80f05008

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612203]  r4:7f1e2ff0

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612342] [<7f1e2ff0>] (video_ioctl2 [videodev]) from [<7f1db1f4>] (v4l2_ioctl+0x4c/0x60 [videodev])

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612421] [<7f1db1a8>] (v4l2_ioctl [videodev]) from [<8034a468>] (sys_ioctl+0x1d4/0x8ec)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612430]  r5:80f05008 r4:c0145608

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612443] [<8034a294>] (sys_ioctl) from [<80100040>] (ret_fast_syscall+0x0/0x28)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612451] Exception stack(0x86f05fa8 to 0x86f05ff0)

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612461] 5fa0:                   711129a8 711129a8 0000000e c0145608 71a97974 00000001

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612472] 5fc0: 711129a8 711129a8 00000000 00000036 00000001 00000000 7115f290 71a97cfc

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612482] 5fe0: 76eebff4 71a9795c 76e8fbf4 76b3b1ac

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612494]  r10:00000036 r9:86f04000 r8:80100204 r7:00000036 r6:00000000 r5:711129a8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612501]  r4:711129a8

Message from syslogd@raspberrypi at Dec  2 14:59:49 ...
 kernel:[ 6668.612511] Code: e24b6040 e284200c e5846010 e14b24f0 (e5836000) 
```

Doing "ctrl-c" stopped these messages and dropped my back in the terminal (so it can't be that bad I guess).

**Additional context**

I flashed the 32 bit image with the Raspberry Pi Imager. Kernel: 5.10.63-v7+

The command had been running ok before that.

flatsiedatsie, 2021-12-02:
P.S. I was hoping for the Pi Zero W 2.. and there it was! Thank you for updating this great platform.

flatsiedatsie, 2021-12-02:
I ran the python code again, and got this:
```
get_snapshot_command = libcamera-jpeg -o /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480
libcamera-jpeg result: Error: [2:04:55.395446412] [2492]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[2:04:55.476103616] [2492]  INFO Camera camera.cpp:751 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/ov5647@36 ***
```

davidplowman, 2021-12-02:
Hi, could you perhaps supply us with some more information, please? Here are some things that would be helpful:

* I understand that you're using a Pi Zero 2 W, with an ov5647 sensor, correct?
* The output of `uname -a`.
* The output of `vcgencmd version`.
* The output of `libcamera-hello --version`.
* Are you running the latest Bullseye release or something else?
* The contents of your /boot/config.txt.
* Are you running the libcamera-apps binaries that were supplied with the OS release? Or if not, where are they from?
* If you're not running the latest Bullseye Raspberry Pi OS, are you able to try that?
* Can you provide a command line or steps to reproduce this bug? I'm afraid "get_snapshot_command" is not familiar to me.

Thanks!

flatsiedatsie, 2021-12-02:
> Hi, could you perhaps supply us with some more information, please? Here are some things that would be helpful:
> 
> * I understand that you're using a Pi Zero 2 W, with an ov5647 sensor, correct?

A Pi Zero W 2, yet. The camera is a v1.3 Pi Camera. I don't know if it's a genuine one. Perhaps this is useful:

P5V04A
SUNNY
R1525

```
v4l2-ctl --list-devices
unicam (platform:3f801000.csi):
	/dev/video0
	/dev/video1
	/dev/media1

bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/media2

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/media0

```

> * The output of `uname -a`.

Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

> * The output of `vcgencmd version`.

Oct 29 2021 10:49:08 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)

> * The output of `libcamera-hello --version`.

libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:21)
libcamera build: v0.0.0+3156-f4070274

> * Are you running the latest Bullseye release or something else?

Yes, Bullseye. Flashed the image a few days ago.
> * The contents of your /boot/config.txt.

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
```

> * Are you running the libcamera-apps binaries that were supplied with the OS release? Or if not, where are they from?

Yes, the supplied ones.

> * If you're not running the latest Bullseye Raspberry Pi OS, are you able to try that?

Already running Bullseye. I didn't even realise the Pi Zero W 2 could run the older versions. When I tried in the past the device froze a few seconds after boot, and it seemed wifi related. So I figured the new wifi chip required bullseye.

> * Can you provide a command line or steps to reproduce this bug? I'm afraid "get_snapshot_command" is not familiar to me.

Fully reproducing my environment will take some time.. but in case you really want to:
- Fresh install of Bullseye on a Pi Zero W 2.
- Install [Webthings Gateway](https://webthings.io/) smart home controller following the [steps on Github](https://github.com/WebThingsIO/gateway).
- Run the gateway from `~/gateway` by calling `./run-app.sh`.
- Manually install [this addon](https://github.com/createcandle/candlecam) by cloning this repository into `./webthings/addons/`
- Run the addon: `python3 /home/pi/.webthings/addon/candlecam/main.py`

Context: the pi zero forms the heart of a [privacy friendly smart doorbell](https://www.candlesmarthome.com/candle-smart-doorbell-development-update). On this board the ReSpeaker hat and software isn't installed yet. For now I'm just trying to get FFMPEG dash streaming to work again on the Pi Zero 2)

> 
> Thanks!

You're more than welcome. Let me know if there's anything else I can do.

davidplowman, 2021-12-02:
Hi, having had a look through this I'm coming to the conclusion that there's more here than I can really do. I can't test anyone else's pre-built images or binaries, and the build from scratch instructions all seem quite long for someone without any familiarity with this software. Are you - or the provider of this software - able to file a bug report against libcamera-apps where there is a straightforward command line, or some other simple test case, that shows this behaviour?

For what it's worth I've been round the loop of installing a fresh Bullseye on a Pi Zero 2 and checking that libcamera-jpeg works fine with the ov5647. Actually I'm suspicious that "oops! SMP ARM" sounds like some software has been built for the wrong platform. Is this software known to support Pi Zero 2s correctly?

flatsiedatsie, 2021-12-02:
Hey no worries. It's not an issue for me - so far the crash was a one-time thing. Maybe it was just a neutrino? :-)

> I can't test anyone else's pre-built images or binaries

To avoid miscommunication: i'm running the image that the Raspberry Pi Imager tool has provided. I'm _not_ running the image provided by Webthings, as that's a year old and crashes as soon as it tries to touch the wifi settings.

The python script essentially just calls the command line:

`libcamera-jpeg -o /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480`

This is to get an initial photo. 

Three seconds later it fires off another shell command, this time to `libcamera-vid` to create an mjpeg stream.

If it happens again, I'll let you know.

closed 2021-12-02 by flatsiedatsie

flatsiedatsie, 2021-12-02:
Hmm, just happened again.

davidplowman, 2021-12-03:
I'll leave a Pi Zero 2 running libcamera-still over and over for the weekend, and let's see if I can catch it doing anything bad.

flatsiedatsie, 2021-12-03:
I ran the command outside of the python sketch

`libcamera-jpeg -o /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480`

..and there was no issue.

I then thought maybe it was caused by running the libcamera-vid while the libcamera-jpeg wasn't finished yet, since the python sketch has a 3 second sleep before it fires that. But libcamera-vid said something was already using the camera, as expected.

```
ibcamera-vid --codec mjpeg - /home/pi/.webthings/media/candlecam/photos/latest.jpg -n --width 640 --height 480
[14:33:12.222256849] [4257]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[14:33:12.301889523] [4257]  INFO Camera camera.cpp:751 Pipeline handler in use by another process
```

Finally, to find if it was another part of the code, I disabled the GPIO pins access parts. That stopped the crashing. So maybe that was it?