raspberrypi/rpicam-apps #176: [Feature Suggestion] v4l2loopback support

dustinkerstein, 2021-12-03:
_Please let me know if there's a better place for feature suggestions_

It would be awesome if libcamera-vid was able to output to v4l2 devices such as [v4l2loopback](https://github.com/umlaeute/v4l2loopback). This currently can be accomplished using Gstreamer + libcamerasrc, but it would be nice to be able to do it directly with libcamera-vid. While it's possible to use a pipe in certain cases, v4l2loopback provides video-centric features like frame buffering and opens up other interoperability. Here's what the libcamerasrc setup looks like:

1. Build and install v4l2loopback
2. Add a loopback device: `sudo modprobe v4l2loopback video_nr=25 max_buffers=32`
3. Run Gstreamer: `gst-launch-1.0 libcamerasrc ! capsfilter caps=video/x-raw,width=1920,height=1088,format=I420,colorimetry=bt601,framerate=30/1,interlace-mode=progressive ! queue ! v4l2sink device=/dev/video25`
4. Capture with FFmpeg: `ffmpeg -y -f video4linux2 -pix_fmt yuv420p -s:v 1920x1080 -r 30 -i /dev/video25 -c:v h264_v4l2m2m -b:v 25M ~/out.mp4`

What do you all think? Is this something that would benefit others in the libcamera world?

davidplowman, 2021-12-03:
Thanks for the suggestion, it sounds interesting.

On one hand, I think it would mostly call for another derived `Output` class and would therefore fit into the code pretty well. There's probably a bit of restructuring that would be required as that class isn't currently expected to deal with camera buffers that would need to be returned asynchronously back to libcamera. But extending our `shared_ptr<CompletedRequest>` mechanism into the `Output` class is possibly the tidy way to do that.

Can you say a bit more about what it is that doesn't work, either using a pipe as you describe, or using the gstreamer method? It would be good to scope out the requirements a bit more beforehand so that we can understand what we might be aiming for. Thanks!

6by9, 2021-12-07:
How does feeding back into v4l2loopback improve things compared to the V4L2 compatibility layer of libcamera?
Shuffling data to/from v4l2loopback is fairly inefficient in copying everything, so doing it "natively" should give better performance.

kbingham, 2021-12-07:
Over on the IPU3 / Microsoft surface devices, uses have got v4l2loopback working with the following:

https://github.com/linux-surface/linux-surface/wiki/Camera-Support#gstreamer-loopback-device-for-other-applications
Of course that's ipu3 specific, and would need to be adapted to the RPi camera names, but it's the general concept.

It's not an ideal solution, and is certainly just a "workaround" though.

The loopback, can use gstreamer and (I assume can) use dma-buf through the whole process, while the libcamera V4L2 compatibility layer doesn't yet support dma-buf.

That needs to be fixed in the libcamera v4l2 compatibility layer, which someone is working on, but might be slow.
See : https://lists.libcamera.org/pipermail/libcamera-devel/2021-December/028125.html

kbingham, 2021-12-07:
> How does feeding back into v4l2loopback improve things compared to the V4L2 compatibility layer of libcamera? Shuffling data to/from v4l2loopback is fairly inefficient in copying everything, so doing it "natively" should give better performance.

So - yes this was the way we 'envisioned' this to be handled, but while the compatibility layer lacks dma-buf support it doesn't support all applications.

I think it would be better to see developers working on improving that (the compatibility layer) ... than creating new systems / workarounds of course, or the longer term goal is to add libcamera or other framework (gstreamer/pipewire) support to applications which want to access cameras.

dustinkerstein, 2021-12-07:
Kk, that all sounds good. One of the use-cases I was envisioning was that libcamera would be able to interface with other v4l2 devices like capture cards (not just v4l2loopback) which might offer their own hardware encoding options. Of course this might not be suitable for Raspberry Pi's, but generally across the libcamera ecosystem. I guess the big difference between using libcamera-vid piped through Gstreamer is the lack of zero copy / dma-buf. I'd assume that libcamerasrc + Gstreamer might provide a reasonable workaround to output to v4l2 devices via Gstreamer's v4l2sink, but libcamerasrc doesn't currently provide much camera control currently. It's possible that once that control is added (see [here](https://bugs.libcamera.org/show_bug.cgi?id=94)), the Gstreamer solution may be suitable for these types of use-cases. I was just thinking it could be nice to have the v4l2sink type support directly in libcamera as it would have fewer dependencies.

davidplowman, 2022-01-26:
I think I'm going to close this one. I did try using v4l2loopback and it seemed another case of "some stuff works, some stuff doesn't" - and the same is true of the compatibility layer, for a different value of "some stuff". As others have said, the direction of travel ought to be making the compatibility layer better, rather than adding something else with different but also incomplete functionality. But I do hear the message here - that the compatibility layer needs to do better!

closed 2022-01-26 by davidplowman