raspberrypi/rpicam-apps #183: [BUG] ERROR: *** failed to start output streaming *** using libcamera_vid.cpp in a loop

JuergenSmo, 2021-12-07:
Hi
I want to use use libcamera_vid.cpp in directly my program. 
So I downloaded and compiled libcamera and libcamera-apps. 
For a first test I've added a loop in the main() function in libcamera_vid.cpp:

int main(int argc, char *argv[])
{
	for (unsigned int count = 0; count < 10; count++) // added a loop
	{
		try
		{
			LibcameraEncoder app;
			VideoOptions* options = app.GetOptions();
			if (options->Parse(argc, argv))
			{
				if (options->verbose)
					options->Print();
				std::cout << "counter: " << count << std::endl;
				event_loop(app);
			}
		}
		catch (std::exception const& e)
		{
			std::cerr << "ERROR: *** " << e.what() << " ***" << std::endl;
			//return -1; // removed because for the loop
		}
		//return 0; // moved out of for-loop
	}
	return 0;
}

**Describe the bug**
This works fine 7 times for me. But at the 8th run I recieve following error:
....
[50:18:40.529517790] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.529550641] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.529977156] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.530119062] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.530454855] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.530735019] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.530752648] [7928]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[50:18:40.596483373] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (7 left)
[50:18:40.654454044] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (6 left)
[50:18:40.689764142] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (5 left)
[50:18:40.721888826] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (4 left)
[50:18:40.752478896] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (3 left)
[50:18:40.780760414] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (2 left)
[50:18:40.812898080] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (1 left)
[50:18:40.843677704] [7928]  INFO RPI raspberrypi.cpp:1706 Dropping frame at the request of the IPA (0 left)
counter: 8
ERROR: *** failed to start output streaming ***
counter: 9
ERROR: *** failed to start output streaming ***

I can increase the loop, it is not working unless I terminate an re-run the application.
I did some deeper investigation and came to following line in libcamera_encoder.hpp which raise the error:

	void StartEncoder()
	{
		createEncoder(); // <-- this line is raising the error
		encoder_->SetInputDoneCallback(std::bind(&LibcameraEncoder::encodeBufferDone, this, std::placeholders::_1));
		encoder_->SetOutputReadyCallback(encode_output_ready_callback_);
	}

**Bug report**
I have attached the output of the bug report. I generated it with:
camera-bug-report -t 600 -o bug.txt -c"/root/projects/testCam2/bin/ARM/Debug/libcamera-vid"

**Additional context**
I did not use an commandline parameter.
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/7667783/bug.txt)

davidplowman, 2021-12-07:
Hi, thanks for filing this bug report. Indeed the h.264 encoder is not being tidied up properly as the code was assuming the process was going to quit. But obviously it should. We have merged a fix here: https://github.com/raspberrypi/libcamera-apps/pull/182

We'll release an updated image and packages "shortly". In the meantime if you want the fixes you can check out the latest code and follow the instructions [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps-without-rebuilding-libcamera)

davidplowman, 2021-12-08:
I think this issue should be fixed now. Please feel free to re-open it if it reappears, or post on the Camera forum if you'd like any further help. Thanks!

closed 2021-12-08 by davidplowman