raspberrypi/rpicam-apps #187: [BUG] Playback with libcamera-hello not smooth and showes stripes

FDelporte, 2021-12-09:
I'm trying if I can use Raspberry Pi as an 'inexpensive' HDMI camera for maker videos (in combination with an Atem), but I see the result is not 100% smooth. Any idea if this is something that can be further improved by additional settings, camera profile...? Or because of the use of the viewfinder to match the resolution of the screen?

Video started with 
```
$ libcamera-hello --viewfinder-width 1920 --viewfinder-height 1080 -f -t 30000
```

Video created and published on Vimeo for three different Raspberry Pi boards.

Pi Zero W Rev 1.1 with inexpensive camera: very shaky, but expected ;-)
https://vimeo.com/654910762

Pi Zero 2 + HQ camera: bit shaky, but with some stripes in the image when moving with the camera
https://vimeo.com/654913465

Pi 4 with 4Gb memory + HQ camera: bit shaky, less stripes compared to Zero 2, but still there
https://vimeo.com/654913691

Screenshot of such a "stripe"

[image]

davidplowman, 2021-12-09:
Is there anywhere else you could post the videos, Vimeo seems to want me to log in and I'd prefer not to make a random accounts all over the place...

FDelporte, 2021-12-09:
wow didn't know Vimeo was doing this... needed to add a setting in the video, should be OK now!

davidplowman, 2021-12-09:
Can I also just clarify exactly how the display is being driven:

- You're showing the image on a monitor connected directly to the Pi through the HDMI port?
- Are you running X Windows or not? There are known to be some display tearing issues with X Windows, but I don't think it's possible to avoid them (other than by quitting X Windows).
- You could try quitting X Windows (or suspend it with Ctrl+Alt+F1, use Ctrl+Alt+F7 to return). You then get the DRM/KMS display path which is much closer to the hardware. There should be no tearing with this method.

davidplowman, 2021-12-09:
Ah, I've seen the videos now, thanks. I assume the videos are recordings of the Pi's HDMI out, right? Yes, it looks like display tearing to me. If it's possible, I think DRM/KMS is going to work better, especially on lower-powered Pis.

FDelporte, 2021-12-09:
Indeed I think you nailed it, Ctrl+Alt+F1 and starting from terminal already makes it a lot smoother.
I should have started with a light OS version.
Thanks a lot!

Next step (for me): create some startup script to run this automatically at startup and then I have my new camera ready ;-)
Will document and share with you when ready.

closed 2021-12-09 by FDelporte

FDelporte, 2021-12-20:
@davidplowman  thanks for the support! I documented this on https://webtechie.be/post/2021-12-20-raspberry-pi-as-hdmi-camera-for-atem-mini/