raspberrypi/rpicam-apps #188: [BUG] Wrong h264 encoder input buffer width alignment.

Doliman100, 2021-12-10:
**Describe the bug**
I captured video at 780x360 resolution (19.5:9 aspect ratio) with the following command. The result looks like the h264 encoder expects the input buffer to be aligned to 32, but it is aligned to 64.
```
libcamera-vid -t 640 --width 780 --height 360 -o video.h264
```

The following command gives the video file encoded with yuv420 codec. It's width is aligned to 64 (832x360).
```
libcamera-vid -t 640 --width 780 --height 360 -o video.yuv --codec yuv420
```

I added the following line to "encoder/h264_encoder.cpp" in V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE format settings:
```
fmt.fmt.pix_mp.plane_fmt[0].bytesperline = ((options->width + 63) & ~63);
```
And now it works right. I'm not sure why the input buffer is aligned to 64, but not 32.

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7690047/bug-report.txt)

naushir, 2021-12-10:
Hi, thanks for the report.  This is indeed a problem, and your fix is valid.  I'll work on a change to fix this by fetching the stride from the stream info rather than had coding it to 64-byte alignment.

The reason this buffer is 64 byte aligned rather than 32-byte aligned is to do with how the ISP driver uses single plane formats for it's output.  With this format, there is no padding allowed between planes, so we must align the Y plane to 64 bytes to ensure the U/V planes are 32-byte aligned.

naushir, 2021-12-10:
https://github.com/raspberrypi/libcamera-apps/pull/189 does some refactoring, including setting up the bytesperline based on the stride configured by the stream.

6by9, 2021-12-10:
https://github.com/raspberrypi/linux/pull/4744 and https://github.com/raspberrypi/linux/pull/4746) fixed the codec driver to expect a multiple of 64 for the I420 stride.

naushir, 2021-12-14:
I'll close this now as all the relevant patches have been merged.

closed 2021-12-14 by naushir