raspberrypi/rpicam-apps #192: [BUG] No post processing stage found for "sobel_cv"

krahlos, 2021-12-10:
**Describe the bug**
The post processing stage for the sobel filter cannot be performed. I also tried it with the sobel_cv.json of this repository but it also did not perform the post-processing.

**Bug report**

[sobel_bug.txt](https://github.com/raspberrypi/libcamera-apps/files/7695806/sobel_bug.txt)

**Additional context**
Add any other context about the problem here.

davidplowman, 2021-12-10:
Hi, could you just confirm whether you've rebuilt libcamera-apps to include the OpenCV post-processing functions? (Instructions are [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps-without-rebuilding-libcamera).)

We don't include them by default because the OpenCV libraries are quite large and not everybody installs them.

krahlos, 2021-12-11:
Hi, I actually did not compile the libcamera-apps myself. Did I read over that in the instructions, or is that not included? Anyway, thanks for the quick reply!

closed 2021-12-11 by krahlos