raspberrypi/rpicam-apps #193: [question] Libcamera-vid capture

ml50us, 2021-12-11:
With libcamera-vid, Is it possible to capture frame by frame with a signal working as a trigger?
You could answer RTFM😀
Thanks

davidplowman, 2021-12-11:
Hi, do you mean that you would send a signal and it would capture just a single frame, and then the camera would keep on running while it waits for the next signal?

It doesn't have that behaviour at the moment, I think if you send a signal it starts recording permanently until another signal pauses it again. I don't think there's any way you can time these so as to guarantee a single frame.

libcamera-still has this behaviour already (though the version in apt implements this incorrectly; we fixed it in the sources and will be rolling that out "shortly"). It also performs a camera mode switch to full-resolution readout for the capture before switching back to the preview mode again, so capturing a frame takes a bit longer, and maybe this isn't what you want.

If you have some familiarity with C/C++ it might be worth editing [libcamera_vid.cpp](https://github.com/raspberrypi/libcamera-apps/blob/main/apps/libcamera_vid.cpp) yourself to get the behaviour you want, the code is not at all difficult. With h.264 you might need to set the GOP size to 1 so that every frame is an I-frame, with MJPEG obviously there are no such problems.

If you wanted to try that, certainly feel free to ask questions (perhaps over on the [Camera forum](https://forums.raspberrypi.com/viewforum.php?f=43)) and we'll try to help.

ml50us, 2021-12-11:
yes , this is exactly what i wish to do . 
i have rebuilt libcamera and the last sources, it works well with libcamera-still -s but i need a process as fast as possible.
thank you for your complete answer, i will try to edit libcamera_vid.cpp and also with MJPEG codec, wich is perfect for me.
and i'll fall over to the forum . thanks again

ml50us, 2021-12-11:
i try libcamera-vid with -s option .
it seems to work just one time if i want to switch pause/resume mode.
libcamera-vid --version
libcamera-apps build: 9d89e1e858b3-intree 11-12-2021 (14:21:06)
libcamera build: v0.0.0+3346-f16acb27

davidplowman, 2021-12-11:
Hmm, just off the top of my head I wonder if [this fix](https://github.com/raspberrypi/libcamera-apps/commit/fa95bcba868f6718cf9673897b602df170cc1d9d) needs applying to libcamera_vid (are you in a position to try that?). Otherwise I'll have to investigate.

ml50us, 2021-12-11:
yes it works just by adding the same fix
signal_received = 0;
thanks

davidplowman, 2021-12-11:
Thanks for letting me know. I'll get that merged into the code.

naushir, 2021-12-14:
Closing this now, as the relevant patch has been merged.

closed 2021-12-14 by naushir