raspberrypi/rpicam-apps #195: [BUG] error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’

krahlos, 2021-12-13:
**Describe the bug**

The compilation of libcamera-apps fails with `error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’`

**System information**
```
$ uname -a
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux

$ sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cmake is already the newest version (3.18.4-2+rpi1).
libboost-program-options-dev is already the newest version (1.74.0.3+b1).
libdrm-dev is already the newest version (2.4.104-1+rpi1).
libexif-dev is already the newest version (0.6.22-3).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

$ git log -1
commit a86d0303ef6635f3f48693e7b435239ab67b76ce (HEAD -> main, origin/main, origin/HEAD)
Author: David Plowman <david.plowman@raspberrypi.com>
Date:   Mon Dec 13 08:52:42 2021 +0000

    libcamera-vid: Signals were not being "cancelled" in signal mode
    
    This copies a previous fix in libcamera-still (fa95bcb), which applies
    here too.
    
    Signed-off-by: David Plowman <david.plowman@raspberrypi.com>
```

**Steps for reproduction**

```
$ git clone https://github.com/raspberrypi/libcamera-apps.git
$ cd libcamera-apps
$ mkdir build
$ cd build
$ cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0
$ cmake --build  . -j 4
```

**Compile log**
```
Scanning dependencies of target libcamera_app
[ 70%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
[ 70%] Building CXX object core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o
[ 72%] Building CXX object core/CMakeFiles/libcamera_app.dir/options.cpp.o
[ 75%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
/home/pi/Coding/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
/home/pi/Coding/libcamera-apps/core/libcamera_app.cpp:669:54: error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’
  669 |     if (i == buffer->planes().size() - 1 || plane.fd.get() != buffer->planes()[i + 1].fd.get())
      |                                                      ^~~
compilation terminated due to -Wfatal-errors.
gmake[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:272: core/CMakeFiles/libcamera_app.dir/all] Error 2
gmake: *** [Makefile:171: all] Error 2
```

davidplowman, 2021-12-13:
Hi, yes there has been a recent change to the libcamera API (which libcamera-apps is matching) so you would need to update libcamera to fix that. But note that recent versions of libcamera require a newer kernel than the one shipped with Bullseye so if you haven't been building libcamera for yourself already, you would need a `sudo rpi-update` first (usual warnings apply there: back up any important data first!).

krahlos, 2021-12-13:
Hi @davidplowman, thank you for your quick reply. I have updated the RPi accrding to your instructions, removed my `build` directory and re-configured the lib with the command given above. However, the compilation still fails with the same error.

I have also reinstalled `libcamera-dev`, `libepoxy-dev`, `libjpeg-dev`and `libtiff5-dev`and repeated the process. But this also did not solve the error.

Do you have any further suggestions to solve the issue?

_System information after the update_
```
$ uname -a
Linux raspberrypi 5.10.83-v7l+ #1499 SMP Tue Dec 7 14:08:09 GMT 2021 armv7l GNU/Linux

$ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date (delete /boot/.firmware_revision to force an update anyway)
```

davidplowman, 2021-12-13:
If you've updated the kernel (which you have) and libcamera-apps then you need to build libcamera from source too ([here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera)). Can you confirm that you've done this (it wasn't clear to me from your reply)? Thanks!

krahlos, 2021-12-13:
I have not  build libcamera from source. Sorry, I did not know that I have to do this. I will do this and close the issue once everything works. Thank you.

krahlos, 2021-12-13:
Worked like a charm.

closed 2021-12-13 by krahlos