raspberrypi/rpicam-apps #196: [BUG] AGC fixed after first convergence in libcamera-still

BUDMAN65, 2021-12-13:
libcamera-stilll -t 0 --timelapse 1000 --raw 1 --wrap 100 --verbose 1 -o /mnt/ramdisk/%05d.jpeg
gain and shutter speed are calculated until convergence then appear to be fixed.  Capture image(s) saturates or becomes dark
with changing lighting conditions.

davidplowman, 2021-12-14:
Are you running libcamera-still as installed in the original Bullseye image? If so it might just be that 1000ms isn't long enough to save everything and do all the camera mode switches, as it's trying to make that the shot-to-shot time. We could consider making that the time for which the preview runs, but then the shot-to-shot time would be longer. (Does it work if you lengthen that time to, say, 3 seconds?)

davidplowman, 2021-12-17:
I have tried running this myself, using a v1 camera and `--timelapse 1` to be sure I was avoiding any preview frames. I found that the AGC does seem to adapt as we would expect, only it changes quite slowly as it's now managing to run through its update loop about once per second (instead of 30 times per second). Do you think this is what you've been seeing?

Another possible fix would be to try and stick to the shot-to-shot interval as we do now, but always to force just a few preview frames (maybe half a dozen or so). Would that work?

naushir, 2022-01-04:
Closing this now due to inactivity.  Please re-open if you need more assistance.

closed 2022-01-04 by naushir